merbanan/rtl_433 #2383: 929.6125 Mhz  accuweather forecaster

Harrball76, 2023-02-16:
has anyone looked into decoding the accuweather radio signal? I have a couple 7 day forecaster that receives data on 929.6125 Mhz. even though accuweather states they disconnected the service in 2019 it is still working. 

the model i have is a Bushnell Weather FX5 950015C.  the weather is updated about ever 15 mins

zuckschwerdt, 2023-02-16:
Interesting. That's just the indoor unit, no outdoor sensor, right?
What ist broadcasting the 929.6125 MHz signal? Can you capture a sample? https://triq.org/rtl_433/ANALYZE.html

Harrball76, 2023-02-16:
[g004_929.612M_1000k.zip](https://github.com/merbanan/rtl_433/files/10759079/g004_929.612M_1000k.zip)

here are the files on that bandwith

zuckschwerdt, 2023-02-16:
Good clean signals. But very much payload data, I don't think there is an easy way to just guess what those bits encode...

Harrball76, 2023-02-16:
[image]

This is the device that is receiving the signal. I am still going to work on this. If anyone else has any ideas please let me know

zuckschwerdt, 2023-02-16:
Great to hear, happy hacking! You might need to decode the payload by hand or with URH. The demod in rtl_433 struggles with GFSK.

merbanan, 2023-02-16:
The signal is clipping so the demodulator fails badly on these signals.

zuckschwerdt, 2023-02-16:
Not by much, but yes, that probably needs adjustment (distance, antenna, or fixed gain).

gdt, 2023-09-26:
This doesn't have actionable content.  Please retest and re-engage with up-to-date git master.

Harrball76, 2023-12-14:
Here is another article about this service

http://forums.radioreference.com/threads/ambient-information-network-datacasting-service.399490/

zuckschwerdt, 2023-12-14:
It's compared to POCSAG and it might be better to write a dedicated receiver for it. Also there is mention of it shutting down perhaps?