merbanan/rtl_433 #2384: Add support for EEZ Tire TPMS

Gliebig, 2023-02-16:
Thanks for your suggestions! I had some time today to make new recordings and I pushed them to my fork and updated the READEME.md and deleted the old data. 

It takes some time to setup the test and to be able to capture since there is no manual trigger on the sensor. The link you sent https://triq.org/spectrogram-next/ was very helpful! Should be able to move to the next phase. I'm a ME and fairly technical but this byte stuff is something that I've struggled with and with some help should be able to get this figured out.

ProfBoc75, 2023-02-17:
Hi @Gliebig ,
I tried to read your [samples](https://github.com/Gliebig/rtl_433_tests/tree/master/tests/EEZ%20Tire%20TPMS/Model%20618E) but they are durty,  lot of noise into the captures.
Are you able to make new records very closer the valves ? 
Also try several options to get the best result as explained here https://triq.org/rtl_433/ANALYZE.html#analyze-the-data-packet
Try with -s 1024k to get better sampling.

zuckschwerdt, 2023-02-17:
We can already confirm that the signal is OOK and the signals looks fine in the spectrogram. (@ProfBoc75 that's how most TPMS look, you won't get a cleaner signal)
There is some clipping, so if at all I'd suggest more distance or a lower gain. The `-s 1024k` is a good suggestion, for some reason neither the `-Y classic` nor the `-Y minmax` can properly demod the signal :(

Gliebig, 2023-02-17:
OK, I will do a few more test runs this morning after I get the walks shoveled. I did notice another signal was popping up from somewhere when recording -S unknown capturing the TPMS the farther I moved the EBIKE from the antenna. I've tried to isolate the signal as much as possible. I'm inside an office inside a steel warehouse with no activity within 500'. 
I have a laptop with a second RTL-SDR running a stopwatch and URH. I can easily see the data packet come in every 5 minutes and then note the g... file number that appears on my Xubuntu system with the 1st RTL-SDR dongle. The the signals look really good with URH but they are in the wrong format for use with rtl_433. 
I then make sure the sample size is correct for the TPMS and append the UID of the sensor (hex) and the temperature and pressure data that was recorded by the head unit that normally sits on the dashboard of the RV to the file name. I'll see what the  rtl_433 -f 433.88M -S unknown -s 1024k option produces. and update my repository.

Gliebig, 2023-02-17:
OK, I think I've got the recordings figured out. ended up manually setting gain to 27. Looking at the spectrograph there are nice rounded "sine" waves with no squared edges. used rtl_433 -f 433.88M -S unknown 1024K -g 27. I've updated my repository with with the data values and file names trying to simplify things a bit. Still used two different sensors, freezer, and heat gun.
g108... and g121 data may be a little off. Removing the sensor put it into an alarm mode. Without a timestamp on the data file I was guessing at the pressure and temperature values as the sensor sends a signal about once per second when there is a fast pressure loss like what happens when removing the sensor from the valve stem.

zuckschwerdt, 2023-02-17:
Oh, those files look perfect! E.g.
`rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert' g071_433.88M_1024k.cu8`
gives:
`{80}ffff 81 0d177e 99 46 0000`

Now get all those codes, annotate and paste into [a BitBench like this](https://triq.net/bitbench#c=7E%20F2%20E8%2081%2066%20B9%20%5Bg071%3A%200d177e%2055%20PSI%2068%20F%5D&f=FLAGS%3A8h%20ID%3A24h%20PRESSURE%3F8h%20TEMP%3F8h&i=true).

Then we need to stare a the numbers until it will makes sense :)

Gliebig, 2023-02-18:
I'll have to fire up my Xubuntu box to do the analysis. It's a Wyse 3040 and it worked to record the signals. I tried using the windows version by extracting the windows version of rtl_433 in the same directory as the recordings but I get the following error, "The procedure entry point rtlsdr_cancel_async could not be located in the dynamic link library C:\Users\Tests\230217cu8\rtl_433.exe. Glad the recordings are good!! Thanks for the feedback!!

ProfBoc75, 2023-02-18:
Hi,

My [BitBench findings](https://triq.net/bitbench#c=fffffc0d177e004a1000&c=ffff850d177e894a1000&c=ffff880d177e8c4a1000&c=ffff8b0d177e8f4a1000&c=fffffd0d177e902b2000&c=ffffec0d177e903a0000&c=ffff870d177e971e3000&c=ffff810d177e99460000&c=ffff870d177e994c0000&c=ffff8f0d177e99540000&c=ffff9c0d177e9a600000&c=ffffce54e3289b243000&c=ffffb754e3289c3c0000&c=ffffbf54e328a32d1000&c=ffffb054e328a32e0000&c=ffffb454e328a3320000&c=ffffc154e328a33f0000&c=ffffc654e328a4430000&c=ffffd454e328a4510000&c=ffffe154e328a45e0000&c=ffffe254e328b3203000&f=CHECKSUM%3A8h%20ID%3A24h%20PSI%3A8h%20TEMP%3A8h%20FLAG%3A8h%208h&a=Preamble&m=ffff&cw=4) 

So, we have after the small FFFF preamble :
```
CHECKSUM:8h ID:24h PSI:8h TEMP:8h FLAG:8h 8h
```

Checksum = Sum off all bytes from ID to FLAG & 0xFF
PSI = the round up (Hexa Value / 2.8) 
Temp = the Hexa Value - 50   --> in Celcius.
Flag = Could be some kind of alarm or Battery indicator ?

zuckschwerdt, 2023-02-18:
> But where did these values come from in your bit bench example?

I copied from the non-inverted output of pdv, sorry, same thing, your table is perfect.

@ProfBoc75 you really found the fields fast and precise!
For pressure I'd guess it's really `× 2.5 kPa` -- same result and a "round" factor.

I've added the checksum in [this BitBench](https://triq.net/bitbench#c=fffffc0d177e004a1000%5Bg121%3A%200d177e%200%20kPa%2024%20C%5D&c=ffff850d177e894a1000%5Bg108%3A%200d177e%20338%20kPa%2024%20C%5D&c=ffff880d177e8c4a1000%5Bg093%3A%200d177e%20345%20kPa%2024%20C%5D&c=ffff8b0d177e8f4a1000%5Bg080%3A%200d177e%20352%20kPa%2024%20C%5D&c=fffffd0d177e902b2000%5Bg206%3A%200d177e%20359%20kPa%20-7%20C%5D&c=ffffec0d177e903a0000%5Bg207%3A%200d177e%20359%20kPa%208%20C%5D&c=ffff870d177e971e3000%5Bg205%3A%200d177e%20372%20kPa%20-20%20C%5D&c=ffff810d177e99460000%5Bg071%3A%200d177e%20379%20kPa%2020%20C%5D&c=ffff870d177e994c0000%5Bg079%3A%200d177e%20379%20kPa%2026%20C%5D&c=ffff8f0d177e99540000%5Bg078%3A%200d177e%20379%20kPa%2034%20C%5D&c=ffff9c0d177e9a600000%5Bg077%3A%200d177e%20379%20kPa%2046%20C%5D&c=ffffce54e3289b243000%5Bg203%3A%2054e328%20386%20kPa%20-14%20C%5D&c=ffffb754e3289c3c0000%5Bg204%3A%2054e328%20386%20kPa%2010%20C%5D&c=ffffbf54e328a32d1000%5Bg193%3A%2054e328%20407%20kPa%20-5%20C%5D&c=ffffb054e328a32e0000%5Bg197%3A%2054e328%20407%20kPa%20-4%20C%5D&c=ffffb454e328a3320000%5Bg198%3A%2054e328%20407%20kPa%200%20C%5D&c=ffffc154e328a33f0000%5Bg199%3A%2054e328%20407%20kPa%2013%20C%5D&c=ffffc654e328a4430000%5Bg201%3A%2054e328%20407%20kPa%2017%20C%5D&c=ffffd454e328a4510000%5Bg202%3A%2054e328%20407%20kPa%2031%20C%5D&c=ffffe154e328a45e0000%5Bg200%3A%2054e328%20407%20kPa%2044%20C%5D&c=ffffe254e328b3203000%5Bg165%3A%2054e328%20441%20kPa%20-18%20C%5D&f=CHECKSUM%3A8h%20ID%3A24h%20KPA%3A8d%20TEMP%3A8d%20FLAG%3A4b%204h8h&a=Preamble&m=ffff&cf=ADD&co=8&cl=56&cw=8) your findings mostly match but there is something not perfect with the upper nibble -- something missing?

Gliebig, 2023-02-18:
The display unit that sits on the dashboard is where the units are set from the data received from the sensors. There might be a flag for a fast pressure loss alarm. g108 and g121 are two data points when the alarm was active when I was taking off 0d177e off the tire. The signal changes from being sent once every 5 minutes to an alarm mode that sends data about once per second.

ProfBoc75, 2023-02-18:
@Gliebig : Thanks for the feedback, it's good information, I was also looking into [its documentation](https://img1.wsimg.com/blobby/go/e8e00d81-5698-4e62-b5e3-550ec39b76fb/downloads/%E8%AF%B4%E6%98%8E%E4%B9%A62020.0417.pdf?ver=1676648393436)

@zuckschwerdt : this could answer to your question about the upper nibble, here the battery level ( 0 = good) , and the low nibble the pressure deflation alarm indicator. On the console I notice also a battery gauge (3 lines barregraphe) and "sensor battery low" light, this means that battery information is coded with a least 2 bits, where the expected 3.2 V with a brand new battery and 2.8 V for a low battery (page 18 into the doc). So logically, we could have 4 levels , 0 = good/full , with 3 bars in the gauge , 1 = 2 bars, 2 = 1 bar, 3 = 0 bar + "Sensor battery low" light on. To be tested with a variable dc or a low battery, since this is my pure imagination 😄 . Compare to TPMS_JANSITE, this one is very close, not the data sent in the same order and not same modulation, but provides same kind of data.

Gliebig, 2023-02-18:
I did not include originally this but here is the data that was recorded when removing sensor 0d177e from the valve stem. I started by creating a "leak" by slightly unscrewing the sensor from the valve stem and watched the file names and recorded the pressure. The problem is the display unit cycles through all 6 tires until the alarm goes off. Then it only displays the tire under alarm and records very quickly. I can't guarantee the pressure is on the correct file name. It's a quick series of info that might prove helpful.
```
{80}ffff 8b 0d177e 8f 4a 1000	[g080: 0d177e 51? PSI 75 F]
{80}ffff 8b 0d177e 8f 4a 1000	[g081: 0d177e 51? PSI 75 F]
{80}ffff 8a 0d177e 8e 4a 1000	[g082: 0d177e 51? PSI 75 F]
{80}ffff 8a 0d177e 8e 4a 1000	[g083: 0d177e 51? PSI 75 F]
{80}ffff 8a 0d177e 8e 4a 1000	[g084: 0d177e 51? PSI 75 F]
{80}ffff 8a 0d177e 8e 4a 1000	[g085: 0d177e 51? PSI 75 F]
{80}ffff 89 0d177e 8d 4a 1000	[g086: 0d177e 51? PSI 75 F]
{80}ffff 89 0d177e 8d 4a 1000	[g087: 0d177e 51? PSI 75 F]
{80}ffff 89 0d177e 8d 4a 1000	[g088: 0d177e 51? PSI 75 F]
{80}ffff 89 0d177e 8d 4a 1000	[g089: 0d177e 51? PSI 75 F]
{80}ffff 89 0d177e 8d 4a 1000	[g090: 0d177e 51? PSI 75 F]
{80}ffff 88 0d177e 8c 4a 1000	[g091: 0d177e 51? PSI 75 F]
{80}ffff 88 0d177e 8c 4a 1000	[g092: 0d177e 51? PSI 75 F]
{80}ffff 88 0d177e 8c 4a 1000	[g093: 0d177e 50? PSI 75 F]
{80}ffff 88 0d177e 8c 4a 1000	[g094: 0d177e 50? PSI 75 F]
{80}ffff 88 0d177e 8c 4a 1000	[g095: 0d177e 50? PSI 75 F]
{80}ffff 88 0d177e 8c 4a 1000	[g096: 0d177e 50? PSI 75 F]
{80}ffff 87 0d177e 8b 4a 1000	[g097: 0d177e 50? PSI 75 F]
{80}ffff 87 0d177e 8b 4a 1000	[g098: 0d177e 50? PSI 75 F]
{80}ffff 87 0d177e 8b 4a 1000	[g099: 0d177e 50? PSI 75 F]
{80}ffff 87 0d177e 8b 4a 1000	[g100: 0d177e 50? PSI 75 F]
{80}ffff 86 0d177e 8a 4a 1000	[g101: 0d177e 50? PSI 75 F]
{80}ffff 86 0d177e 8a 4a 1000	[g102: 0d177e 50? PSI 75 F]
{80}ffff 86 0d177e 8a 4a 1000	[g103: 0d177e 50? PSI 75 F]
{80}ffff 86 0d177e 8a 4a 1000	[g104: 0d177e 50? PSI 75 F]
{80}ffff 86 0d177e 8a 4a 1000	[g105: 0d177e 50? PSI 75 F]
{80}ffff 86 0d177e 8a 4a 1000	[g106: 0d177e 50? PSI 75 F]
{80}ffff 85 0d177e 89 4a 1000	[g107: 0d177e 50? PSI 75 F]
{80}ffff 85 0d177e 89 4a 1000	[g108: 0d177e 49? PSI 75 F]
{80}ffff 85 0d177e 89 4a 1000	[g109: 0d177e 49? PSI 75 F]
{80}ffff 85 0d177e 89 4a 1000	[g110: 0d177e 49? PSI 75 F]
{80}ffff 85 0d177e 89 4a 1000	[g111: 0d177e 49? PSI 75 F]
{80}ffff 85 0d177e 89 4a 1000	[g112: 0d177e 49? PSI 75 F]
{80}ffff f4 0d177e 88 4a 0000	[g113: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g114: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g115: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g116: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g117: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g118: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g119: 0d177e 49? PSI 75 F]
{80}ffff fc 0d177e 00 4a 1000	[g120: 0d177e 49? PSI 75 F]
```

ProfBoc75, 2023-02-18:
Thx, so low bit =1 =  Deflate alarme,  from this extract, the last g114 to g120 pressure should be 0.  88 or 89 = 49 PSI for sure.
Well, anyway , it's not enough to be sure how to decode the flag ... since from other extract we have also 2 or 3 at this position, and we know that the battery is also part of the data since you have the indicator on the central console.

Gliebig, 2023-02-18:
The main console has a battery indicator in the lower right of the screen that is the battery indicator of the display. It runs on 12v plugged into the RV which charges the internal battery. The display was on the internal batteries for all my testing and did not look for a battery indicator for each sensor. I could do some testing next week using a power supply hooked to the sensor and do some recordings varying the voltage holding pressure and temperature consistent. Are there preferred voltages to try?

ProfBoc75, 2023-02-18:
Ok, the gauge is for the internal battery of the console. And only the "Sensor battery low" will flash. page 17 (Sensor Low Battery alert).

ProfBoc75, 2023-02-18:
> Oh, those files look perfect! E.g. `rtl_433 -R 0 -X 'n=name,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert' g071_433.88M_1024k.cu8` gives: `{80}ffff 81 0d177e 99 46 0000`
> 
> Now get all those codes, annotate and paste into [a BitBench like this](https://triq.net/bitbench#c=7E%20F2%20E8%2081%2066%20B9%20%5Bg071%3A%200d177e%2055%20PSI%2068%20F%5D&f=FLAGS%3A8h%20ID%3A24h%20PRESSURE%3F8h%20TEMP%3F8h&i=true).
> 
> Then we need to stare at the numbers until it makes sense :)

I tried the cu8 files, but I don't have the same result as you or @Gliebig , I'm using the last version of rtl_433 ,202302162313. 
lot of single line and [sdr_callback] Sample buffer length not aligned to sample size! messages, 
Under Debian, in Windows WSL, last upd update/upgrade, all pre-req install to make rtl_433, no error at the build...

 [Input] Test mode active. Reading samples from file: g071_433.88M_1024k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.016974s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.016974s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 3            data      : 8
codes     : {3}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.016974s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 2            data      : c
codes     : {2}c
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.019377s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.019377s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.019911s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.019911s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.020480s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.020480s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 2            data      : 8
codes     : {2}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.021091s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _  [... cut ...]
codes     : {1}8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.255943s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 1            data      : 8
codes     : {1}8

Did you put other option in the command line ?

merbanan, 2023-02-18:
Add -s 1024k.

ProfBoc75, 2023-02-18:
same result with -s 1024k

zuckschwerdt, 2023-02-18:
Something must be different/broken on your setup? It works for at least @Gliebig and me. Just
`rtl_433 -X "n=test,m=OOK_MC_ZEROBIT,s=50,l=50,r=120,invert" g071_433.88M_1024k.cu8`

ProfBoc75, 2023-02-18:
@zuckschwerdt : thanks for the confirmation, I will check where is the issue, thx. 

I'm annoyed because I wrote the device driver "tpms_eezrv" but I can't test it.. 😢

zuckschwerdt, 2023-02-18:
For a quick test you can also use e.g. `-y ffff860d177e8a4a1000`

ProfBoc75, 2023-02-18:
it works with that option : (without invert)

>rtl_433 -X "n=test,m=OOK_MC_ZEROBIT,s=50,l=50,r=120" g071_433.88M_1024k.cu8 -y ffff860d177e8a4a1000
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.000000s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 80           data      : ffff860d177e8a4a1000
codes     : {80}ffff860d177e8a4a1000

```

ProfBoc75, 2023-02-18:
@Gliebig : I just put a PR with my [draft device driver](https://github.com/ProfBoc75/rtl_433_tests/blob/patch-3/tests/EEZ%20Tire%20TPMS/Model%20618E/src/devices/tpms_eezrv.c), into your rtl_433_test 

Going to diner , to be continued later ...

zuckschwerdt, 2023-02-18:
> it works with that option : (without invert)

Use only `rtl_433 -R 0 -X "n=test,m=OOK_MC_ZEROBIT,s=50,l=50,r=120" -y ffff860d177e8a4a1000`

The file is not read. Do not use -y and a file together.

Gliebig, 2023-02-18:
I'm using rtl_433 version 21.12 (2021-12-14) inputs file rtl_tcp RTL-SDR SoapySDR on an Xubuntu system installed on a Wyse 3040 thin client. Small, Fanless, and 5 volts. I'm planning to install this in my RV for collecting all the SDR data for my home assistant server that is on a PI4. I'm using the command line provided by @zuckschwerdt for getting the test results. I could easily modify a sensor for use with a power supply to collect data for the low battery test since the wheel does not need to rotate. Let me know.

ProfBoc75, 2023-02-18:
> I could easily modify a sensor for use with a power supply to collect data for the low battery test since the wheel does not need to rotate. Let me know.

If this does not require to much efforts , yes please then we will be sure about the battery bits position to decode the value accordingly. Thanks.

ProfBoc75, 2023-02-18:
Some time I'm stupid .... my rtl_433 is working like a charm , just if I download properly the cu8 file, it works better, the source of my mistake is because I download the files with wget with the git url page and not the raw link of the files ...

Gliebig, 2023-02-18:
> Some time I'm stupid .... my rtl_433 is working like a charm , just if I download properly the cu8 file, it works better, the source of my mistake is because I download the files with wget with the git url page and not the raw link of the files ...

Don't worry about it! This has all been a bunch of firsts for me. First pull request, first fork, first time building a unix machine, first time making rtl_433 and using it for recordings. First time doing a merge and only the second time collaborating on a GitHub project and nervous as heck about not making a mistake and I'm an old guy.

Gliebig, 2023-02-18:
I understand the 2nd and 3rd packets (duplicated 2nd) contain the data needed for what we're trying to accomplish. I don't understand what the 1st packet it for but to establish some sort of communication with the main display.

ProfBoc75, 2023-02-19:
> I understand the 2nd and 3rd packets (duplicated 2nd) contain the data needed for what we're trying to accomplish. I don't understand what the 1st packet it for but to establish some sort of communication with the main display.

In fact all the 3 packets are the same, depends on the synchronization, it could be shift by one bit and hexa seams different, but this is the same values, the reason why we are using a preamble research at bit level into the data to synchronise them and get the good values at the good position into the frame.
I understand your remark, because here in the first packet , you have a lot of 0xFF , then the data, where the 2nd and 3rd packets have less with only FFFF.

ProfBoc75, 2023-02-19:
> I understand the 2nd and 3rd packets (duplicated 2nd) contain the data needed for what we're trying to accomplish. I don't understand what the 1st packet it for but to establish some sort of communication with the main display.

@Gliebig : you're right, the first packet sounds different from 2 others, where usually the 1st and the other packets are repeats, the reason why I did not deep dive until now. The preamble is very long, this is normal to let this small device start and stabilise the signal frequency before sending the data. I'll let you know my findings about this 1st packet, if any  ...

Gliebig, 2023-02-20:
I will run the voltage tests tomorrow. I designed and built a power supply adapter for a CR1632 coin cell battery today. I added pictures to the README. I'll post the results when I get them done.
[image]
[image]

zuckschwerdt, 2023-02-20:
Very nice build! This should be a template for all further TPMS hacking :)

Gliebig, 2023-02-20:
Updated my repository with the new test data. Let me know what files (if any) you would like me to include under test data as you suggest to keep uploaded recordings to a minimum. The coin cell adapter worked great! Also, there is no indication on the display console that there is a low battery. There is a single beep after about 15 minutes without receiving any data from the sensor.

[image]
[image]

ProfBoc75, 2023-02-20:
Good job ! Unfortunatly from your samples, I did not see battery information from the 2sd/3rd packets need to check the 1st packet may be here. I'll check that later. And what I identified as flags could be lower nibble for the temp.

Gliebig, 2023-02-21:
I did upload the recorded signals from the voltage tests. The file names are in the table of values if thats helpful.

ProfBoc75, 2023-02-28:
@Gliebig : Hi, I was not able to find relationship between your voltage tests and the data. The device driver is now merged into the last version of rtl_433,  did you test it ?

gdt, 2023-09-26:
Looks like git master has acquired improved code.  Please restest with that and post a summary of where we are.  Otherwise I'll assume this is solved.

flyboy013, 2023-09-30:
Trying to get EezTire-E618 to read all my EezTire 515 tire sensors (same as E618?).  I have a total of 12 (coach + car).  Only about half of the sensor are being decoded.   I’m currently using “rtl_433 -s 1024k”.  I’ve also tried with various combinations of “-f 433.88M” and “-g 27”, but no change in processed signals.  In addition to receiving half of my sensors, I’m receiving another 10 or so from nearby RVs .

I’m using “[NooElec R820T2 SDR & DVB-T NESDR Mini 2](https://www.nooelec.com/store/sdr/sdr-receivers/nesdr-mini-2.html)” receiver.   I’m using the receiver on a RPi CM4.  I’ve also installed rtl_433 on my MacBook and I’m getting same results.  Also relocated MacBook receiver outside of RV thinking location of receiver may have been issue, but I’m getting same result.  The EezTire 515 receiver is receiving signal from all tire sensors.  

Maybe related (or not), I’ve noticed that I always get “PLL not locked!”.  Below is sample rtl_433 output.

```
mark@Marks-MacBook-Pro /tmp % /usr/local/bin/rtl_433 -s 1024k -vv -R 241
rtl_433 version nightly-1-g72c0f21b branch master at 202309292256 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/mark/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registering protocol [241] "EezTire E618 (TPMS10ATC)"
[Protocols] Registered 1 out of 246 device decoding protocols
[Input] The internals of input handling changed, read about and report problems on PR #1978
[SDR] Found 1 device(s)
[SDR] trying device 0: Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[R82XX] PLL not locked!
[SDR] Sample rate set to 1024000 S/s.
[Input] Bit detection level set to 0.0 (Auto).
[SDR] Tuner gain set to Auto.
[Input] Reading samples in async mode...
[SDR] Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-09-30 12:54:32
model     : EezTire-E618 type      : TPMS          id        : c1833c
Pressure  : 540 kPa      Temperature: 20.0 C       Flags     : 0             Integrity : CHECKSUM
[pulse_slicer_manchester_zerobit] EezTire E618 (TPMS10ATC)
codes     : {80}ffff9ec1833cd8460000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-09-30 12:54:32
model     : EezTire-E618 type      : TPMS          id        : c1833c
Pressure  : 540 kPa      Temperature: 20.0 C       Flags     : 0             Integrity : CHECKSUM
[pulse_slicer_manchester_zerobit] EezTire E618 (TPMS10ATC)
codes     : {80}ffff9ec1833cd8460000
```

Any suggestions on what other options to try.  Also willing to dig deeper and contribute to decoder development although I’m am new to rtl_sdr and rtl_433.  Thanks!

gdt, 2023-10-01:
Great progress!   Also, I suggest getting a better dongle, metal case, TCXO.  They are amazingly better.  Even if it is a checksum issue.   Also graph your frequencies and make sure they are in-band.  Also please post if you are using up-to-date git master.

I am thinking the original issue is solved, as you are decoding 5/8 and thinking there is a bug with your variant.  If so, maybe we should close this and have you open up a  "persistent checksum failures on some EEZ Tire E515 sensors" issue which is finer-grained.  Would appreciate your opinion on whether the original issue is fixed, and if there is information there that should be captured in the source code before we close.

flyboy013, 2023-10-01:
I'm good with closing this issue and opening another if that is preferred.   I believe the checksum calculation needs updating. The checksum is the sum of 6 bytes (id, pressure, temperature, and flags).  In my case the, sum is greater than 0xFF.  It seems that when the computed checksum is 0x1-- (ie greater than 0xFF), that the checksum is the sum of the 6 bytes & 0xFF plus 1.  The one gets added to the computed checksum.  When the computed checksum is 0x2--, the computed checksum needs to be or'ed with 0x80.   This based on the one sample that I have so far.   The car is back, so I need to capture the other 4 sensors and see if this logic holds true.

gdt, 2023-10-01:
It's clear that EEZ Tire is mostly working, so I'm closing this as done.   Please do open an issue about the checksum, and then eventually a PR when you know how to fix it.

closed 2023-10-01 by gdt

TheButterZone, 2026-09-06:
I've just gotten some beyond-belief EezTire-E618 decodes from somewhere nearby at almost the midnight hour...

time      : 2026-09-06 06:44:28
model     : EezTire-E618 type      : TPMS          id        : c3f000
Battery_OK: 0            Pressure  : 5 PSI         Temperature: 397.4 F      Flags     : 8000          Fast Leak : 0             Inflate   : 0             Integrity : CHECKSUM

time      : 2026-09-06 06:34:54
model     : EezTire-E618 type      : TPMS          id        : ffc290
Battery_OK: 1            Pressure  : 93 PSI        Temperature: -58.0 F      Flags     : 3fef          Fast Leak : 0             Inflate   : 1             Integrity : CHECKSUM

time      : 2026-09-06 06:19:35
model     : EezTire-E618 type      : TPMS          id        : e00000
Battery_OK: 1            Pressure  : 93 PSI        Temperature: -58.0 F      Flags     : 001f          Fast Leak : 0             Inflate   : 0             Integrity : CHECKSUM

time      : 2026-09-06 06:19:11
model     : EezTire-E618 type      : TPMS          id        : 0003fe
Battery_OK: 0            Pressure  : 0 PSI         Temperature: 401.0 F      Flags     : fe00          Fast Leak : 0             Inflate   : 1             Integrity : CHECKSUM

time      : 2026-09-06 06:15:03
model     : EezTire-E618 type      : TPMS          id        : 800003
Battery_OK: 0            Pressure  : 185 PSI       Temperature: 401.0 F      Flags     : ffff          Fast Leak : 0             Inflate   : 1             Integrity : CHECKSUM

Some sensors put in a scientific-grade freezer, others in a mega-autoclave?

zuckschwerdt, 2026-09-06:
Either a collision with some quite different sender or a very similar sender but with a different layout of data fields. Or, because the data looks like lots of clustered 1's and 0's, it could be noise that passes the checksum test.

If you can get a .cu8 then we can check if it's a legitimate sender. If you can only get the codes (`-R 241:v`) then we can at least check why the checksum is fooled.

TheButterZone, 2026-09-07:
For now...

```
./src/rtl_433 -f 315M -s 1024k -C customary -g 42.1 -R 241:v | awk '{print $0; printf "\a"}'
rtl_433 version 25.12-364-gf9041e81 branch pr-3680-monotonic at 202609051730 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000010, "Generic RTL2832U OEM"
[R82XX] PLL not locked!
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2026-09-07 03:11:01
model     : EezTire-E618 type      : TPMS          id        : 001000
Battery_OK: 0            Pressure  : 10 PSI        Temperature: 120.2 F      Flags     : f000          Fast Leak : 0             Inflate   : 1             Integrity : CHECKSUM
[pulse_slicer_manchester_zerobit] EezTire E618, Carchet TPMS, TST-507 TPMS
codes     : {68}ffffff0010001c63f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2026-09-07 03:13:00
model     : EezTire-E618 type      : TPMS          id        : e7781d
Battery_OK: 0            Pressure  : 92 PSI        Temperature: 172.4 F      Flags     : 8000          Fast Leak : 0             Inflate   : 0             Integrity : CHECKSUM
[pulse_slicer_manchester_zerobit] EezTire E618, Carchet TPMS, TST-507 TPMS
codes     : {135}fffffbe7781dff8080000000000000077e
```

Awaiting .cu8 from 
`./src/rtl_433 -f 315M -s 1024k -g 42.1 -R 241:v -S all | awk '{print $0; printf "\a"}''
`

TheButterZone, 2026-09-07:
[EezTire-E618 cu8 files trimmed.zip](https://github.com/user-attachments/files/31904635/EezTire-E618.cu8.files.trimmed.zip)

zuckschwerdt, 2026-09-07:
Careful with the codes, there is a `bitbuffer_invert()` in the decoder, so we need to reverse the invert.
Interestingly I don't think the Checksum matches in these examples? See [this BitBench](https://triq.org/bitbench/#c=%7B68%7D%20ffff%20ff0010001c63f&c=%7B135%7D%20ffff%20fbe7781dff8080000000000000077e&c=%7B92%7Df7%20ffff%20871f8c07d8d0003df&c=%7B176%7D%20ffff%20fffc000000008003c0f80007c0fc1e00000000ff&c=%7B85%7D80103%20ffff%208078700802108&c=%7B127%7D8000000100%20ffff%20ffff800000001fe1fe&c=%7B217%7D83%20ffff%207ff003dfffc3fff80f87ffffec3fe07ff80ffc03ffffffff8&f=CHK%3A8h%20ID%3A24h%20PRES%3A8h%20TEMP%3A8h%20FLAGS%3A16h%20TRAILER%3A200h&z=1&a=Preamble&m=ffff&i=true&cf=ADD&co=8&cl=56&cw=8).
I have to look closer why this happens. I.e. this should not work but does output: `rtl_433 -v -R 241:vv -y {68}000000ffefffe39c00`

Also `-S all` really should work. There should be outputs like `*** Saving signal to file g004_433.92M_250k.cu8 (287587 samples, 655360 bytes)`

TheButterZone, 2026-09-07:
I CMD-F'd the entire terminal tab, "Saving signal to file" never appeared.

zuckschwerdt, 2026-09-07:
I've fixed the length check in the decoder with cf82363
Most of the codes still pass through, the checksum accidentally matches.

zuckschwerdt, 2026-09-07:
With 090f13c there is a strict preamble check now, should get rid of most false positives. Otherwise we need to add e.g. maximum length checks.

TheButterZone, 2026-09-07:
> I've fixed the length check in the decoder with [cf82363](https://github.com/merbanan/rtl_433/commit/cf823633200ab838b20c610a894ce5380ead8f31) Most of the codes still pass through, the checksum accidentally matches.

I've been running ./src/rtl_433 -f 315M -s 1024k -g 42.1 -C customary | awk '{print $0; printf "\a"}' after making those +2 -1 edits, haven't seen any Eez decode since, just other models I'd previously seen - plus one decode of Acurite-Leak channel C all 0s & checksum, which I didn't see before.

I guess I'll test the https://github.com/merbanan/rtl_433/commit/090f13c0bcb07d278bd157fb0dc7878187578e82 +6 -1 edits after another 30 minutes.

Dangit, just as I say that...

> time      : 2026-09-07 09:42:57
> model     : EezTire-E618 type      : TPMS          id        : 7fbfe0
> Battery_OK: 0            Pressure  : 185 PSI       Temperature: 401.0 F      Flags     : e003          Fast Leak : 0             Inflate   : 1             Integrity : CHECKSUM

TheButterZone, 2026-09-07:
Just before I restarted to apply https://github.com/merbanan/rtl_433/commit/3792adf1075acb7c6447f7c27df9920fd4f2d41a

EezZombie is back again...

> time      : 2026-09-07 10:44:57
> model     : EezTire-E618 type      : TPMS          id        : 03fc00
> Battery_OK: 1            Pressure  : 0 PSI         Temperature: -58.0 F      Flags     : 0000          Fast Leak : 0             Inflate   : 0             Integrity : CHECKSUM

TheButterZone, 2026-09-08:
[TPMS E618 Anti-Theft sensors. Works with T515, E518 and the new E618 monitors](https://eezrvproducts.com/shop/ols/products/tpms-system-e518-anti-theft-replacement-sensor-1-ea) says

> Operation temperature range: -40℃ to +80℃
> Storage temperature range: -40℃ to +85℃
> Pressure range: 0 to 188psi (0 to 13bar)

-40 C is -40 F, 85 C is 185 F, so all but a couple of the decodes above should fail temp range sanity.