merbanan/rtl_433 #2388: Arexx IP-HA90

MacH-21, 2023-02-19:
Hi All,
I am attempting to decode a couple of sensors, 

https://www.conrad.com/p/arexx-ip-ha90-data-logger-sensor-unit-of-measurement-temperature-40-up-to-125-c-1313632
https://www.conrad.com/p/arexx-ip-th78ext-data-logger-sensor-unit-of-measurement-temperature-humidity-40-up-to-100-c-20-up-to-100-rh-1313634

I have caught and decoded most of the ip-ha90 temperature sensor this is a microchip rfpic12f679f 433.92M and a microchip mcp9808 temperature sensor.

rtl_433 -R 0  -X "n=name,m=FSK_MC_ZEROBIT,s=210,l=210,r=2500"

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-17 13:33:00
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 122          data      : 2aaaaaaad57c38ff7f6ffbbbadd27f8
codes     : {122}2aaaaaaad57c38ff7f6ffbbbadd27f8
*** Saving signal to file g002_433.92M_250k.cu8 (45807 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-17 13:33:52
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 121          data      : 2aaaaaaad57c38ff7f6ffb93c03fff8
codes     : {121}2aaaaaaad57c38ff7f6ffb93c03fff8
*** Saving signal to file g003_433.92M_250k.cu8 (45784 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-17 13:34:43
model     : name         count     : 1             num_rows  : 1             rows      :
len       : 121          data      : 2aaaaaaad57c38ff7f6ffb73c738ff8
codes     : {121}2aaaaaaad57c38ff7f6ffb73c738ff8
*** Saving signal to file g004_433.92M_250k.cu8 (45799 samples, 131072 bytes)

I have attached a zipfile with my info - captures and bitbench links at various stages, 

I believe the last two bytes to be CRC of sorts but unable to confirm any advice/help would be appreciated.
[ID_65934-Decode.txt](https://github.com/merbanan/rtl_433/files/10776986/ID_65934-Decode.txt)
[cu8-captures.zip](https://github.com/merbanan/rtl_433/files/10776989/cu8-captures.zip)

zuckschwerdt, 2023-02-19:
With the ID you already found polarity and alignment, that's the hard part ;) The second to last byte is a CRC-8 (poly 0x31, init 0) on reflected bytes (MSB-LSB swap). Not sure what the last byte is, seems XOR based.

MacH-21, 2023-02-19:
Wow, thanks for quick response I have been fixed on a 16 bit crc
07 8E 01 01 20 F6 48 2B D4 so.. 2B is the 8 bit crc  ... D4 is the invert of the crc :)
what bytes do the crc cover is it 0x8E thru to 0x48

zuckschwerdt, 2023-02-19:
the preceeding 7 bytes are covered by the crc
`crc8(&b[0], 8, 0x31, 0x00) == 0;` and `crc8(&b[0], 7, 0x31, 0x00) == b[7];`

zuckschwerdt, 2023-04-20:
There is now a decoder in #2487 -- let's discuss and finish this there.

closed 2023-04-20 by zuckschwerdt