merbanan/rtl_433 #2389: Add support for Celsia CZC1 Thermostat

libanp, 2023-02-19:
Hello,

Thanks for the help so far on the mailing list. Moving here as suggested.

The current flex decoder is:

`rtl_433 -X n=CZC1,m=OOK_PCM,s=1220,l=1220,r=4880,preamble=cccccccccccccccccccc`

I've created an annotated bitbench from the unique samples I've managed to get from the 3 thermostats I have running.

There are 2 packet types, one for pairing and one for commands. Pairing packets are 5 bytes, command packets are 6 bytes.

Pairing packet:
1. 0
2. ID
3. ID
4. Type
5. CRC

Command packet:
1. 0
2. ID
3. ID
4. Type
6. Heat level (0-255 LSB)
7. CRC

As shown on the mailing list, pre-padding the data reveals a known CRC (still valid with the extra samples):

```
./reveng -w 8 -s 000000002750f00017 000000002750f0ecab 000000002750f0d6b5 000000002750f08598 000000002750f0ab01 000000002750f0ffbb 000000002750f0c050 00000000ec46f00087 00000000ec46f08c80 00000000f0d3f0c0f4 00000000f0d3f000b3 00000000f0d3f0ec0f 00000000f0d3f0d611 00000000f0d3f0853c 00000000f0d3f0aba5 00000000f0d3f0ff1f 

width=8  poly=0x31  init=0xff  refin=false  refout=false  xorout=0x00  check=0xf7  residue=0x00  name="CRC-8/NRSC-5"
```

There are some samples that decode wrong, is this an issue with the timing specified in the flex decoder? See samples g2150, g363

[bitbench](https://triq.net/bitbench#c=%5BCMD%20is%20the%20duty%20cycle%20of%20the%20radiator%20out%20of%207%5D&c=&c=002750f00017%20%5B0%20(off)%5D&c=002750f0ecab%20%5B1%20or%202%5D&c=002750f0d6b5%20%5B3%5D&c=002750f08598%20%5B4%20or%205%5D&c=002750f0ab01%20%5B6%5D&c=002750f0ffbb%20%5B7%20(max)%5D&c=002750f0c050%20%5Bunknown%5D&c=&c=&c=00ec46f00087%20%5Boff%5D&c=00ec46f08c80%20%5Bunknown%5D&c=&c=00f0d3f0c0%20%20%20%5Bmissing%20CRC%5D&c=00f0d3f0c0f4%20%5Bunknown%5D&c=00f0d3f000b3%20%5B0%20(off)%5D&c=00f0d3f0ec0f%20%5B1%20or%202%5D&c=00f0d3f0d611%20%5B3%5D&c=00f0d3f0853c%20%5B4%20or%205%5D&c=00f0d3f0aba5%20%5B6%5D&c=00f0d3f0ff1f%20%5B7%20(max)%5D&c=00f0d30073%20%20%20%5Bpairing%20packet%2C%20always%201%20byte%20shorter%5D&c=&c=&c=001d88de0010%20%5Bshould%20be%20ec46f00087%5D&c=003c34fc002%20%20%5Bshould%20be%20f0d3f000b%5D&c=003c34fc002c%20%5Bshould%20be%20f0d3f000b3%5D&f=8x%20ID%3A%20%5E8h%5E8h%20%7C%20TYP%3A%20%5E8h%20%7C%20CMD%3A%20%5E8d%20%7C%20CRC%3A%20%5E8h&cw=4)
[czc1_samples.zip](https://github.com/merbanan/rtl_433/files/10777598/czc1_samples.zip)

zuckschwerdt, 2023-02-19:
Well done, especially figuring out the reflected bytes.

Usually you only want to match the end of a preamble and then some some sync.
I.e. a preamble is a regular pattern and breaking of the pattern is the sync, payload starts after that.
So I guess the full preamble+sync might be `cccccccccccccccccccc55555555` and matching just e.g. `cccccccc55555555` should do.

Not sure there is a need for padding is, a CRC only changes the init under 0-bit prepends ;) The init is 0xd7 if you start at the ID.

klohner, 2023-02-20:
Perhaps there's no need to zero pre-pad the data to get a working CRC if one accepts a seemingly awkward CRC init byte value.

This works to find a CRC for these 6-byte messages:

```
$ reveng -w 8 -s 00f0d3f0ec0f 00f0d3f0d611 00f0d3f0853c
width=8  poly=0x31  init=0x4b  refin=false  refout=false  xorout=0x00  check=0x9b  residue=0x00  name=(none)
```

as do these:

```
$ reveng -w 8 -s d3f0ec0f d3f0d611 d3f0853c
width=8  poly=0x31  init=0x11  refin=false  refout=false  xorout=0x00  check=0xad  residue=0x00  name=(none)

$ reveng -w 8 -s 0000f0d3f0ec0f 0000f0d3f0d611 0000f0d3f0853c
width=8  poly=0x31  init=0x81  refin=false  refout=false  xorout=0x00  check=0x32  residue=0x00  name=(none)

$ reveng -w 8 -s 000000f0d3f0ec0f 000000f0d3f0d611 000000f0d3f0853c
width=8  poly=0x31  init=0xac  refin=false  refout=false  xorout=0x00  check=0x29  residue=0x00  name=(none)
```

However, something interesting happens when zero pre-padding these 6-byte message to 9-byte messages:

```
$ reveng -w 8 -s 00000000f0d3f0ec0f 00000000f0d3f0d611 00000000f0d3f0853c
width=8  poly=0x31  init=0xff  refin=false  refout=false  xorout=0x00  check=0xf7  residue=0x00  name="[CRC-8/NRSC-5]"
```

[CRC-8/NRSC-5](https://reveng.sourceforge.io/crc-catalogue/1-15.htm#crc.cat.crc-8-nrsc-5) is a named CRC algorithm with a more normalized init byte value (0xFF), and I thought this might be significant and not a coincidence.

But perhaps not.  As @libanp stated, messages might be either 5 or 6 bytes.  My first thought with this info was that received messages might be internally zero pre-padded to 9 bytes before checking the CRC.  However, this turned out to be an incorrect assumption.  Shorter 5-byte messages (such as "00f0d30073") must remain a byte shorter than the 6-byte messages for the same CRC algorithm to work.

```
$ reveng -w 8 -s 00000000f0d3f0ec0f 00000000f0d3f0d611 00000000f0d3f0853c 0000000000f0d30073
reveng: no models found
[when normalizing all messages to zero pre-padded, 9-byte messages]

$ reveng -w 8 -s 00000000f0d3f0ec0f 00000000f0d3f0d611 00000000f0d3f0853c 00000000f0d30073
width=8  poly=0x31  init=0xff  refin=false  refout=false  xorout=0x00  check=0xf7  residue=0x00  name="CRC-8/NRSC-5"
[when simply pre-padding received messages with 0x000000, regardless of initial length]
```

Also, to @libanp's point that g2150 and g363 decode wrong, the issue in both these files (and also in g153) is that there is a signal from another non-thermostat device overlapping part of the thermostat's signal.  (I was able to manually pull the signal from these files, though)

I went through the samples in the .zip file and tried to pair them up to the bitbench at the beginning of this thread.  I removed the ones marked "should be" since those were probably the overlapped captures.  The only one I cannot account for with a .cu8 file is the "pairing packet".  It would be very useful to see that one!  @libanp - do you have a .cu8 file with a pairing packet example signal?

[revised bitbench](https://triq.net/bitbench#c=%5BCMD%20is%20the%20duty%20cycle%20of%20the%20radiator%20out%20of%207%5D&c=&c=002750f00017%20%5B0%20(off)%20(g055)%5D&c=002750f0ecab%20%5B1%20or%202%20(g1541)%5D&c=002750f0d6b5%20%5B3%20(g1512)%5D&c=002750f08598%20%5B4%20or%205%20(g003)%5D&c=002750f0ab01%20%5B6%20(g012)%5D&c=002750f0ffbb%20%5B7%20(max)%20(g037)%5D&c=002750f0c050%20%5Bunknown%20(g1732)%5D&c=&c=00f0d3f000b3%20%5B0%20(off)%20(g058%2C%20g153%2C%20and%20g363)%5D&c=00f0d3f0ec0f%20%5B1%20or%202%20%20(g001)%5D&c=00f0d3f0d611%20%5B3%20(g004)%5D&c=00f0d3f0853c%20%5B4%20or%205%20(g024)%5D&c=00f0d3f0aba5%20%5B6%20(g047)%5D&c=00f0d3f0ff1f%20%5B7%20(max)%20(g676)%5D&c=00f0d3f0c0f4%20%5Bunknown%20(g1102)%5D&c=&c=00ec46f00087%20%5Boff%20(g002%20and%20g2150)%5D&c=00ec46f08c80%20%5Bunknown%20(g592)%5D&c=&c=00f0d30073%20%20%20%5Bpairing%20packet%2C%20always%201%20byte%20shorter%20(missing%20.cu8%20file%3F)%5D&f=8x%20ID%3A%20%5E8h%5E8h%20%7C%20TYP%3A%20%5E8h%20%7C%20CMD%3A%20%5E8d%20%7C%20CRC%3A%20%5E8h&cw=4)

libanp, 2023-02-20:
Oops! Not sure how I managed to leave the pairing sample out. I've attached an updated sample archive that includes pairing signals from each thermostat (so 2 new pairing signals as well as the missing one) and added the new files to the bitbench notes 

[revised bitbench](https://triq.net/bitbench#c=%5BCMD%20is%20the%20duty%20cycle%20of%20the%20radiator%20out%20of%207%5D&c=&c=002750f00017%20%5B0%20(off)%20(g055)%5D&c=002750f0ecab%20%5B1%20or%202%20(g1541)%5D&c=002750f0d6b5%20%5B3%20(g1512)%5D&c=002750f08598%20%5B4%20or%205%20(g003)%5D&c=002750f0ab01%20%5B6%20(g012)%5D&c=002750f0ffbb%20%5B7%20(max)%20(g037)%5D&c=002750f0c050%20%5Bunknown%20(g1732)%5D&c=&c=00f0d3f000b3%20%5B0%20(off)%20(g058%2C%20g153%2C%20and%20g363)%5D&c=00f0d3f0ec0f%20%5B1%20or%202%20%20(g001)%5D&c=00f0d3f0d611%20%5B3%20(g004)%5D&c=00f0d3f0853c%20%5B4%20or%205%20(g024)%5D&c=00f0d3f0aba5%20%5B6%20(g047)%5D&c=00f0d3f0ff1f%20%5B7%20(max)%20(g676)%5D&c=00f0d3f0c0f4%20%5Bunknown%20(g1102)%5D&c=&c=00ec46f00087%20%5Boff%20(g002%20and%20g2150)%5D&c=00ec46f08c80%20%5Bunknown%20(g592)%5D&c=&c=00f0d30073%20%20%20%5Bpairing%20packet%2C%20always%201%20byte%20shorter%20(pair_g002)%5D&c=002750001d%20%20%20%5Bpairing%20packet%20(pair_g004)%5D&c=00ec460096%20%20%20%5Bpairing%20packet%20(pair_g010)%5D&f=8x%20ID%3A%20%5E8h%5E8h%20%7C%20TYP%3A%20%5E8h%20%7C%20CMD%3A%20%5E8d%20%7C%20CRC%3A%20%5E8h&cw=4)

[czc1_samples_withpairing.zip](https://github.com/merbanan/rtl_433/files/10779008/czc1_samples_withpairing.zip)

libanp, 2023-02-21:
I've got a working decoder written which I'll open a PR for. Thanks for the help with the CRC @zuckschwerdt and @klohner, its definitely the part of this that I understand the least. What I went with in the end was CRC8 with  `poly=0x31` and `init=0xd7`. `reveng` gave me two options so I went with the one with no remainder:

```
./reveng -w 8 -s f0d3f0d611 f0d3f0853c f0d3f0aba5 f0d3f0ff1f f0d3f0ec0f f0d3f000b3 ec46f08c80 f0d30073 2750001d ec460096
width=8  poly=0x31  init=0x38  refin=false  refout=false  xorout=0xef  check=0xef  residue=0xef  name=(none)
width=8  poly=0x31  init=0xd7  refin=false  refout=false  xorout=0x00  check=0xef  residue=0x00  name=(none)
```

 (the last three provided values are the shorter pairing packets)

gdt, 2023-09-26:
Looks like the PR was merged.   Please open a new issue if there is something residual.

closed 2023-09-26 by gdt