merbanan/rtl_433 #2394: Request for Baldr Rain Gauge

garudaonekh, 2023-02-23:
Hi,
Please add protocol for Baldr Rain Gauge.
https://www.baldrelectronic.com/products/digital-wireless-indoor-outdoor-thermometer-hygrometer-with-rain-gauge
[BaldrRainGauge_CU8_files.zip](https://github.com/merbanan/rtl_433/files/10810873/BaldrRainGauge_CU8_files.zip)

It's also a Rainpoint product. I think it's similar to RainPoint Soil moisture sensor.

I have collected some cu8 files in my attachment.
Regards;

zuckschwerdt, 2023-02-23:
All files except g003_433.92M_250k.cu8 are empty. Drop the files https://triq.org/pdv/ on to check.

It's PPM 1000/2000. This will get the data:
`rtl_433 -X 'n=baldr,m=OOK_PPM,s=1000,l=2000,g=3000,r=5000' g003_433.92M_250k.cu8`
as `{36}75b000000`

Run this live as `rtl_433 -X 'n=baldr,m=OOK_PPM,s=1000,l=2000,g=3000,r=5000,bits>=36'`
and collect the codes. Write the readings you have (from the head unit) with each code, like this:

`{36}75b000000 [10 C 60 % 3mm]`

Post a list of annotated codes here.

garudaonekh, 2023-02-23:
Here it's
```
****************************  0mm ***************************
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-23 21:44:39
model     : baldr        count     : 13            num_rows  : 13            rows      : 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000, 
len       : 36           data      : 75b000000
codes     : {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000

****************************   5.2mm  ****************************
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-23 21:50:27
model     : baldr        count     : 13            num_rows  : 13            rows      : 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf, 
len       : 36           data      : 75b8000cf
codes     : {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf, {36}75b8000cf

---
9.6mm
time      : 2023-02-23 21:56:15
model     : baldr        count     : 13            num_rows  : 13            rows      : 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a, 
len       : 36           data      : 75b80017a
codes     : {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a, {36}75b80017a

****************************   13.9mm   ****************************
time      : 2023-02-23 22:02:03
model     : baldr        count     : 13            num_rows  : 13            rows      : 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224, 
len       : 36           data      : 75b800224
codes     : {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224, {36}75b800224

****************************    17.1mm  ****************************

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-23 22:07:51
model     : baldr        count     : 13            num_rows  : 13            rows      : 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3, 
len       : 36           data      : 75b8002a3
codes     : {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3, {36}75b8002a3

```

zuckschwerdt, 2023-02-23:
A shorter list is preferred, edited above. What was temperature and humidity at those times?

garudaonekh, 2023-02-23:
It only report rain in mm. There's a separate temperature sensor--captured by your Nexus-TH.

zuckschwerdt, 2023-02-23:
Is there a pairing method or button?

Can you remove the batteries, wait a moment and insert them back? Does it still start with `75b` or does that change? We must try to find the sender ID.

Also do you have flat batteries or a variable voltage power supply to simulate near dead batteries? Then we can find the "battery_low" flag.

garudaonekh, 2023-02-23:
> Is there a pairing method or button?
>
> Can you remove the batteries, wait a moment and insert them back? Does it still start with `75b` or does that change? We must try to find the sender ID.
>
> Also [...]

I already remove battery several times, it's still 75b.
Synchornization and registration is done at the LCD monitor side.
I have voltage regulator so I can simulate low battery. But I don't see any mention of battery status in its manual or on the LCD panel

zuckschwerdt, 2023-02-23:
One interesting flag is the 0/8, e.g. the first code had `75b 0` but other codes then `75b 8` -- any idea what that means or when/why this appears?

garudaonekh, 2023-02-23:
I remove battery and I run again. There's no 8. only 75b0000f9. I am trying to find the pattern

garudaonekh, 2023-02-23:
```
{36}75b0001f2 [11.6mm]
{36}75b000245 [13.7mm]
```

zuckschwerdt, 2023-02-23:
Maybe that is a "did it rain since last transmission" flag? Can you capture the same "mm" multiple times? Does the 8 become 0 then?

garudaonekh, 2023-02-24:
Here's an update. I notice b8 still there for several minutes and when I make another click. it goes to b0
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-24 11:01:01
codes     : {48}00075b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000, {36}75b000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-24 11:03:56
codes     : {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c, {36}75b00007c
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-24 11:06:50
codes     : {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-24 11:24:12
codes     : {65}00000003adc000a98, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153, {36}75b800153

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-02-24 11:27:08
codes     : {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6, {36}75b0001a6
```

garudaonekh, 2023-02-24:
> Maybe that is a "did it rain since last transmission" flag? Can you capture the same "mm" multiple times? Does the 8 become 0 then?

No, it will not become 0.

garudaonekh, 2023-02-24:
I think it the same as your Nexus Temp/humidity protocol but with different payload.

zuckschwerdt, 2023-02-24:
Yes. Then `75` must be the ID and `b0`/`b8` some flags?

The strange thing is the rain counter:
Its either 20-bit rain in 1/40 mm or 16-bit rain in 1/2.5 mm steps (scale tips) and some 4-bit checksum.
I don't think a bucket tip is 1/40 =0.025 mm rain. A cup of 1/2.5 =0.4 mm sounds more reasonable.

- Can you record codes with just one single bucket tip?
- Can you measure the cup (either size or volume) ?

zuckschwerdt, 2023-02-24:
That is very strange. I would have expected:
`75b000027` -> `75b000028` -> `75b000029` ...
or `75b000027` -> `75b00003?` -> `75b00004?` ...

Why would it be 27 -> 50? Can you test again and for more tips?
With "tip" I mean the bucket in the rain sensor moving once. This is the minimum the sensor can "measure" and should be "1" unit.

garudaonekh, 2023-02-24:
Yes, it's the smallest unit.

this is 1 tip: [75b000027 0.9mm]
this is 2 tip: [75b000050 2.0mm]
3 tip: 75b00007a
4 tip: 75b0000a3
5 tip: 75b0000cd
6 tip: 75b0000f7

garudaonekh, 2023-02-28:
I think I get a pattern which may be correct, but it increment double every 3-4 tips maybe it's not 1.25mm per tip:

{36}75b000000 [2023-02-23 21:44:39 0tip]
{36}75b000025 [2023-02-23 21:50:27 1tip]
{36}75b000050 [2023-02-23 21:56:15 2tip]
{36}75b00007a [2023-02-23 22:02:03 3tip]
{36}75b0000a6 [2023-02-23 22:07:51 4tip]
{36}75b0000D1 [2023-02-23 22:07:51 5tip]
{36}75b0000fb [2023-02-23 22:07:51 6tip]
{36}75b000125 [2023-02-23 22:07:51 7tip]
{36}75b00014e [2023-02-23 22:07:51 8tip]
{36}75b000178 [2023-02-23 22:07:51 9tip]
{36}75b00019f [2023-02-23 22:07:51 10tip]
{36}75b0001c8 [2023-02-23 22:07:51 11tip]
{36}75B0001f2 [2023-02-23 22:07:51 12tip]
{36}75B00021b [2023-02-23 22:07:51 13tip]
{36}75B000245 [2023-02-23 22:07:51 14tip]

[image]

garudaonekh, 2023-02-28:
I think this one is better suited
[image]

zuckschwerdt, 2023-02-28:
Using `-2` you are cutting of the last 2 bits, so the mm-factor ist now `*10` and the 2 bits could be a parity check. I'll test that.

zuckschwerdt, 2023-02-28:
Ohh, I see now. The `20d` value is correct. But it's not i `mm` but in `thou` -- that's `1/1000 inch`:
```
75b0 0000000 : 0mm = 0 thou
75b8 0000207 : 5.2mm = 205 thou
75b8 0000378 : 9.6mm = 378 thou
75b8 0000548 : 13.9mm = 547 thou
75b8 0000675 : 17.1mm = 673 thou
```

garudaonekh, 2023-02-28:
Great! anything else you need from me?

zuckschwerdt, 2023-02-28:
We can implement this now. But it will likely be default-disabled as we can't check anything about the packet (would be too many false positives).

garudaonekh, 2023-02-28:
What do you mean by false positives?

zuckschwerdt, 2023-02-28:
Many signals can "seen" as this protocol. This decoder will match and decode too many other signals.

garudaonekh, 2023-02-28:
I see. Thanks. Looking forward for testing it in RTL_SDR

closed 2023-03-03 by zuckschwerdt

garudaonekh, 2023-03-04:
Hi,

`rtl_433 -R 242`

Response:

```
[Protocols] Registered 1 out of 242 device decoding protocols [ 242 ]
Found Rafael Micro R820T tuner
[SDR] Using device 0: Generic, RTL2832U, SN: 77771111153705700, "Generic RTL2832U"
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
vdata_make() bad data type (197414298)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-03-04 14:22:28
vdata_make() bad data type (197414298)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-03-04 14:22:56

```

zuckschwerdt, 2023-03-04:
Oops, thanks, fixed with 1647d9af

garudaonekh, 2023-03-05:
Look correct. The result is look correct. I am waiting for my second rain gauge to verify the id is correctly identified.

The first one: **188.171 in** is actually the Temperature Sensor. It is also Baldr. 

`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-03-05 15:46:07
model     : Baldr-Rain   id        : 1c8
Flags     : 1            Rain      : 188.171 in
[pulse_slicer_ppm] Baldr / RainPoint rain gauge.
codes     : {0}, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {36}1c812df0b, {35}1c812df0a, {756}000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[pulse_slicer_ppm] Baldr / RainPoint rain gauge.
codes     : {323}000000000000000000000000000000000000000000000000000000000000000000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-03-05 15:47:12
model     : Baldr-Rain   id        : 75b
Flags     : 0            Rain      : 0.627 in
[pulse_slicer_ppm] Baldr / RainPoint rain gauge.
codes     : {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273, {36}75b000273
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2023-03-05 15:48:01
model     : Baldr-Rain   id        : 1c8
Flags     : 1            Rain      : 196.363 in
[pulse_slicer_ppm] Baldr / RainPoint rain gauge.
codes     : {0}, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {36}1c812ff0b, {35}1c812ff0a, {756}000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
[pulse_slicer_ppm] Baldr / RainPoint rain gauge.
codes     : {323}000000000000000000000000000000000000000000000000000000000000000000000000000000000`

zuckschwerdt, 2023-03-05:
Thanks for the feedback.

> The first one: **188.171 in** is actually the Temperature Sensor. It is also Baldr. So it's that false positive you mentioned.

Yes, I have now default-disabled the Baldr-Rain protocol. You now need `-R 242` or a conf file

garudaonekh, 2023-03-07:
I just receive another rain gauge, its id is 12b. Thus, I think the id is correct

zuckschwerdt, 2023-03-07:
If we have `75b` and `12b` then the `b` is likely not part of the ID?

garudaonekh, 2023-03-08:
its LCD receiver has rain gauge and temp sensor. Thus b might be used to identified it is a rain gauge. I am buying a few more. I will give more update

commit d3fb7169 (2023-03-03): Add support for Baldr/Rainpoint rain gauge (closes #2394)