merbanan/rtl_433 #2396: MQTT doesn't show signals that default mode does

coogle, 2023-02-24:
I'm encountering a problem where I'm not seeing the same data from the built-in MQTT client as I am from the "default" configuration... In all cases I'm running version 22.11.

When running a stock configuration, I get devices I expect to see:

```
rtl_433 version 22.11 (2022-11-19) inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
[Protocols] Registered 207 out of 240 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-240 ]
[SDR] Found 1 device(s)
[SDR] trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[SDR] Using device 0: Generic RTL2832U OEM
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : 2023-02-24 04:51:15
model     : Ambientweather-F007TH
House Code: 244
Channel   : 3
Battery   : 1
Temperature: -13.4 F
Humidity  : 42 %
Integrity : CRC
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
bitbuffer_add_bit: Warning: row count limit (50 rows) reached
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

time      : 2023-02-24 04:51:46
model     : Acurite-986
id        : 16488
channel   : 2F
Battery   : 1
temperature: 48.000000 F
status    : 0
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
```

However, when I run with `-F mqtt://localhost:1883` I see the "Ambientweather" but I do *not* see the Acurite-986 ever get transmitted to my MQTT server. 

I also see my Acurite-986 when I run `-F json` so for the moment I can workaround by piping to `mosquitto_pub` but I'd like to know what I'm doing wrong to prevent all of the captures from showing up in my MQTT feed here..

zuckschwerdt, 2023-02-24:
There are no filters. Perhaps you missed the data listening on a different topic. Use e.g. `mosquitto_sub -t '#' -v` to monitor.

Closing as this is not something to change or fix in rtl_433, feel free to discuss further though.

closed 2023-02-24 by zuckschwerdt

coogle, 2023-02-24:
I didn’t miss the data listening in on a different topic. I’m very familiar with MQTT and wildcard topics. The data just didn’t show up! I think closing this is in error, I honestly think there is a bug here.

coogle, 2023-02-24:
What would be the best way for me to demo what im seeing? I’ve seen *all* the other devices around me except this one (even when I force transmission)

zuckschwerdt, 2023-02-24:
When I try one of our samples with the change from #2372 I get e.g.
```
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/events', ... (222 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/time', ... (26 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/id', ... (5 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/channel', ... (2 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/battery_ok', ... (1 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/temperature_F', ... (4 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/status', ... (1 bytes))
Received PUBLISH from rtl_433-b6bfffff (d0, q0, r0, m0, 'rtl_433/Host/devices/Acurite-986/1R/17021/mic', ... (3 bytes))
```

There really is no code path to filter out some devices. You have an exlusion hiding somewhere in your MQTT listener. Is it maybe the channel as that's no a number here?

coogle, 2023-02-24:
Hmm. The only thing I can think is there is a bug in my MQTT GUI client. I will test again now with just the CLI and report back.

coogle, 2023-02-24:
Okay it looks like there is a bug, just not in this code base  (sorry about that!) My MQTT GUI client wasn't actually reporting all of the messages it is getting. When I used `mosquitto_sub` with the exact same subscription I saw the messages come across. Never seen this issue before so it never occurred to me to even try a different MQTT client.

Sorry again for the noise!