merbanan/rtl_433 #2397: Auto discovery does not properly decode devices

invisible999, 2023-02-24:
I am using rtl_433 and rtl_433_auto_discovery with Home Assistant. 

I see that MQTT explorer sees devices and their properties one this way:
[image]

However after auto discovery decodes them, it ends up like this

```
automation_type: trigger
type: button_short_release
subtype: button_1
topic: rtl_433/9b13b3f4-rtl433/devices/Markisol/0/1/channel
platform: mqtt
device:
  identifiers:
    - Markisol-0-1
  name: Markisol-0-1
  model: Markisol
  manufacturer: rtl_433

automation_type: trigger
type: button_short_release
subtype: button_1
topic: rtl_433/9b13b3f4-rtl433/devices/Regency-Remote/7/channel
platform: mqtt
device:
  identifiers:
    - Regency-Remote-7
  name: Regency-Remote-7
  model: Regency-Remote
  manufacturer: rtl_433
  ```
And honda car remote does not come with any properties after decoding.
  
  ```
 "device": {
      "id": "fee23ea4e674ee8a119d67aebbbd294a",
      "name": "Honda-CarRemote-65535",
      "name_by_user": null,
      "disabled": false,
      "disabled_by": null,
      "entities": [
        {
          "device_class": null,
          "disabled_by": "integration",
          "disabled": true,
          "entity_category": "diagnostic",
          "entity_id": "sensor.honda_carremote_65535_utc",
          "icon": null,
          "original_device_class": "timestamp",
          "original_icon": "mdi:clock-in",
          "state": null,
          "unit_of_measurement": null
        }
      ]
    },
    "mqtt_debug_info": {
      "entities": [],
      "triggers": []
    }
```

I have seen a number of decoding errors with other devices as well, some temp sensors showing outside temperature as 670F. Well, guess we are living on the surface of Venus...

zuckschwerdt, 2023-02-24:
Do you mean `rtl_433_mqtt_hass.py` otherwise not sure what `rtl_433_auto_discovery` is?
@gdt @rct is there a case where the raw string `/channel` could end up in the topic?

invisible999, 2023-02-24:
> Do you mean `rtl_433_mqtt_hass.py` otherwise not sure what `rtl_433_auto_discovery` is? @gdt @rct is there a case where the raw string `/channel` could end up in the topic?

I am using [this plugin](https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433_mqtt_autodiscovery)

zuckschwerdt, 2023-02-24:
Ok, thanks. The author @deviantintegral is also a rtl_433 contributor, he might have an idea about this.

invisible999, 2023-02-24:
So my question is two fold:

First is about auto-discovery because it is limited only to the keywords it works with. So even if the sensor data is properly decoded, it is not properly imported into HA.

Example from the screenshot: Regency-Remote identified as a device under protocol 211. The properties are channel and fan speed. 

In reality, it might be a completely different device under a different protocol and as a result - different properties.

- How can I force it to decode it as a different protocol number?
- If I am I am satisfied with the decoded information, how can I import it into HA as a sensor, if auto_discovery script is not helping me?

rct, 2023-02-24:
First, the Home Assistant Add On you are referring to wraps two programs, rtl_433, and the contributed python auto discovery script.  rtl_433 is the main product of this repo, which deals with the RTL-SDR and implements decoders for many devices. 

In that add on rtl_433 is configured to output to MQTT. 

In rtl_433, a number of people have reverse engineered many devices and contributed decoders. Support to additional devices needs to be implemented by adding a new decoder (code in C), or where possible using a flex decoder that can break out bytes using only configuration.

> Example from the screenshot: Regency-Remote identified as a device under protocol 211. The properties are channel and fan speed.
> 
> In reality, it might be a completely different device under a different protocol and as a result - different properties.

So if I'm following you correctly, you've got a device which isn't being decoded correctly by rtl_433. Some devices may be incorrectly decoded as a different device. This tend to happen with wireless remotes as they often don't have a lot of information in the message that can be used to differentiate. 

> How can I force it to decode it as a different protocol number?

If you there is a protocol number (a decoder) which decodes the device correctly, you can make sure to enable it -R NN, and disable the protocol that decodes it incorrectly if you don't need it.  The recommended practice is to run rtl_433 with only the protocols (device decoders) enabled that you actually need. This will cut down on false positives where messages are interpreted incorrectly.

> If I am I am satisfied with the decoded information, how can I import it into HA as a sensor, if auto_discovery script is not helping me?

The Home Assistant auto discovery script is a contributed piece of software that listens to the messages rtl_433 publishes to MQTT and tries to publish MQTT messages that will auto configure entities in Home Assistant.  

You don't need to use the auto discovery script. You can manually configure MQTT entities in Home Assistant.  Or you can write and publish your own auto discovery JSON config.  I tend to use MQTT explorer to edit and customize my auto discovery configurations.

> [...]  is about auto-discovery because it is limited only to the keywords it works with. So even if the sensor data is properly decoded, it is not properly imported into HA.

Yes, the current auto discovery script is very limited because it works on the individual fields (what I believe you are referring to by keywords).  It doesn't take the device type, what rtl_433 refers to as `model` into account.  It tends to work fairly well for all the devices that report generic fields like temperature, humidity, battery, etc.

gdt, 2023-02-24:
> I am using [this plugin](https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433_mqtt_autodiscovery)

If you are having a problem with that plugin I suggest that you inquire of that project instead of here.

deviantintegral, 2023-02-25:
We actually pull the script directly from this repo. Occasionally we'll apply patches to it, but only if there is an open PR here.

https://github.com/pbkhrv/rtl_433-hass-addons/blob/main/rtl_433_mqtt_autodiscovery/Dockerfile#L4-L6

I think @rct 's comment pretty much covers what I would suggest for next steps.

gdt, 2023-09-26:
This feels like a question/support request.   I am not seeing a specific sensor that's wrong, or testing with recent master.   Therefore I think we should close this and the original poster could open a discussion or ask for help.   There are a lot of processing steps in getting to HA, and debugging needs to be done on the one that is troubled.  A broad comment that the sum of multiple steps doesn't do what someone wants is not really going to lead to a fix.

closed 2023-09-26 by invisible999