merbanan/rtl_433 #2398: Bresser 5in1 wrong 'rain' value

PopcornAWH, 2023-02-25:
Hi, 
Been using rtl_433 for some years to capture and decode my Bresser 5in)1.

On a newly installed Pi4B (raspiOS bullseye) the installed rtl_433 version is:
rtl_433 version nightly-2-g371ccfd5 branch master at 202302201511 inputs file rtl_tcp RTL-SDR

This version shows the rain value wrong... 
time      : 2023-02-25 15:29:10
model     : Bresser/mczakk Weather Center 5-in-1   count     : 1             num_rows  : 1             rows      :
len       : 285          data      : 80555555555516ea720ebfebffed77cc7fd43bf47f8df1401400128833802bc40b800000
codes     : {285}80555555555516ea720ebfebffed77cc7fd43bf47f8df1401400128833802bc40b800000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-02-25 15:29:10
model     : Bresser-5in1 id        : 226
Battery   : 1            Temperature: 6.7 C        Humidity  : 57            Wind Gust : 4.0 m/s       Wind Speed: 2.5 m/s       Direction : 0.0           Rain      : 178.8 mm
Integrity : CHECKSUM

Bresser 5in1 has a cycling rain value 0 to 100 (or maybe 99), in any case, it should never be greater than 100.

cu8 and ook file:
https://www.dropbox.com/sh/slicjbnjygf2x4e/AADwc_tcWFnWucCsSHI2FYIqa?dl=0

Let me know if you need anything more from me, happy to provide it...

Paul

matthias-bs, 2023-02-26:
Please see here for expanation:https://github.com/merbanan/rtl_433/pull/2312

https://www.dropbox.com/sh/slicjbnjygf2x4e/AADwc_tcWFnWucCsSHI2FYIqa?dl=0

zuckschwerdt, 2023-02-26:
Thanks @matthias-bs
Closing as this seems to be a misunderstanding only, value goes to `999.9 mm` now, but please let us know if you see anything suspicious or wrong with the values.

closed 2023-02-26 by zuckschwerdt

PopcornAWH, 2023-02-26:
I'll adapt my backend, will report if something looks wrong...