merbanan/rtl_433 #2402: Add support for french GazPar smart gaz meter

fthiery, 2023-03-01:
Hi there, i live in France where the national gaz provider deploys a smart meter called Gazpar.

[image]

This [reportedly](https://fr.wikipedia.org/wiki/Gazpar) use the Wize protocol that seems to be based on EN13757-4 (Wireless M-Bus) and sends current metering twice a day over the 169 MHz frequency, over a central frequency that depends on the serial number visible on the unit (among 5 possible central frequencies: 169,40625 MHz, 169,41875 MHz, 169,44375 MHz, 169,45625 MHz and 169,46875 MHz).

My goal is to measure my gaz consumption into Home Assistant over mqtt; i would need to filter only my gaz meter obviously.

Based on the information above, i launched rtl_433 with `/usr/bin/rtl_433 -f 169468750 -S unknown` and looked at the catched samples. I think a few may be of interest, especially this one:

[image]

[g022_169.469M_250k.zip](https://github.com/merbanan/rtl_433/files/10859643/g022_169.469M_250k.zip)

I tried the existing m-bus drivers 104-107 but nothing came out.

I think the protocol spec is this: https://www.wize-alliance.com/content/download/620/2430?version=6

Another possibly interesting publication is this: https://innovation.grdf.fr/statics/filemanager/AAP%20169%20MHz/Informations%20sur%20l%E2%80%99appel%20%C3%A0%20projets%20Hybridation.pdf

Here is the output of `rtl_433 -r g022_169.469M_250k.cu8 -a 4 -A` :

```
rtl_433 version -128-NOTFOUND branch master at 202211192216 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/fthiery/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Test mode active. Reading samples from file: g022_169.469M_250k.cu8
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
Detected OOK package	@0.209664s
Analyzing pulses...
Total count:    1,  width: 149.08 ms		(37271 S)
Pulse width distribution:
 [ 0] count:    1,  width: 149084 us [149084;149084]	(37271 S)
Gap width distribution:
Pulse period distribution:
Pulse timing distribution:
 [ 0] count:    1,  width: 149084 us [149084;149084]	(37271 S)
 [ 1] count:    1,  width: 100004 us [100004;100004]	(25001 S)
Level estimates [high, low]:  12516,    437
RSSI: -1.2 dB SNR: 14.6 dB Noise: -15.7 dB
Frequency offsets [F1, F2]:    -859,      0	(-3.3 kHz, +0.0 kHz)
Guessing modulation: Single pulse detected. Probably Frequency Shift Keying or just noise...
view at https://triq.org/pdv/#AAB102FFFFFFFF8155

*** signal_start = 42525, signal_end = 99618, signal_len = 57093, pulses_found = 8
Iteration 1. t: 5598    min: 1728 (5)    max: 9469 (3)    delta 4787549
Iteration 2. t: 5598    min: 1728 (5)    max: 9469 (3)    delta 0
Pulse coding: Short pulse length 1728 - Long pulse length 9469

Short distance: 2, long distance: 4, packet distance: 8

p_limit: 5598
bitbuffer:: Number of rows: 4 
[00] { 2} 00        : 00
[01] { 1} 00        : 0
[02] { 1} 80        : 1
[03] { 4} a0        : 1010
Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta 3567587328
Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
Distance coding: Pulse length 0

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 0
bitbuffer:: Number of rows: 0 
```
How can i help decoding this ?

zuckschwerdt, 2023-03-01:
Do not ever use `-a`.
That signal is clipping (too "loud"). Use more distance or lower gain.
Offset tune for best results, e.g. `-f 169.50M`

fthiery, 2023-03-01:
Just learnt that Wize is AES-encoded, so that's game over i guess ;) Sorry for the noise, i'll go [the hardware route](https://github.com/reefab/homeassistant-gazpar-zigbee)

closed 2023-03-01 by fthiery

zuckschwerdt, 2023-03-01:
Thanks for the feedback, I'll update the guidelines.