merbanan/rtl_433 #2404: Partially detected Govee H5054 water leak sensors

Vlad-Star, 2023-03-03:
Re: https://github.com/pbkhrv/rtl_433-hass-addons/issues/124

I've bought two 5-packs of Govee H5054 sensors, one directly from Govee and another one from Amazon. In both sets the battery is detected by auto-discovery plugin, but the sensor itself is not. I am using the latest "next" version of rtl_433 and auto-discovery. 

Please, review and let me know if you need any further information. Thanks!

# Button press:
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (123 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:55:18\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Button Press\",\"code\":\"218b05050750\",\"mic\":\"PARITY\"}"
DEBUG:root:homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config:{"device_class": "timestamp", "name": "Govee-Water-56948-UTC", "entity_category": "diagnostic", "enabled_by_default": false, "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948/time", "unique_id": "Govee-Water-56948-UTC", "device": {"identifiers": ["Govee-Water-56948"], "name": "Govee-Water-56948", "model": "Govee-Water", "manufacturer": "rtl_433"}}
DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m54065), 'b'homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config'', ... (389 bytes)
INFO:root:Published Govee-Water/56948: time
INFO:root:Skipped Govee-Water/56948: event, code

# Shortened contacts:
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:56:32\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:56:34\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config

# Battery re-insert:
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:59:34\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (160 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-02-28 16:59:36\",\"model\":\"Govee-Water\",\"id\":56948,\"battery_ok\":1.0,\"battery_mV\":3000,\"event\":\"Battery Report\",\"code\":\"218b039b035a\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-B/config`

# My rtl_433.conf.template:
output mqtt://hass.example.com:1883,user=USER,pass=PASS,devices=rtl_433/9b13b3f4-rtl433/devices[/type][/model][/subtype][/channel][/id],events=rtl_433/9b13b3f4-rtl433/events,states=rtl_433/9b13b3f4-rtl433/states
frequency 433.92M
protocol 73  # LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth, (TFA, ORIA) sensor
protocol 41  # Acurite 986 Refrigerator / Freezer Thermometer
protocol 192 # Govee Water Leak Detector H5054, Door Contact Sensor B5023
protocol 231 # Govee Water Leak Detector H5054
convert si

zuckschwerdt, 2023-03-03:
Read the newer issues on https://github.com/merbanan/rtl_433/issues?q=is%3Aissue+Govee

Maybe @aav7fl and @jasonmichalski from https://github.com/merbanan/rtl_433/pull/2273 can help?

jasonmichalski, 2023-03-03:
I think this is working as expected.  There are two different (known) revisions floating around.  

One of them uses a PARITY check and reports these fields: https://github.com/merbanan/rtl_433/blob/master/src/devices/govee.c#L224

The newer version that I implemented uses a CRC check and reports these fields: https://github.com/merbanan/rtl_433/blob/master/src/devices/govee.c#L375

I don't have any of the older models to experiment with to say whether or not a `leak_num` exists in that message.

When you short the contacts, clear the alarm, wait a little bit, and short the contacts again, do you see any changing / incrementing values in the code word?

Vlad-Star, 2023-03-03:
Ok... to be clear, I just reinstalled both "next" versions,and  rtl_433 reported the following version upon startup:
```
Starting rtl_433 with rtl_433.conf...
[rtl_433] rtl_433 version nightly-2-g80d332cd branch master at 202303021545 inputs file rtl_tcp RTL-SDR
```
# Shortened the contacts:
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-03-03 13:56:03\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config:{"device_class": "timestamp", "name": "Govee-Water-56948-UTC", "entity_category": "diagnostic", "enabled_by_default": false, "icon": "mdi:clock-in", "state_topic": "rtl_433/9b13b3f4-rtl433/devices/Govee-Water/56948/time", "unique_id": "Govee-Water-56948-UTC", "device": {"identifiers": ["Govee-Water-56948"], "name": "Govee-Water-56948", "model": "Govee-Water", "manufacturer": "rtl_433"}}
DEBUG:paho.mqtt.client:Sending PUBLISH (d0, q0, r1, m15), 'b'homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config'', ... (389 bytes)
INFO:root:Published Govee-Water/56948: time
INFO:root:Skipped Govee-Water/56948: event, code
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-03-03 13:56:05\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config

# Pressed the button:
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (123 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-03-03 13:56:57\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Button Press\",\"code\":\"218b05050750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config

# Waited for a minute, then shortened the contacts again:
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-03-03 13:58:07\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config
DEBUG:paho.mqtt.client:Received PUBLISH (d0, q0, r0, m0), 'rtl_433/9b13b3f4-rtl433/events', ...  (121 bytes)
DEBUG:root:MQTT message: "{\"time\":\"2023-03-03 13:58:09\",\"model\":\"Govee-Water\",\"id\":56948,\"event\":\"Water Leak\",\"code\":\"218b04040750\",\"mic\":\"PARITY\"}"
DEBUG:root:Discovery timeout in the future for: homeassistant/sensor/Govee-Water-56948/Govee-Water-56948-UTC/config

Vlad-Star, 2023-03-03:
```
Data layout:
    II II XE DD CC CC
- I: 16 bit ID, does not change with battery change
- X: 4 bit, always 0x3 for the sensors evaluated
- E: 4 bit event type
- D: 8 bit event data
- C: CRC-16/AUG-CCITT, poly=0x1021, init=0x1d0f

```
In case of my H5054 sensors, I am getting 218b04040750 (leak) and 218b05050750 (button press) codes:
1) X is always 0x0
2) E is 0x4 for the leak and 0x5 for the button press
3) The DD field doesn't seem to be incrementing, even after re-inserting the batteries. Probably just repeating the XE field.

jasonmichalski, 2023-03-03:
unfortunately it looks like the version of the sensor you have doesn't have that `leak_num` in the message -- I was hoping maybe it was there and just wasn't extracted in the original implementation, but it seems like it's really not there at all

wish i had better news, but hope that helps!

Vlad-Star, 2023-03-03:
Thank you folks for helping out, I truly appreciate it! It seems this problem is not really related to rtl_433 code, but rather to the "old" revision of the sensors I've got and MQTT auto-discovery plugin, that seems to be expecting "new" version with `leak_num` field, in order to fully discover the sensor.

Closing it.

closed 2023-03-03 by Vlad-Star