merbanan/rtl_433 #2405: Doorbell with energy-harvesting button

TheChatty, 2023-03-03:
I've got some doorbells whose buttons have no battery, i.e. they harvest energy when a user pushes the button.

I can't get it to get decoded. Here are some samples: 
[Samples.zip](https://github.com/merbanan/rtl_433/files/10886859/Samples.zip)

zuckschwerdt, 2023-03-03:
The signal is badly clipping (too loud), remove the antenna, add more distance or set a low manual gain.

Also you are hitting the frequency dead on. DC will not work well, try to offset by e.g. 20k - 30k (`-f 433.90M` or `-f 433.89M`)

And then these senders usually are very fast to conserve energy, set a higher rate, e.g. `-s 1024k`

TheChatty, 2023-03-04:
[Samples.zip](https://github.com/merbanan/rtl_433/files/10888098/Samples.zip)
I recorded all bells now. From the neighboring room.

TheChatty, 2023-03-04:
I also took samples without antenna: 
[Samples-no-antenna.zip](https://github.com/merbanan/rtl_433/files/10888893/Samples-no-antenna.zip)

zuckschwerdt, 2023-03-04:
Still "too loud" and you really need `-f 433.90M`

TheChatty, 2023-03-04:
This time I shifted frequency and moved stepwise away from sender till I got no reception anymore.
[Uploading Samples-no-antenna.zip…]()

zuckschwerdt, 2023-03-04:
It looks like a simple PWM 50/150 µs signal. This will work:
`rtl_433 -R 0  -X 'n=name,m=OOK_PWM,s=50,l=150,g=200,r=3000,bits>=24'`
Or to match exactly this one button use:
`rtl_433 -R 0  -X 'n=name,m=OOK_PWM,s=50,l=150,g=200,r=3000,bits>=24,match=e0d50d,countonly'`

Write a conf documented [conf file](https://github.com/merbanan/rtl_433/tree/master/conf) and open a PR to include that if you like.

TheChatty, 2023-03-04:
When I execute the above command it fails after about one second with `Async read stalled, exiting!`
I'm using `rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR`on macOS 13.2 on a M2 MacBook Air.

TheChatty, 2023-03-04:
Bell 1 and 2 can be received with the above command **when `-s 1024k` is set**. This is the [specific product](https://www.amazon.de/gp/product/B09P5Y4G1B/).

Bell 3 is [this type](https://github.com/RFD-FHEM/RFFHEM/issues/946) and seems to need a different decoder.

gdt, 2023-09-26:
Do we have a lace to store these flex configs?   Open issues don't seem like the right place!

Cyberbeni, 2024-02-13:
I have this doorbell: [link](https://vi.aliexpress.com/item/1005002054725196.html)

The decoder mentioned above worked fine after also adding `-s 1024k`

zuckschwerdt, 2024-02-13:
If you have the time please write a documented conf file as suggested and PR.

TheChatty, 2024-05-26:
I filed this PR #2940.

closed 2024-05-26 by TheChatty