merbanan/rtl_433 #2407: Oregon Scientific WMR500

rVs21, 2023-03-04:
Hello,

The first analysis of Christian  :

_It's ~25 µs PCM, not something we have seen from OS before.
We get the PCM bits: {200} 55 55 55 55 A7 23 A7 23 E2 02 7D B5 C4 0C 44 41 D4 F5 4C 73 81 60 C5 1E 2C
Put that in BitBench and shift one bit, now we get
aaaaaaa d391 d391 f1013edae2062220ea7aa639c0b0628f16
That's a well know sync word (d391), not sure what the payload is though, too much data and very random looking._

I've uploaded some better cu8 captures and the result of rtl_433 -f 868.2M -X 'n=WMR500,m=FSK_PCM,s=25,l=25,r=3000,preamble=aad391d391'
(in log.txt from 7.1°C, 59%hum to 9.9°C, 49%hum)

[cu8.zip](https://github.com/merbanan/rtl_433/files/10888127/cu8.zip)
[log.txt](https://github.com/merbanan/rtl_433/files/10888126/log.txt)

Regards

zuckschwerdt, 2023-03-04:
Everybody else doing this: please compile your data into [a BitBench like this](https://triq.net/bitbench#c=%7B136%7Df1013edae2062220ea7ac0390cb0c8930a%20%5B11%3A01%3A48%5D&c=%7B224%7De6013ddae2062220ea7ac0390cb04cec6a7f4c7836f2453c5743b203%20%5B11%3A02%3A03%5D&c=%7B136%7Df1013cdae2062220ea7a60390cb06a5cc6%20%5B11%3A02%3A18%5D&c=%7B224%7De6013fdae2062220ea7a70390cb04cec6a7f4c7836f2453c57f55ef6%20%5B11%3A02%3A32%5D&c=%7B137%7Df1013edae2062220ea7a70390cb078a42a0%20%5B11%3A02%3A47%5D&c=%7B136%7Df1013cdae2062220ea7ae0390cb0ea63c6%20%5B11%3A03%3A17%5D&c=%7B225%7De6013fdae2062220ea7ae0390cb04cec6a7f4c7836f2453c57653efc0%20%5B11%3A03%3A31%5D&c=%7B136%7Df1013edae2062220ea7ae0390cb0e89cca%20%5B11%3A03%3A46%5D&c=%7B224%7De6013ddae2062220ea7ae0390cb04cfc697f4c7836f2453c57100710%20%5B11%3A04%3A01%5D&c=%7B137%7Df1013cdae2062220ea7ae0390cb0ea63c60%20%5B11%3A04%3A16%5D&c=%7B224%7De6013fdae2062220ea7ae0390cb04cfc697f4c7836f2453c5712cbf4%20%5B11%3A04%3A31%5D&c=%7B137%7Df1013edae2062220ea7af0390cb0f89b2a0%20%5B11%3A04%3A45%5D&c=%7B223%7De6013ddae2062220ea7af2391db04c6c6b7f18f06de48a78af87992c%20%5B11%3A05%3A00%5D&c=%7B136%7Df1013cdae2062220ea7ae2391db0fbe207%20%5B11%3A05%3A15%5D&c=%7B225%7De6013fdae2062220ea7ae2391db04c6c6b7f4c7836f2453c57f560aa0%20%5B11%3A05%3A30%5D&c=%7B135%7Df1013ed5c40c4441d4f5c4723b61f23a16%20%5B11%3A05%3A44%5D&c=%7B224%7De6013ddae2062220ea7af2391db04c6c147f4c7836f2453c57c26aff%20%5B11%3A05%3A59%5D&c=%7B137%7Df1013cdae2062220ea7af2391db0cb65240%20%5B11%3A06%3A14%5D&c=%7B224%7De6013fdae2062220e87a12391db04c6c147f4c7836f2453c57664fa8%20%5B11%3A06%3A29%5D&c=%7B135%7Df1013edae2062220ea7a12391db06269d6%20%5B11%3A06%3A43%5D&c=%7B224%7De6013ddae2062220ea7a12391db04cac6a7f4c7836f2453c57201ca6%20%5B11%3A06%3A58%5D&c=%7B136%7Df1013cdae2062220ea7a32391db00b45a4%20%5B11%3A07%3A13%5D&c=%7B224%7De6013fdae2062220ea7a32391db04cac6a7f4c7836f2453c57c212c2%20%5B11%3A07%3A28%5D&c=%7B137%7Df1013edae2062220ea7a32391db009baa80%20%5B11%3A07%3A42%5D&c=%7B225%7De6013ddae2062220ea7a32391db04ccc6a7f4c7836f2453c57e026f40%20%5B11%3A07%3A57%5D&c=%7B136%7Df1013cdae2062220ea7a32391db00b45a4%20%5B11%3A08%3A12%5D&c=%7B225%7De6013fdae2062220ea7a32391db04ccc6a7f4c7836f2453c57e2ea100%20%5B11%3A08%3A27%5D&c=%7B136%7Df1013edae2062220ea7a02391db019324b%20%5B11%3A08%3A41%5D&c=%7B224%7De6013ddae2062220ea7a02391db04cfc6a7f4c7836f2453c57807b1e%20%5B11%3A08%3A56%5D&c=%7B136%7Df1013cdae2062220ea7a02391db01bcd47%20%5B11%3A09%3A11%5D&c=%7B224%7De6013fdae2062220ea7a02391db04cfc6a7f4c7836f2453c5782b7fa%20%5B11%3A09%3A26%5D&c=%7B136%7Df1013edae2062220ea7a02391db019324b%20%5B11%3A09%3A41%5D&c=%7B225%7De6009eed71031110753d611c8ed8267e353fa63c1b79229e2ba03d888%20%5B11%3A09%3A55%5D&c=%7B136%7Df1013cdae2062220ea7ac2391db0dbedc7%20%5B11%3A10%3A10%5D&c=%7B224%7De6013fdae2062220ea7ac2391db04cfc6a7f4c7836f2453c5742b7f5%20%5B11%3A10%3A25%5D&c=%7B135%7Df1013ed5c40c4441d4f584723b61b22596%20%5B11%3A10%3A40%5D&c=%7B224%7De6013ddae2062220ea7ac2391db04cfc6a7f4c7836f2453c57407b11%20%5B11%3A10%3A54%5D&c=%7B137%7Df1013cdae2062220ea7a72391db04b5a240%20%5B11%3A11%3A09%5D&c=%7B224%7De6013fdae2062220ea7a62391db04cfc6a7f4c7836f2453c57e277ff%20%5B11%3A11%3A24%5D&c=%7B135%7Df1013edae2062220ea78c4723b60f24416%20%5B11%3A11%3A39%5D&c=%7B225%7De6013ddae2062220ea7a32391db04ccc6a7f4c7836f2453c57e026f40%20%5B11%3A11%3A53%5D&c=%7B136%7Df1013cdae2062220ea7ac2391db0dbedc7%20%5B11%3A12%3A08%5D&c=%7B224%7De6013fdae2062220ea7a12391db04ccc6a7f4c7836f2453c57c22a10%20%5B11%3A12%3A23%5D&c=%7B136%7Df1013edae2062220ea7a12391db06934eb%20%5B11%3A12%3A38%5D&c=%7B225%7De6013ddae2062220ea7a12391db04c8c6a7f4c7836f2453c57001a348%20%5B11%3A12%3A53%5D&c=%7B136%7Df1013cdae2062220ea7a12391db06bcbe7%20%5B11%3A13%3A07%5D&c=%7B224%7De6013fdae2062220ea7a12391db04c8c6a7f4c7836f2453c5702f88d%20%5B11%3A13%3A22%5D&c=%7B136%7Df1013edae2062220ea7a92391db0e90beb%20%5B11%3A13%3A37%5D&c=%7B225%7De6013ddae2062220ea7a92391db04c8c6b7f4c7836f2453c5783d47e0%20%5B11%3A13%3A52%5D&c=%7B136%7Df1013cdae2062220ea7ab2391db08b7aa4%20%5B11%3A14%3A06%5D&c=%7B225%7De6013fdae2062220ea7a62391db04c8c6b7f4c7836f2453c571539ce0%20%5B11%3A14%3A21%5D&c=%7B136%7Df1013edae2062220ea7a12391db06934eb%20%5B11%3A14%3A36%5D&c=%7B224%7De6013ddae2062220ea7a1c7915b04c6c6b7f4c7836f2453c57d96329%20%5B11%3A14%3A51%5D&c=%7B136%7Df1013cdae2062220ea7a1c7915b0d54a0e%20%5B11%3A15%3A05%5D&c=%7B224%7De6013fdae2062220ea7a1c7915b04c6c6b7f4c7836f2453c57dbafcd%20%5B11%3A15%3A20%5D&c=%7B136%7Df1013edae2062220ea7a1c7915b0d33519%20%5B11%3A15%3A35%5D&c=%7B225%7De6009eed71031110753d0e3c8ad8262e35bfa63c1b79229e2b94ce818%20%5B11%3A15%3A50%5D&c=%7B136%7Df1013cdae2062220ea7a6c7915b025dead%20%5B11%3A16%3A05%5D&c=%7B224%7De6013fdae2062220ea7a1c7915b04c5c6b7f4c7836f2453c572b51e7%20%5B11%3A16%3A19%5D&c=%7B136%7Df1013edae2062220ea7a1c7915b0d33519%20%5B11%3A16%3A34%5D&c=%7B224%7De6013ddae2062220ea7a1c7915b04c3c6b7f4c7836f2453c570965d1%20%5B11%3A16%3A49%5D&c=%7B136%7Df1013cdae2062220e87a6c7915b0235c9a%20%5B11%3A17%3A04%5D&c=%7B224%7De6013fdae2062220ea7a1c7915b04c3c6b7f4c7836f2453c570ba935%20%5B11%3A17%3A18%5D&c=%7B136%7Df1013edae2062220ea7a0c7915b0c332f9%20%5B11%3A17%3A33%5D&c=%7B224%7De6013ddae2062220ea7a3c7915b04ddc147f4c7836f2453c57499aad%20%5B11%3A17%3A48%5D&c=%7B136%7Df1013cdae2062220ea7afc7915b0b5e64d%20%5B11%3A18%3A03%5D&c=%7B224%7De6013fdae2062220ea7a0c7915b04ddc147f4c7836f2453c575b768a%20%5B11%3A18%3A17%5D&c=%7B136%7Df1013edae2062220ea7a0c7915b0c332f9%20%5B11%3A18%3A32%5D&c=%7B136%7Df1013cdae2062220ea7a0c7915b0c54dee%20%5B11%3A19%3A02%5D&c=%7B224%7De6013fdae2062220ea [... cut ...] 13edae2062220ea7a106975b07f73578%20%5B12%3A03%3A51%5D&f=8h8h%20). Much easier to work with.

Interesting, there seems to be a shorter `f101` and a longer `e601` message and sometimers also a `f102`.

But if we invert the bit we get `0efe` `19fe` and `19ff` -- the bits look "too heavy" when inverted, but 0x0e=14 and 0x19=25 are the length of those messages (minus length and checksum fields).

Last two byte are a CRC-16 with poly 0x8005, but the init varies with length, so something isn't prefect.

klohner, 2023-03-04:
I poked at this data a bit and couldn't find a way to get the CRC init values to align between the one used for the short messages and the one for the long messages.  Using inverted bits so that the `0x0e` and `0x19` bytes indicate length is probably not a coincidence.

I did discover that the byte just before the CRC in the short messages is a parity byte. (`poly=0x01`) but don't see a similar parity byte in the long messages.

```
$ reveng -w 16 -s 0efec1251df9dddf15853fc6f34f376cf5 0efec3251df9dddf15859fc6f34f95a339 0efec1251df9dddf15858fc6f34f875bd5 0efec3251df9dddf15851fc6f34f159c39
width=16  poly=0x8005  init=0x4ed0  refin=false  refout=false  xorout=0x0000  check=0x4260  residue=0x0000  name=(none)

$ reveng -c -w 16 -p 8005 -i 4ed0 -b 0efec1251df9dddf15853fc6f34f37
6cf5

$ reveng -w 8 -s 0efec1251df9dddf15853fc6f34f37 0efec3251df9dddf15859fc6f34f95 0efec1251df9dddf15858fc6f34f87 0efec3251df9dddf15851fc6f34f15
width=8  poly=0x01  init=0x10  refin=false  refout=false  xorout=0x00  check=0x21  residue=0x00  name=(none)
width=8  poly=0x01  init=0x08  refin=true  refout=true  xorout=0x00  check=0x21  residue=0x00  name=(none)

$ reveng -c -w 8 -p 01 -i 10 -b 0efec1251df9dddf15853fc6f34f
37
```

klohner, 2023-03-07:
FWIW, here's the flex spec I'm using for these samples:

`rtl_433 -Y minmax -X n=WMR500,m=FSK_PCM,s=26,l=26,r=312,invert,preamble=552c6e2c6e`

These files contained recognized Oregon-THGR810 data (instead of the relevant WMR500 data):

- g004_868.2M_1000k.cu8
- g009_868.2M_1000k.cu8
- g014_868.2M_1000k.cu8
- g019_868.2M_1000k.cu8
- g024_868.2M_1000k.cu8
- g029_868.2M_1000k.cu8
- g034_868.2M_1000k.cu8

Here's various updated bitbench views (with some manual data corrections):

- [All data bitbench](https://triq.net/bitbench#c=%7B224%7D19fec0251df9dddf15857bc6f74fb233e980b387c90dbac3a8997272%20%5Bg001_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf1585abc6f74fa755a5%20%5Bg002_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf1585abc6f74fb233e980b387c90dbac3a8cb5d9a%20%5Bg003_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf1585ebc6f74fe5b529%20%5Bg005_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf1585ebc6f74fb233e980b387c90dbac3a8091278%20%5Bg006_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15853bc6f74f376d45%20%5Bg007_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15853bc6f74fb223e980b387c90dbac3a8ab6bb6%20%5Bg008_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15853bc6f74f3592498%20%5Bg010_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15852bc6f74fb223e980b387c90dbac3a8594511%20%5Bg011_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15852bc6f74f276aa5%20%5Bg012_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15852bc6f74fb203ea80b387c90dbac3a8ba8383%20%5Bg013_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15852bc6f74f2595a9%20%5Bg015_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15859bc6f74fb203ea80b387c90dbac3a868ee2d%20%5Bg016_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15859bc6f74f975d85%20%5Bg017_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15859bc6f74fb3c3eb80b387c90dbac3a8aa7002%20%5Bg018_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15859bc6f74f95a289%20%5Bg020_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15859bc6f74fb3c3eb80b387c90dbac3a8a8bce6%20%5Bg021_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15859bc6f74f975d85%20%5Bg022_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf1585fbc6f74fb393ea80b387c90dbac3a8fb54ae%20%5Bg023_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15851bc6f74f159d89%20%5Bg025_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15851bc6f74fb393ea80b387c90dbac3a8195845%20%5Bg026_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15851bc6f74f176285%20%5Bg027_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15851bc6f74fb3939480b387c90dbac3a81d52ce%20%5Bg028_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15851bc6f74f159d89%20%5Bg030_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15851bc6f74fb3939480b387c90dbac3a81b1e31%20%5Bg031_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15851bc6f74f176285%20%5Bg032_868.2M_1000k.cu8%20corrected%5D&c=%7B224%7D19fec2251df9dddf15851bc6f74fb3539480b387c90dbac3a85d236f%20%5Bg033_868.2M_1000k.cu8%20corrected%5D&c=%7B136%7D0efec3251df9dddf15851bc6f74f159d89%20%5Bg035_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15851bc6f74fb3539480b387c90dbac3a85b6f90%20%5Bg036_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15851bc6f74f176285%20%5Bg037_868.2M_1000k.cu8%5D&f=LEN%3A8d%20%3F%3Ahh%20TYPE%3Ahh%20%3F%3Ahhhhhhhh%20%3F%3Ahhhhhh%20%3F%3Ahh%20%3F%3Ahhhhhh%20%3F%3Ahh%20hhhh%20hhhhhhhh%20hhhhhhhh%20hh%20hhhh&cw=4)
- [Short codes bitbench](https://triq.net/bitbench#c=%7B136%7D0efec1251df9dddf1585abc6f74fa755a5%20%5Bg002_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf1585ebc6f74fe5b529%20%5Bg005_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15853bc6f74f376d45%20%5Bg007_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15853bc6f74f359249%20%5Bg010_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15852bc6f74f276aa5%20%5Bg012_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15852bc6f74f2595a9%20%5Bg015_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15859bc6f74f975d85%20%5Bg017_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15859bc6f74f95a289%20%5Bg020_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15859bc6f74f975d85%20%5Bg022_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15851bc6f74f159d89%20%5Bg025_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15851bc6f74f176285%20%5Bg027_868.2M_1000k.cu8%5D&c=%7B136%7D0efec3251df9dddf15851bc6f74f159d89%20%5Bg030_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15851bc6f74f176285%20%5Bg032_868.2M_1000k.cu8%20corrected%5D&c=%7B136%7D0efec3251df9dddf15851bc6f74f159d89%20%5Bg035_868.2M_1000k.cu8%5D&c=%7B136%7D0efec1251df9dddf15851bc6f74f176285%20%5Bg037_868.2M_1000k.cu8%5D&f=LEN%3A8d%20%3F%3Ahh%20TYPE%3Ahh%20%3F%3Ahhhhhhhh%20%3F%3Ahhhhhh%20%3F%3Ahh%20%3F%3Ahhhhhh%20PARITY%3Ahh%20CRC-%F0%9D%9F%AA%F0%9D%9F%A2%F0%9D%9F%A2%F0%9D%9F%A7%3Ahhhh&cw=4)
- [Long codes bitbench](https://triq.net/bitbench#c=%7B224%7D19fec0251df9dddf15857bc6f74fb233e980b387c90dbac3a8997272%20%5Bg001_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf1585abc6f74fb233e980b387c90dbac3a8cb5d9a%20%5Bg003_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf1585ebc6f74fb233e980b387c90dbac3a8091278%20%5Bg006_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15853bc6f74fb223e980b387c90dbac3a8ab6bb6%20%5Bg008_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15852bc6f74fb223e980b387c90dbac3a8594511%20%5Bg011_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15852bc6f74fb203ea80b387c90dbac3a8ba8383%20%5Bg013_868.2M_1000k.cu8%5D&c=%7B224%7D19fec0251df9dddf15859bc6f74fb203ea80b387c90dbac3a868ee2d%20%5Bg016_868.2M_1000k.cu8%5D&c=%7B224%7D19fec2251df9dddf15859bc6f74fb3c3eb80b387c90dbac3a8aa7002%20%5B [... cut ...]
$ reveng -w 16 -s 0efec1251df9dddf1585abc6f74fa755a5 0efec3251df9dddf1585ebc6f74fe5b529 0efec1251df9dddf15853bc6f74f376d45 0efec3251df9dddf15853bc6f74f359249 0efec1251df9dddf15852bc6f74f276aa5 0efec3251df9dddf15852bc6f74f2595a9 0efec1251df9dddf15859bc6f74f975d85 0efec3251df9dddf15859bc6f74f95a289 0efec1251df9dddf15859bc6f74f975d85 0efec3251df9dddf15851bc6f74f159d89 0efec1251df9dddf15851bc6f74f176285 0efec3251df9dddf15851bc6f74f159d89 0efec1251df9dddf15851bc6f74f176285 0efec3251df9dddf15851bc6f74f159d89 0efec1251df9dddf15851bc6f74f176285 
width=16  poly=0x8005  init=0x4ed0  refin=false  refout=false  xorout=0x0000  check=0x4260  residue=0x0000  name=(none)

```

GardenShedDevelopments, 2023-10-04:
Hi all, I have a wmr500 that I would also like  to get connected. 

Is there any data I can help to submit to aid this issue? If so what is the best method to collect and submit?

gdt, 2023-10-06:
`more docs/*.md`.   Feel free to ask after reading if you still have questions.

gdt, 2024-06-04:
This needs work by the people with the devices to move to a PR.

klohner, 2024-06-05:
I took another look through all this.

- The WMR500 system consists of a base unit (which only receives the RF sensor data and is also a wifi transciever) and an "All-In-One" sensor (which is what transmits the RF data we're interested in).  This All-In-One sensor transmits at either 868 MHz (EU) or 915 MHz (US).  The captured data provided here is at 868 MHz.  [Oregon Scientific WMR500 Professional All-In-One Weather Station](https://www.oregonscientificstore.com/p-407-oregon-scientific-wmr500-professional-all-in-one-weather-station.aspx)

- In the 37 provided captures, 7 were already recognized at Oregon-THGR810 data (g004, g009, g014, g019, g024, g029, g034).  Typical Oregon sensor data is ASK (not the FSK we see in the other sample files) and I suspect these are from the All-In-One sensor.   But are the FSK samples from this device as well?  I suppose it's possible that there is a FSK transmitter for the wind and pressure data in the All-In-One, but it's strange that Oregon would use a new transmission method and protocol for this data.  [Oregon Scientific RF Protocol Description](https://www.osengr.org/WxShield/Downloads/OregonScientific-RF-Protocols-II.pdf)

Without more sample captures from other users of this device, I'm not sure how to go any further.  @GardenShedDevelopments - would you be able to capture signal data from your sensor and post here?

FWIW, here's flex decoders for the FSK signals.  I haven't had much luck in making sense of their data, though.

```
decoder {
    name        = Oregon_WMR500_LEN_25,
    modulation  = FSK_PCM,
    short       = 26,
    long        = 26,
    reset       = 312,
    invert,
    preamble    = {40}552c6e2c6e,
    bits        >= 287,
    bits        <= 288,
    get         = LEN:@0:{8}:%d,
    get         = TYPE:@16:{8}:%x,
    get         = CRC-8005:@208:{16}:%x,
}
decoder {
    name        = Oregon_WMR500_LEN_14,
    modulation  = FSK_PCM,
    short       = 26,
    long        = 26,
    reset       = 312,
    bits        >= 199,
    bits        <= 200,
    invert      ,
    preamble    = {40}552c6e2c6e,
    get         = LEN:@0:{8}:%d,
    get         = TYPE:@16:{8}:%x,
    get         = CRC-8005:@120:{16}:%x,
}
```

Here are examples of using rtl_433 to decode some data posted by @rVs21:

`$ rtl_433 -Y minmax -r g001_868.2M_1000k.cu8`
[image]

`$rtl_433 -Y minmax -r g002_868.2M_1000k.cu8`
[image]

`$rtl_433 -r g004_868.2M_1000k.cu8`
[image]