merbanan/rtl_433 #2408: Capturing 868Mhz water meter with unknown interval.

JoSSte, 2023-03-04:
I want to capture packets from my water meter which uses a Kamstrup Multical 2100 flow sensor communicating at 868MHz

https://www.kamstrup.com/en-en/water-solutions/meters-devices/meters/multical-21/documents

I set up a conf file:
```conf
output json
frequency 868M
``` 
and ran `rtl_433 -c /etc/rtl_433/service.conf -S unknown`
and let it sit for a while... but most of the captures seem to be random noise, at least compared to running the same but at 433.92MHz (I have attached a couple as examples)

### Conditions:
* The plumber who installed it mentioned that they could read these two streets away, in towns, but I have more than 60 meters to the nearest neighbour, so I hope that I don't get interference from them. 
* I am currently measuring from my office, which is quite a bit away from the device - should I move closer?
* I don't see any indication that it is sending...

What steps can I take to capture the sensor properly?

[scans.zip](https://github.com/merbanan/rtl_433/files/10888362/scans.zip)

zuckschwerdt, 2023-03-04:
https://triq.org/rtl_433/ANALYZE.html
Move close and remove the antenna to eliminate other signals.

Also it could be better to use `-f 868.3M`

merbanan, 2023-03-04:
First this is most likely a known protocol, wmbus. And second the wmbusmeters project will most likely be able to help you better.

JoSSte, 2023-03-04:
So I'll have to have a secondary sdr/rtl dongle and setup to capture that sensor?

closed 2023-03-05 by JoSSte

merbanan, 2023-03-05:
To record signals you need to use -S known. For wmbus I have used frequency 868.8M and 868.6M sucessfully. https://github.com/weetmuts/wmbusmeters has support for your meter and support for decryption which rtl_433 does not.