merbanan/rtl_433 #2415: Feature Request: option to make http API read only.

rct, 2023-03-11:
The HTTP client has gotten to be very impressive. I'm thinking about enabling HTTP by default on my (long running, aka "production") rtl_433 instances. 

To feel somewhat comfortable about doing that. It would be helpful to have an option to make the HTTP API read only to prevent unwanted changes.  This way the HTTP port can be enabled for monitoring with somewhat less concern about accidental problems

While it is somewhat intuitive that changing frequency/sampling rate would change the running rtl_433 process, options and modes aren't obvious.  Changes to options and modes might not be seen unless you are watching the logfile output. 

Thanks

zuckschwerdt, 2023-03-11:
With the recent merge of #1978 we now finally have the event loop available for the HTTP API.
The features you mention are on my roadmap for both the API and the web UI.

Lets keep this open to collect a wishliat of all things around the HTTP API and web UI.

rct, 2023-03-12:
Sounds good, and since you mentioned wishlist:

It would be helpful to have an RF/reception focused view in the web UI like @hdtodd's rtl_watch (#2409) that could be used to understand reception performance. 

A tabular view similar to rtl_watch with count of messages decoded, last SNR, SNR min/max, mean SNR, last received time (or seconds/minutes since last heard)

rct, 2023-04-15:
I didn't see it mentioned elsewhere: on iOS with both Safari and Chrome, the web UI fails to start leaving just a blank page.    I didn't get any errors logged with chrome://inspect  So I don't yet have any clues as to what the problem is. 

I did set Safari to request the desktop version of the site.

This was on iPadOS 16.3.1.

Somewhat surprising to me is that the Web UI does seem to work on a Kindle Fire with Amazon's Silk browser.

zuckschwerdt, 2023-04-15:
I know that Safari is missing an API. I never test on Chrome, it should work. I'll update the UI shortly and test more thoroughly. Very interesting about Kindle and Silk, thanks!

rct, 2023-04-17:
Ok, I seem to have been wrong.  I had left the blank tab up in iPadOS Safari, next time that tab was brought to the forefront, it started filling in the RXUI page and was eventually working.