merbanan/rtl_433 #2416: Feature Request: Disable decoders by type (i.e. TPMS)

rct, 2023-03-11:
(Apologies if this issue already exists, I though it did, couldn't find it.)

It would be nice to disable (or enable) decoders by type.  Would be handy to disable all TPMS decoders.

It also would be handy if remotes (devices that transmit when a button is pressed) that are more likely to generate false positives could be grouped under a type.

merbanan, 2023-03-11:
We dont have that information available. What we do have is the disabled flag that is enabled on all devices that dont have a crc or a checksum.

rct, 2023-03-11:
Right, but a number of the TPMS decoders have been coded to output a type field of "TPMS".  So decoders could supply that info when they register ...

gdt, 2023-10-15:
I guess the big question is if people think it's a reasonable idea to go down this path, of having a taxonomy of decoder types defined, and a new type field on registration, and finally options to process based on that.

There is also the question of existing fields and doing a rationalization rototill pass.  But it seems we are protective of backwards compat, which makes that nearly impossible.

rgregg, 2023-11-07:
I would love this feature - my MQTT log is filling up with more TPMS outputs - I can go squash the individual decoders, but it would be a lot easier to just turn off the decoders as a group, if that information were available.

zuckschwerdt, 2023-11-07:
Just editing the conf file isn't much trouble though.
`sed -i .bak 's/.*TPMS.*//' conf/rtl_433.example.conf`
or even better
`sd '.*TPMS.*' '#$0' conf/rtl_433.example.conf`