merbanan/rtl_433 #2422: Help me please, I can't read anything!

silkyclouds, 2023-03-13:
hi guys, 

I've got a new water meter and did watch this youtube video telling me I could read its content using a DVB-T usb key : https://www.youtube.com/watch?v=m5R6sfsGmvE&lc=UgwZeWsIuUNGae--8pV4AaABAg.9mux3sd2R1_9n2kaFudq2J

So, I ordered it, and it looks like, even if the key is 100% similar design wise, it runs another chip. 

FYI, the water meter is a Diehl G3 model, and the frequency is 868Mhz (it's written on the counter...) 

here is what I can see when I'm running rtl_433 : 

```
root@DietPi:/dev/serial/by-id# rtl_433 -f 868M -s 1024k
rtl_433 version nightly-3-g376f1b02 branch master at 202303101654 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" if you need the old defaults

[Protocols] Registered 209 out of 243 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119-121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-228 230-232 234-241 243 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[SDR] Using device 0: Realtek, RTL2838UHIDIR, SN: 00000001, "Generic RTL2832U OEM"
[R82XX] PLL not locked!
Allocating 15 zero-copy buffers
```

aaaaan that's about it. I can let it run for hours, nothing is displayed. I'd like to get this to work first, and see what the water meter broadcasts, before I start using this with wmbusmeters. 

Can anybody help ? Should I replace the key for a e4000 model ? 
Could it be that my water meter actually isn't broadcasting on the channels that it states ? 

any helps is welcome. Thanks !!!

zuckschwerdt, 2023-03-13:
The R820T tuner is fine. Acutally I prefer it over the E4000.

Go through https://triq.org/rtl_433/ANALYZE.html to find and grab a transmission. Then upload here as zip.
You can also run this to get more output:
`rtl_433 -f 868M -s 1024k -Y autolevel -M level -M noise -vv`
maybe even add `-A` to see every single piece of transmission

silkyclouds, 2023-03-13:
Thanks for such a fast reply @zuckschwerdt !

how long am I supposed to let it run to be sure I'm not mising my Diehl water meter announces ? 

oh and for what it's worth, I moved the rock64 device on which the usb key/antenna is plugged to the basement, just to be sure it's close enough of the water meter, the distance between the antenna and the meter is now less than a meter...which I believe should be allright. ;) 

for what it's worth (2), here is exactly what's mentionned on the meter : DIEHL metering, HRL-o- G3 868Mhz v2.3 

there is also a little "on" sign on the right side but I cannot see any button or anything... could it be that this specific meter needs to be "enabled" when the guy comes to make the yearly check of the meter?

I'm already sharing the first 5 minutes log here : https://pastebin.com/TsBBXn4B

thanks again !!!

silkyclouds, 2023-03-13:
OK so I remove the antenna as it's super close from the water meter, indeed. 

I restart the exact same exact command but do add -Y unknown in the params. And will analyse the exported file thourgh the website you provided. Cool. I'll keep you posted !!!

silkyclouds, 2023-03-13:
hmm -Y unknown is simply not working, can you please let me know the exact command to run in order to end up with a capture file ? Would be helpful. thanks !

merbanan, 2023-03-13:
-S and you can increase the sample rate to 1200k.

zuckschwerdt, 2023-03-13:
Yes, thanks. `-S unknown` sorry. And *do not* use `-W` -- it is not a txt log!

silkyclouds, 2023-03-13:
ok so just to be sure : 

rtl_433 -f 868M -S unknown -Y autolevel -M level -M noise -vv

is that right ? 

what will be the output file then ? do I need to add something ?

zuckschwerdt, 2023-03-13:
`rtl_433 -f 868M -s 1200k -Y autolevel -M level -M noise -v -S unknown`
you will get a bunch of .cu8 files, one per transmission.

silkyclouds, 2023-03-13:
ok cool, I'm letting this running in a screen for how long ? 

I can see it is creating the files by the way : 

*** Saving signal to file g001_868M_1200k.cu8 (33685 samples, 131072 bytes)
*** Saving signal to file g002_868M_1200k.cu8 (33358 samples, 131072 bytes)
*** Saving signal to file g003_868M_1200k.cu8 (33017 samples, 131072 bytes)
[Auto Level] Current noise level -33.0 dB, estimated noise -33.0 dB
*** Saving signal to file g004_868M_1200k.cu8 (33023 samples, 131072 bytes)
*** Saving signal to file g005_868M_1200k.cu8 (33089 samples, 131072 bytes)
*** Saving signal to file g006_868M_1200k.cu8 (33549 samples, 131072 bytes)
*** Saving signal to file g007_868M_1200k.cu8 (33221 samples, 131072 bytes)
*** Saving signal to file g008_868M_1200k.cu8 (35095 samples, 131072 bytes)

zuckschwerdt, 2023-03-13:
I guess 10-20 files, then look at them and zip up the interesting ones.

silkyclouds, 2023-03-13:
ok ok, so, I have a lot of these extracts now , and all of them seem to display the exact same thing on https://triq.org/pdv/

[image]

zuckschwerdt, 2023-03-13:
That is empty, just noise. you want a yellow/red streak in there like the usual SDR spectrograms ("waterfall").

silkyclouds, 2023-03-13:
Ok I just did notice I was allowed to upload several scan results at once. 

On 200+ checks, only a few ones do have some lines. I zipped them. 

[Archive.zip](https://github.com/merbanan/rtl_433/files/10961146/Archive.zip)

zuckschwerdt, 2023-03-13:
Those are too faint, likely just some random temperature sensors.
Take a peek here, that's what you are looking for (or roughly similar): https://github.com/merbanan/rtl_433_tests/tree/master/tests/m-bus/01

silkyclouds, 2023-03-13:
well, here is a picture of the sticker, not sure if it's a C or an O to be honest. Are you telling me that both models do exist ? 

https://imgur.com/a/arswqVa

zuckschwerdt, 2023-03-13:
It looks a bit more like a "c". And I found no mention of a "o" anyway.
LoRa is doable and mostly cheap (NodeMCU+LoRa) but the mention of keys is not good at all.

gdt, 2023-09-26:
This seems not actionable.

Jezza34000, 2024-04-26:
Hi ! I'm interessted in this subject, any news on it ?

zuckschwerdt, 2024-05-03:
Closing as LoRaWAN is not something rtl_433 can do. Anybody interested would need a LoRa adapter or tranceiver and the keys to read the transmission.

closed 2024-05-03 by zuckschwerdt

Blaise75, 2025-09-26:
Hello @silkyclouds , did you succed to read the information of the Diehl G3 ?
I have the same and i can't read the information.

[image]