merbanan/rtl_433 #2423: wont compile on RPi 2B+

Satsatt58, 2023-03-13:
sorry i followed every available installation guide, but without success.

`[ 93%] Linking C executable rtl_433
libr_433.a(mongoose.c.o): In function `mg_ssl_if_ossl_psk_cb':
mongoose.c:(.text+0x5730): undefined reference to `SSL_get_SSL_CTX'
mongoose.c:(.text+0x5738): undefined reference to `SSL_CTX_get_ex_data'
libr_433.a(mongoose.c.o): In function `mg_destroy_conn':
mongoose.c:(.text+0x651c): undefined reference to `SSL_free'
mongoose.c:(.text+0x6594): undefined reference to `SSL_CTX_free'
libr_433.a(mongoose.c.o): In function `mg_mgr_init_opt':
mongoose.c:(.text+0x66d8): undefined reference to `OPENSSL_init_ssl'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_init':
mongoose.c:(.text+0x77ac): undefined reference to `OPENSSL_init_ssl'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_conn_accept':
mongoose.c:(.text+0x77f4): undefined reference to `SSL_new'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_conn_init':
mongoose.c:(.text+0x789c): undefined reference to `TLS_client_method'
mongoose.c:(.text+0x78a0): undefined reference to `SSL_CTX_new'
mongoose.c:(.text+0x78b4): undefined reference to `SSL_CTX_set_options'
mongoose.c:(.text+0x78c0): undefined reference to `SSL_CTX_set_options'
mongoose.c:(.text+0x78cc): undefined reference to `SSL_CTX_set_options'
mongoose.c:(.text+0x7928): undefined reference to `SSL_CTX_set_verify'
mongoose.c:(.text+0x7938): undefined reference to `SSL_CTX_load_verify_locations'
mongoose.c:(.text+0x795c): undefined reference to `SSL_CTX_set_cipher_list'
mongoose.c:(.text+0x7988): undefined reference to `TLS_server_method'
mongoose.c:(.text+0x798c): undefined reference to `SSL_CTX_new'
mongoose.c:(.text+0x79e0): undefined reference to `SSL_CTX_use_certificate_file'
mongoose.c:(.text+0x7a38): undefined reference to `SSL_CTX_ctrl'
mongoose.c:(.text+0x7a44): undefined reference to `BIO_new_file'
mongoose.c:(.text+0x7a5c): undefined reference to `PEM_read_bio_DHparams'
mongoose.c:(.text+0x7a68): undefined reference to `BIO_free'
mongoose.c:(.text+0x7a84): undefined reference to `SSL_CTX_ctrl'
mongoose.c:(.text+0x7a90): undefined reference to `SSL_CTX_set_options'
mongoose.c:(.text+0x7a98): undefined reference to `DH_free'
mongoose.c:(.text+0x7bec): undefined reference to `SSL_CTX_set_psk_client_callback'
mongoose.c:(.text+0x7bfc): undefined reference to `SSL_CTX_set_ex_data'
mongoose.c:(.text+0x7c24): undefined reference to `SSL_ctrl'
mongoose.c:(.text+0x7c58): undefined reference to `SSL_CTX_use_PrivateKey_file'
mongoose.c:(.text+0x7c80): undefined reference to `SSL_new'
mongoose.c:(.text+0x7cc4): undefined reference to `SSL_CTX_use_certificate_chain_file'
mongoose.c:(.text+0x7cf0): undefined reference to `BIO_new_mem_buf'
mongoose.c:(.text+0x7d04): undefined reference to `PEM_read_bio_DHparams'
mongoose.c:(.text+0x7d10): undefined reference to `BIO_free'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_handshake':
mongoose.c:(.text+0x7d84): undefined reference to `SSL_get_fd'
mongoose.c:(.text+0x7d9c): undefined reference to `SSL_accept'
mongoose.c:(.text+0x7dbc): undefined reference to `SSL_get_error'
mongoose.c:(.text+0x7e08): undefined reference to `SSL_connect'
mongoose.c:(.text+0x7e1c): undefined reference to `SSL_set_fd'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_read':
mongoose.c:(.text+0x7e70): undefined reference to `SSL_read'
mongoose.c:(.text+0x7ed4): undefined reference to `SSL_get_error'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_write':
mongoose.c:(.text+0x7f60): undefined reference to `SSL_write'
mongoose.c:(.text+0x7fbc): undefined reference to `SSL_get_error'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_conn_close_notify':
mongoose.c:(.text+0x8040): undefined reference to `SSL_shutdown'
libr_433.a(mongoose.c.o): In function `mg_ssl_if_conn_free':
mongoose.c:(.text+0x806c): undefined reference to `SSL_free'
mongoose.c:(.text+0x809c): undefined reference to `SSL_CTX_free'
libr_433.a(mongoose.c.o): In function `mg_ssl_handshake':
mongoose.c:(.text+0xd368): undefined reference to `SSL_get_fd'
mongoose.c:(.text+0xd380): undefined reference to `SSL_accept'
mongoose.c:(.text+0xd39c): undefined reference to `SSL_get_error'
mongoose.c:(.text+0xd428): undefined reference to `SSL_connect'
mongoose.c:(.text+0xd478): undefined reference to `SSL_set_fd'
libr_433.a(mongoose.c.o): In function `mg_do_recv':
mongoose.c:(.text+0xd82c): undefined reference to `SSL_read'
mongoose.c:(.text+0xda14): undefined reference to `SSL_get_error'
libr_433.a(mongoose.c.o): In function `mg_close_conn':
mongoose.c:(.text+0xdbac): undefined reference to `SSL_shutdown'
libr_433.a(mongoose.c.o): In function `mg_if_accept_tcp_cb':
mongoose.c:(.text+0xdf08): undefined reference to `SSL_new'
collect2: error: ld returned 1 exit status
src/CMakeFiles/rtl_433.dir/build.make:98: die Regel für Ziel „src/rtl_433“ scheiterte
make[2]: *** [src/rtl_433] Fehler 1
CMakeFiles/Makefile2:1109: die Regel für Ziel „src/CMakeFiles/rtl_433.dir/all“ scheiterte
make[1]: *** [src/CMakeFiles/rtl_433.dir/all] Fehler 2
Makefile:138: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2

`

zuckschwerdt, 2023-03-13:
Seems you have a broken OpenSSL install. Use `cmake -DENABLE_OPENSSL=OFF ..` then build.

Satsatt58, 2023-03-13:
I want to marry you -- ;-))) Thanks a lot.

zuckschwerdt, 2023-03-13:
Glad it worked out. You may want to check /usr/local or /opt for remains of a broken OpenSSL install and remove those, then reinstall the original distro package.

Files to look for might be these (but with /usr/local prefix)
https://packages.debian.org/de/bullseye/armhf/openssl/filelist
https://packages.debian.org/bullseye/armhf/libssl-dev/filelist

Satsatt58, 2023-03-13:
only /usr/lib/openssh/       and  10days ago /usr/include/openssl/         nothing suspicious in /usr/local or /opt

zuckschwerdt, 2023-03-13:
Might be something else then. But if you don't need InfluxDB or MQTT with TLS you are good anyway.

zardoz99, 2023-08-26:
For what it's worth, I just hit the exact same problem. Testing the work around but I suspect that the a Raspbian issue.

# cat /etc/os-releasen
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"