merbanan/rtl_433 #2424: LaCrosse-TX141W

davlaw, 2023-03-14:
I have a LaCrosse-TX141W that often does not get fully charged by solar, but that is another project!

Sure this is my ignorance of the config, but is the following preventable?

Items will show up in home assistant fine as it did with LaCrosse-TX141W-0-265712-F etc etc

But what happens when the battery runs down and the unit later recharges  I will get a different id, per mqtt

rtl_433/raspberrypi/devices/LaCrosse-TX141W/0/328931
rtl_433/raspberrypi/devices/LaCrosse-TX141W/0/390625

davlaw, 2023-03-14:
disregard, found out this is normal for these devices

zuckschwerdt, 2023-03-14:
It's normal for most sensor devices, "burned-in" IDs are the exception sadly.

closed 2023-03-14 by zuckschwerdt