merbanan/rtl_433 #2433: Itron Cyble HRF 868Mhz V5.2

jonathanD91, 2023-03-18:
Hello, I have an Itron Cyble brand water meter. the mbus protocol is C2 and I can't listen to it. Are there any specific parameters for this protocol?

merbanan, 2023-03-18:
Hi, follow these instructions:

https://github.com/merbanan/rtl_433/blob/master/docs/ANALYZE.md

merbanan, 2023-03-18:
And the wmbusmeters project might be able to help also.

jonathanD91, 2023-03-18:
> Et le projet wmbusmeters pourrait également aider.

I opened an issue of wmbusmeters also to understand the configuration on C2 but I still have no answer. I will try while waiting for your solution.

gdt, 2023-09-26:
Please feel free to re-engage if you make progress.  This issues doesn't have actual content and it's stale, so I'm closing it.

closed 2023-09-26 by gdt

Anas92230, 2023-10-29:
> Hello, I have an Itron Cyble brand water meter. the mbus protocol is C2 and I can't listen to it. Are there any specific parameters for this protocol?

Hello did you find a solution? I have the same water meter.