merbanan/rtl_433 #2435: Strange fineoffset

Pascal66, 2023-03-19:
```
[fineoffset_WH24_callback] Raw @ bit_offset [60]
codes     : {136}24401f81ee6300000ca7001900713425eb
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-03-19 08:00:49
model     : Fineoffset-WH65B                       ID        : 64
Battery   : 1            Temperature: 9.4 C        Humidity  : 99 %          Wind direction: 287       Wind speed: 0.0 m/s       Gust speed: 0.0 m/s
Rainfall  : 822.7 mm     UV        : 25            UVI       : 0             Light     : 2898.0 lux    Integrity : CRC
[pulse_slicer_pcm] Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000, Misol WS2320 Temperature/Humidity/Pressure Sensor
codes     : {204}aaaaaaaaaaa2dd424401f81ee6300000ca7001900713425eb42
[ambientweather_whx_decode] WH31E/WH31B/WH40 detected, buffer is 204 bits length
[ambientweather_whx_decode] unknown message type 24 (expected 0x30/0x40/0x68)
[fineoffset_WH51_callback] Msg family unknown: 24
[fineoffset_wh1080_callback] Fine Offset WH1080 data
codes     : {80}ff24401f81ee6300000c
[lacrosse_breezepro_decode] Wrong packet length: 204
[lacrosse_wr1_decode] Packet too long: 204 bits
[lacrosse_th_decode] packet length: 204
[lacrosse_th_decode] Sync word not found
[lacrosse_r1_decode] packet length: 204
[lacrosse_r1_decode] Sync word not found
[honeywell_cm921_decode]
codes     : {417}cccccccccccccccccccccc0cf3f3300c30300001ffe001fe7e1e0780000000003c330fc0000078600003f030f300c33fccf8c0220
[honeywell_cm921_decode] preamble_start=417 start=447 len=-30
[fineoffset_wh31l_decode] WH31L detected, buffer is 125 bits length
[inkbird_ith20r_callback] 144 too short
[lacrosse_tx34_callback] LaCrosse IT frame bad CRC: calculated 30, received ee
[fineoffset_wn34_decode]
codes     : {72}24401f81ee6300000c
```

closed 2023-03-19 by zuckschwerdt

Pascal66, 2023-03-19:
Why is it closed ?

zuckschwerdt, 2023-03-19:
No error, all is good. Ambient pressure is measured at the base station, not the outdoor sensor.

Pascal66, 2023-03-19:
Ok...