merbanan/rtl_433 #2436: Dakota Alert 3000 series

macropin, 2023-03-20:
Hi all,

I've got a Dakota Alert 3000 series driveway monitor that I'm trying to decode. To my amateur eye it looks like FSK. I've tried a wide range of decoding options and haven't had any success.

Captures: [Dakota Alert.zip](https://github.com/merbanan/rtl_433/files/11013441/Dakota.Alert.zip) (cu8) are attached.

Any help, or pointers would be appreciated.

macropin, 2023-03-20:
Thanks, I'll try that and report back.

macropin, 2023-03-21:
Is this better? Captured with `rtl_433 -S all -g 15`. Anything less than 15dB and nothing is captured, same without an antenna. 

[g004_433.92M_250k_15db.cu8.zip](https://github.com/merbanan/rtl_433/files/11023988/g004_433.92M_250k_15db.cu8.zip)

zuckschwerdt, 2023-03-21:
200 meter? That's quite a good range! The signal looks good now. (But could be louder, 0 dB is clipping, at e.g. < -1 dB all is well).
Drop on https://triq.org/pdv/ and compare with the older signal.

Do not put "200m" in the filename -- it will be seen as "200M" (200MHz).
I don't think 250k is enough sample rate (the signal is too fast), use `-s 1000k`

zuckschwerdt, 2023-03-21:
Great decoding @klohner, it's rare to see PWM with FSK but it really could be PWM with leading gaps.
Packet gaps filled with mark frequency (the upper one) are strange, maybe it's designed flipped? Then it's "regular" PWM :)

macropin, 2023-03-22:
> 200 meter? That's quite a good range!

Actually 150m LoS, but I was indoors! I think the device is rated to 1km.

> Do not put "200m" in the filename -- it will be seen as "200M" (200MHz).

Oh thanks for the heads up, that explains the "weirdness" with my other captures at 100M coming up as 99Mhz.

And thanks @klohner, that's amazing. I'll try tomorrow, if it's not raining. It's a bit hard to trigger the sensor while I'm at a suitable distance which currently involves turning the sensitivity up to max and waiting for a semi to roll past...

gdt, 2024-06-04:
@macropin Any progress?  It seems like you are most of the way there.

macropin, 2024-06-07:
@gdt Life, responsibility and device reliability problems got in the way of my intentions to complete this and integrate using rtl_433. I ended up going a different route and used a Shelly relay (esp based) to integrate the device.

If you really want the decode config for this sorted, I can have another go, but can't give any definitive time frames. Feel free to close this ticket if necessary. Thank you to everyone that helped.

gdt, 2024-06-07:
Thanks for the status.  I'm often on the fence about keeping partial info open vs closing, but it's good to know where things stand.   Please don't majorly spend time on my account; I'm only really trying to push people to PR a flex decoder that alraedy exists so we can capture partial work.

gdt, 2024-06-07:
Linked into a partial decoder wiki page.   Happy to reopen if it becomes alive.

https://github.com/merbanan/rtl_433/wiki/Decoder-Issues-with-Some-Useful-Information

closed 2024-06-07 by gdt