merbanan/rtl_433 #2442: which command of intertechno remote is which?

Marc1200, 2023-03-21:
Hello, 

i have an klikaanklikuit(Kaku) motion sensor, it has been detected as intertechno-remote. i can see commando's when there is motion and when not. but is there some documentation for each command?
[image]

i also get a huge list of intertechno remotes, but i have only 1 motion sensor.
[image]

zuckschwerdt, 2023-03-21:
That's a false positive. Grab a sample to know what's really in the transmission. https://triq.org/rtl_433/ANALYZE.html

klohner, 2023-03-21:
Oops, maybe this?  I guess this makes more sense since it's on 433MHz.

- [KlikAanKlikUit Wireless Motion Sensor AWST-6000 70013 By COCO Smart Home](https://www.massivelighting.co.uk/klikaanklikuit-wireless-motion-sensor-awst-6000-70013-by-coco-smart-home-8836-p.asp)
- [Wireless motion sensor AWST-6000](https://switch-in.eu/en/product/wireless-motion-sensor/)

Marc1200, 2023-03-21:
@klohner it's indeed the awst-6000

Marc1200, 2023-03-21:
@zuckschwerdt so i should generate a cu8 sample file for you?

zuckschwerdt, 2023-03-21:
Yes, please grab a cu8 sample file for the AWST-6000 and check that it looks ok in https://triq.org/pdv/ then upload here as zip.

gdt, 2023-09-26:
This looks like a timed-out request with no actionable data.

gdt, 2024-06-04:
abandoned by submitter

closed 2024-06-04 by gdt