merbanan/rtl_433 #2444: Issues on Raspberry -> Receive Signal is weak on Pi -> on Mac OS X not ...

MadMacMad, 2023-03-22:
made some tests…
definitive the PI has not so good reception as another computer (Mac)
i’ll cleaned all build from source … and using rtl_433 on my Mac i’ll have much better Signal as when using it (fresh build) on two different Raspberry Pi 3+
with Pi i’ll have to keep a distance around 10-15m
with my MacBookPro i’ll can go more than 50 meter away and it’s still working fine… nooo signal loss …
don’t know where the different is but the Debian build of rtl_433 sucks on Raspberry …

ProfBoc75, 2023-03-22:
Hi, there are so many reasons for that.

You compared 2 computers that are not running into the same team and not at the same price too.

MacBookPro is certainly better shielded than the PI.

Raspberry is very sensitive to the RF disturbances, on the Pi 4, for example, the HDMI 1 alters the Wifi channel 1 to 4 because of the same frequency. the HDMI 2 is better and a workaround for that issue or change the Wifi channel to more than 4.

Raspberry power supply must also be of good quality, 5V / 3A is minimum (especially if the PI is loaded) and with good filtering and lower noise as possible,  help with noise filter cable ring (ferrite), you can add also this kind of filter to the USB cable between the PI and the SDR dongle if you have such installation.

The Bluetooth and Wifi could be stopped / disabled to remove interference too, if not required for sure.

And the antenna is the most important device into your installation, put the SDR dongle and antenna as far as possible from the computer will help too.

I'm using a PI 3+ since years now, with a RFXcom device (indoor antenna) and I'm able to get data from my garden sensors > 30 m , with windows solar protect (Pb) and trees in the middle.

Same, with another PI 4B + RTL R820T dongle, I'm able to get my weather station into the garden, at 868Mhz, no issue, and at more than 20 m and my PI 4 is at the opposite side room from the garden, so the signal needs to cross inside and outside walls.

Hope this can help you improve your rtl_433 experience on PI.

merbanan, 2023-03-22:
I think a cable between the dongle and the usb port is important on the sbc like devices.

MadMacMad, 2023-03-28:
> Hi, there are so many reasons for that.
>
> You compared 2 computers that are not running into the same team and not at the same price too.
>
> MacBookPro is certainly better shielded than the PI. [...]

many thx for reply !
yes you're right with the Pi compared to MacBook... 
disable Wifi & BT is a must and yes the Antenna has to be far more away from the PI ... 
after getting longer cable for the Antenna it's much better now... 
did not know that the PI is so sensible ... 
many thx
M.

ProfBoc75, 2023-03-28:
@MadMacMad , good ! 

In fact, any computer radiates and generates noise. More or less, depending on its shielding, and no shielding on the PI.
To limit the impact on the SDR dongle which is very sensitive to noise, it is therefore necessary to move it away from this noise, as well as its antenna, and to use good cables which will also limit this noise.
This explains your improvement.

closed 2023-09-26 by gdt