merbanan/rtl_433 #2445: Missing data in live capture mode for a new decoder

mgrenonville, 2023-03-22:
Hello, 

Thanks for you wonderful work. 
I'm currently trying to integrate a new device, a french hot water and heater named frisquet eco radio system vision, [based on a SX1211 chipset](https://github.com/merbanan/rtl_433/compare/master...mgrenonville:rtl_433:master) by semtech. 

Everything was running smoothly until somebody on a different location, with different hardware (SDR, computer) tried to help me by getting samples of his installation. His dump were missing data. The protocol is bi-directionnal, speaking from an address to another. And we had only on side of the communication. We suspected first a bad reception, but it appeared that missing frames where the nearest from the antenna.

Long story short, we recorded samples with `-S all` and it appears that some samples indeed contained missing data. When decoding these frame with `-r sample.cu8`, we got all events. We then use `-w raw_capture_868.9M_1000k.cu8`, to analyse spectrogram, and when decoded with `-r raw_capture_868.9M_1000k.cu8` it works flawlessly on both computer (mine and his)

But when capturing live data, we only have half of the events. 

To sum up the events we are talking about, here is a representaton [from_addr -> to_addr] to explain my self. There is 4 events : 
```
[A -> B]
~100ms
[B -> C]
~100ms
[C -> B]
~300ms
[B -> A]
```
We have these settings:
```
        .short_width = 40,   // short gap is ~38 us
        .long_width  = 40,   // long gap is ~38 us
        .reset_limit = 40200, 
        .tolerance   = 10,   
```
We only get [A -> B] and [C -> B] in live capture, but all of them in `-r file_FREQ_SAMPLING.cu8`

I've also made some tests, with a sx1211-like device of my conception that transmit the same events in the same order with approximatively the same timings, and on my computer, it works. I've also made a test on a raspberrypi to have less computing power, and sometimes, I miss some data.

Do you have an idea on why this occurs ? Did I misunderstand the [bitbuffer->rows](https://github.com/merbanan/rtl_433/compare/master...mgrenonville:rtl_433:master#diff-e84b3747126e630815ef4be9ee5642322f0915b0086ade7ad32bb4662e4b9394R146-R156) in decoding and I get 2 events in the same bit buffer, parsing only the first one?

Also, what do rtl_433 do when CPU is overwhelmed ? Does it drop data ? Is there a way to see it with a warning message ?

Thanks for you time and you wonderful work !

zuckschwerdt, 2023-03-22:
When you look at the .cu8 (use https://triq.org/spectrogram-next/ so you can scroll-zoom): do you see all messages?
Do they look the same? Or is there a big difference? Maybe upload a raw capture to some hoster and share a link here?

Also:
Reset is much too long. With PCM you need to ask "what is the longest run of 0's?" Then round up. E.g. if you have Manchester coding the longest run is 2 bit = 80 µs -> reset = 100. If you here have at least one 1-bit in each byte then worst case is 15 0's = 600 µs -> reset = 700.

If your CRC init is random then you perhaps started at the wrong bit.

Your decoder should not be named "sx1211". The protocol is much like "CC1101" but the real specific thing here is maybe "frisquet_eco"?

mgrenonville, 2023-03-22:
Hello, thanks for your response.

Maybe my issue wasn't clear enough. We are 100% sure that SDR receive radio correctly and cu8 file contains all data.

The biggest issue is that when capturing data with `-w` option while outputing data on stdout, the output is not the same than when reading the dump with `-r` option.

I'm not allowed by the other person to share publicly data since his network id may be clearly identified by a connected device to frisquet servers.

Loading the graph on the new spectrogram (the UX is way better !) gives me : 
[image]

We see the 4 packets, 2 with a very high RSSI, and 2 with a lower. The 2 packets with the highest RSSI are  missing in parser output.

merbanan, 2023-03-22:
This sounds like an issue with the FSK demod. If you attach a sample file and post the used command line we can know for sure.

merbanan, 2023-03-22:
> Also, what do rtl_433 do when CPU is overwhelmed ? Does it drop data ? Is there a way to see it with a warning message ?

And this practically never happens even though the FSK demod could use some optimizations.

merbanan, 2023-03-22:
Ok, the command line show that the minmax FSK demod is active because the frequency is higher then 800MHz. But when you do a play back with -r the frequency parameter is missing so it will use the classic FSK demod instead.

merbanan, 2023-03-22:
Add -Y classic to the command line and all will be good.

mgrenonville, 2023-03-22:
Woow ! You are right ! Adding the `-Y minmax` to the read command leads to missing data ! 

Thank you !!

closed 2023-03-23 by zuckschwerdt