merbanan/rtl_433 #2457: Acurite Sensor 00592TX signals stopped with release 22.11

NorthernMan54, 2023-03-27:
@rct @DAVe3283 I recently updated my rtl_433 installation to release 22.11, and found that after the update some of my Acurite Tower sensor devices stopped reporting.  I have double clicked on the issue, and found that the pr #1953 and the signal validation checks based on humidity introduced in the PR are the trigger for the problem.

When running analysis on the signal I'm seeing this in the log

```
acurite_txr_callback: row 1 bits 57, bytes 7, extra bits 1, msg type 0x04
acurite_tower_decode: 0x1765 Ch A : Impossible humidity: 127 %rH
acurite_txr_callback: stats: row 1, msg type 0x04, bytes 7, decoded 0, error -4
acurite_txr_callback: row 2 bits 57, bytes 7, extra bits 1, msg type 0x04
acurite_tower_decode: 0x1765 Ch A : Impossible humidity: 127 %rH
acurite_txr_callback: stats: row 2, msg type 0x04, bytes 7, decoded 0, error -4
acurite_txr_callback: row 3 bits 57, bytes 7, extra bits 1, msg type 0x04
acurite_tower_decode: 0x1765 Ch A : Impossible humidity: 127 %rH
acurite_txr_callback: stats: row 3, msg type 0x04, bytes 7, decoded 0, error -4
```

And after reviewing the code found that this line is the issue - https://github.com/merbanan/rtl_433/blob/0ce05017205a30a10033f89ae4171984d47c3629/src/devices/acurite.c#L1018

Acurite 00592TX - The problematic sensor ( it does not have a humidity sensor )
[image]

Acurite 0592TXR - Another sensor I had received with the Acurite tower, this one supports humidity.
[image]

In my setup I did change the problematic line to     `if ((humidity < 0 || humidity > 100) && humidity != 127 ) {` and the previous behaviour was restored.

zuckschwerdt, 2023-03-27:
Thanks, Support for 592TX added.

closed 2023-03-27 by zuckschwerdt

NorthernMan54, 2023-03-27:
Tks very much for this, I have updated my installation and can report that it is working fine.

commit 9e973ce5 (2023-03-27): Add support for Acurite 592TX (closes #2457)