merbanan/rtl_433 #2461: Inkbird-ITH20R

okolelovav, 2023-03-28:
Inkbird-ITH20R  
rtl_433 MQTT Auto Discovery result log
INFO:root:Published Inkbird-ITH20R/57658: time, temperature_C, humidity
INFO:root:Skipped Inkbird-ITH20R/57658: battery, sensor_num, temperature2_C
temperature2_C additional temperature sensor
Please add support battery and temperature2_C

zuckschwerdt, 2023-03-28:
Use a recent rtl_433 version. S.a #2220, i.e. 2f1e2a2.

Ninja1283, 2023-06-12:
With the ITH-20R-I onboard sensor, ITH-20R-O onboard sensor, and ITH-20R-O external temp probe, shouldn't there be three temperature topics, and two humidity topics? I'm only seeing two temps and one humidity.

zuckschwerdt, 2023-06-12:
What is `-I` and `-O`? Are those two different devices you have? You need to check the ID then.

The protocol transmits only 2 temps and 1 hum, see https://github.com/merbanan/rtl_433/blob/master/src/devices/inkbird_ith20r.c
And `temperature_2_C` is `external temperature probe`

Ninja1283, 2023-06-12:
> What is `-I` and `-O`? Are those two different devices you have? You need to check the ID then.

They are the **I**nside and **O**utside sensor modules:
[image]

Even the parameter list from Inkbird's [website](https://inkbird.com/products/digital-hygrometer-ith-20r) calls out five different readings:
[image]