merbanan/rtl_433 #2462: Unknown wireless M-Bus device

lio0909, 2023-03-28:
Hello,
I'm using your great project to get the values of my Bresser waether station, which works fine.
I recently discovered that my home water meter sends signal through wireless m-bus. I was able to get something through [104]  Wireless M-Bus, Mode C&T, 100kbps (-f 868.95M -s 1200k) but it looks like my device is not recognized correctly (device type: oil instead of water). Is there anyway to fix?
I've tested wmbusmeters, device was recognized correctly and data readable. But I would like to get everything through rtl_433. Is that possible? Can the izar driver be integrated?

Thanks you!

rtl_433 reports as follow:
```
center_frequency: 868300000                        frequencies: 868950000, 868300000                     hop_times : 20
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-03-29 00:17:16
model     : Bresser-6in1 id        : 42725167
channel   : 0            Battery OK: 1             Sensor type: 1            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 1             Rain      : 3.2 mm        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-03-29 00:17:28
model     : Bresser-6in1 id        : 42725167
channel   : 0            Battery OK: 1             Temperature: 8.6 C        Humidity  : 66            Sensor type: 1            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s       Direction : 0             Integrity : CRC
Tuned to 868.950MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
center_frequency: 868950000                        frequencies: 868950000, 868300000                     hop_times : 20
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-03-29 00:17:38
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 40902450
Version   : 132          Device Type: 0x01         Device Type String: Oil   Control   : 0x44          Data Length: 26           Data      : 1744304c4ac489408401a21102001329fe377d604b008def14697c8e          Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2023-03-29 00:17:46
model     : Wireless-MBus Mode      : T            Manufacturer: SAP         ID        : 40902450
Version   : 132          Device Type: 0x01         Device Type String: Oil   Control   : 0x44          Data Length: 26           Data      : 1744304c4ac489408401a22102001329c746a79d2974e7823f6e2144          Integrity : CRC
Control Info: 0xA2       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0000
```

wmbusmeters:
```
(telegram) DLL L=19 C=44 (from meter SND_NR) M=4c30 (SAP) A=4089c44a VER=00 TYPE=07 (Water meter) (driver izar) DEV=rtlwmbus[00000001] RSSI=87
(wmbus) parseELL @10 16
(wmbus) parseNWL @10 16
(wmbus) parseAFL @10 16
(wmbus) parseTPL @10 16
(telegram) TPL CI=a2
Received telegram from: 4089c44a
          manufacturer: (SAP) Sappel (0x4c30)
                  type: Water meter (0x07)
                   ver: 0x00
                device: rtlwmbus[00000001]
                  rssi: 87 dBm
                driver: izar
(wmbus) 000   : 19 length (25 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 304c dll-mfct (SAP)
(wmbus) 004   : 4ac48940 dll-id (4089c44a)
(wmbus) 008   : 00 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : a2 tpl-ci-field (Mfct specific)
(wmbus) 011 C?: 5102001329A28A5BCCCC4FEA837063 mfct specific
(wmbus) telegram from 4089c44a ignored by all configured meters!
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2023-03-28 22:36:40.000;92;92;4089C44A;0x1944304c4ac489408401a25102001329a28a5bcccc4fea837063<0A>"

Started auto rtlwmbus[00000001] listening on t1
telegram=|1944304C4AC489408401A2_71020013298CD4C89A8FE859354266|+9
{"media":"water","meter":"izar","name":"watermeter","id":"4089c44a","prefix":"C90AB","serial_number":"028682","total_m3":1270.277,"last_month_total_m3":1265.235,"last_month_measure_date":"2023-03-01","remaining_battery_life_y":1,"current_alarms":"no_alarm","previous_alarms":"no_alarm","transmit_period_s":8,"manufacture_year":"1990","timestamp":"2023-03-28T22:35:48Z","device":"rtlwmbus[00000001]","rssi_dbm":86}
```

lio0909, 2023-03-29:
Hi, thanks for your answer. It sounds this kind of implementation is not planned for rtl_433?
Since I don't need a constant high frequency update, do you think it would be possible to have kind of a script executing rtl_433 for 1 min, pause 5 min, execute wmbusmeters for 1 min and so on and so on?
I'm really not fimiliar with linux if someone could help me or put me on a lead?
Thanks!

gdt, 2023-09-26:
Seems like the answer above is "wontfix".

closed 2023-09-26 by gdt