merbanan/rtl_433 #2465: Release says Friedland EVO added but no ref found in main branch

Morriz, 2023-04-03:
Following [this release link](https://github.com/merbanan/rtl_433/blob/master/CHANGELOG.md#release-2011-2020-11-13) one can read that “Friedland EVO” was added, but it is nowhere found in the main branch. Does anybody know which code Friedland uses? We could then update the docs ;)

zuckschwerdt, 2023-04-03:
That's a .conf flex decoder, as usual with doorbell(-style) devices. It should be installed, but you need to locate and enable it with `-c path/to/friedlandevo.conf`. Otherwise view and get it here: https://github.com/merbanan/rtl_433/blob/master/conf/friedlandevo.conf

zuckschwerdt, 2023-04-03:
Oh, an important: change the .conf by deleting the `match =` line, then watch which code your doorbell is and add/edit that.

closed 2023-04-03 by zuckschwerdt

Morriz, 2023-04-04:
Tnx! Any idea what conf to use to make it spit out the same format? Using that decoder I see entries like this:
```
time      : 2023-04-04T23:07:58.376896+0200
model     : FriedlandEvo count     : 1             num_rows  : 1             len       : 110           data      : 4084102080824084104082104000
```

But when I use a device that has out of the box support I see entries like this:
```
time      : 2023-04-04T23:09:26.999956+0200
model     : Nexa-Security House Code: 5600755
Channel   : 3            State     : ON            Unit      : 3             Group     : 0
```
The latter actually does send the house code as id when I set the output to `mqtt` and makes it usable.

Morriz, 2023-04-04:
I would expect the flex decoder to have a config option to map raw fields to output fields, for example:
```
decoder {
    name        = FriedlandEvo,
    modulation  = OOK_PCM,
    short       = 750,
    long        = 750,
    reset       = 9000,
    preamble    = AA,
    match       = 0x410408210400,
    unique,
    data_map = {
        Channel    = 123
        State      = ON
        Group      = 0
        Unit       = 3
        House Code = 1234567
    }
}
```

zuckschwerdt, 2023-04-05:
There is an option to extract data fields with a flex conf (`get=`)  but doorbells just transmit one long unique "ID" or "serial number".

Morriz, 2023-04-06:
Ah, interesting. The only field that contains an identifier is indeed called `data`, and all I hope for is to get that field to be mapped instead to `id`, as it is the expected format for a home assistant mqtt device as I understand it.

zuckschwerdt, 2023-04-06:
That's easy: `get=@0:{48}:id`
But seeing `data : 4084102080824084104082104000` means the timings are not correct -- those are two and a partial repeat and they are not split and not even matching.

Morriz, 2023-04-07:
Tnx! Seeing that the data never changes over time I can use it as an id I'd think.