merbanan/rtl_433 #2467: Device not recognized

mtmmtm9, 2023-04-03:
I use rtl_sdr on OSX (install through brew). It works fine:
rtl_sdr  -s 2400000 -f 88500000  capture.bin
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sampling at 2400000 S/s.
Tuned to 88500000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...

It does not work using rtl_433:
rtl_433 
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/mp/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
No supported devices found.

zuckschwerdt, 2023-04-04:
It not obvious why this might happen. Maybe some mixup with libusb or librtlsdr libraries?
MacPorts and Brew versions are known to work well.

mtmmtm9, 2023-04-12:
I got it working now. Not sure exactly why it started to work.

closed 2023-04-12 by mtmmtm9