merbanan/rtl_433 #2470: Just stopped!

SolarCzar, 2023-04-04:
Hey guys,

I've been using RTL-433 for long time, and I'm ashamed to say, that I've forgotten everything I did to set it up. Sorry.  I simply use it to push temperatures to Home Asst.  I have had RTL-433 running on a RPi and it's been great.  This week it's not receiving any communications.  I thought it must be the RPi, so I loaded the RTL-433 app on a VirtualBox instance.  Same thing, but this time, my Thermo Hydro Screen stopped showing temps.  I have converted many of my sensors using AC -> DC transformers, to avoid battery loss.  I power cycled the sensors and it came up.  I unfortunately have to go in and redefine the MQTT id's in Home Asst, which is a PITA by the way.  Hope someone could come up with a way to cure this.

But why did it lose comm?

zuckschwerdt, 2023-04-05:
So reseting rtl_433 or the Raspi, and even using another host (VM) did not help but reseting the sensor fixed the issue?

It could be frequency drift but fixing everything with a sensor reset sounds strange.

ProfBoc75, 2023-04-05:
Hi , 

It can be due to another equipment that interferes at the same frequency, especially if this one is facing a low battery.

Another point, some sensors may become damaged over time and thus create problems, depending on the climatic conditions in which they are exposed to heat and humidity. I had the case of spiders in the temperature and humidity sensors that distorted the humidity value.

Try with RTL-SDR tools (like SDRSharp ) to see if you have some noise at that frequency, this help for me.

SolarCzar, 2023-04-06:
As I go through it, I got all my sensors back except my Govee H5054 (Protocol 192) which stopped being captured.  I've got about 8 of them.  Replaced batteries, but no joy.  I notice that there is a newer Protocol 231, but that is not included in the 21.12 version I'm running on Ubuntu Linux.  An

drew-hill, 2023-04-10:
22.11 also does not include Protocol 231, in fact. I'm unable to adequately capture my Govee 5054 as well. Using the following command ([pulled from here](https://github.com/merbanan/rtl_433/issues/1518)) gives me the ability to see it, but I'm not sure which Protocol is being used here, and I don't seem to be getting any useful data in the readouts: 
```rtl_433 -X "n=Govee_H5054,m=OOK_PULSE_PWM,s=440,l=940,g=900,r=9000,bits=48,unique"```

gdt, 2023-09-26:
This doesn't seem actionable.  Please update to the latest git master, and then if you can articulate that there's a bug, feel free to open a new issue.

closed 2023-09-26 by gdt