merbanan/rtl_433 #2472: Wrong rain_mm in JSON at Bresser Weather Center 5-in-1 (type 172)

dl7ata, 2023-04-05:
I use this command to send a mqtt-JSON string:
_rtl_433 -F log -F json -M iso -f 868M -R 172 -C si -Y squelch | mosquitto_pub -t 7ata/wx/raw -l_

By random, I received some 68258.2 values in the transmitted JSON (rain_mm) instead of correct total amount of rain numbers like 92.4.
You can see these "random spikes" at [https://de.aprs.fi/weather/a/DB0TGO-14?range=day](url) and also at [https://aprs.fi/weather/a/DM0EWE-13?range=week](url)
Any idea what to do?

zuckschwerdt, 2023-04-05:
Rain is absoulte not relative. Derive the timeframe youself. E.g. substract daily values.

closed 2023-04-05 by zuckschwerdt

zuckschwerdt, 2023-04-05:
Oh and if you mean exactly the "Bresser-5in1" then there should only be 3 digits and one decimal, true. Seems like a broken transmission then. Try to get codes with `rtl_433 -R 173:vv`
And there is built-in MQTT with e.g. `-F mqtt://host,events=7ata/wx/raw`

dl7ata, 2023-04-05:
> Rain is absoulte not relative. Derive the timeframe youself. E.g. substract daily values.

That's what I do with the received mqtt - frame in a separate python script. But sometimes the absolute rain becomes > 60000.

I'll give it a try with -R 173:vv
But this means, that 172 - which is my station - doesn't work proper.

zuckschwerdt, 2023-04-05:
Sorry, I meant `-R 172:vv` i.e. just add `:vv` (verbose) to what you already have.

dl7ata, 2023-04-05:
Will try to find out. At first, I have to learn the protocol.   :cry: