merbanan/rtl_433 #2477: Multiple frequencies reported by sensor Fineoffset-WN34

FosterSM, 2023-04-09:
Hi, I have a weather system that monitors a number of sensors from various manufacturers. I have found that the various sensors report slightly different tx frequencies and to get a reliable overall capture of data I have opted to use the "hop" feature from the rtl_433:

rtl_433 -H 7 -f 433910000 -f 433915000 -f 433920000 -f 433925000 -Y autolevel -F json -M bst -M level -d 0

This has significatly improved the reliability of my overall capture responses, the frequencies chosen and the hop value should be considered a work in progress and in no way cast in stone but at the moment I am happy with these values.

While analysing the responses to try and tune my frequency and hop selections I noticed that one of the sensors (Fineoffset-WN34) reports two frequencies freq1 and freq2 does anyone know the signifcance of this and in general can the returned freq values be relied upon in any way.
Thanks

gdt, 2023-04-09:
This seems like not a bug in rtl_433.

I would advise looking at and graphing the metadata in decoded packets.  You are only moving 15 kHz and normally the receive bandwidth is much more than that.   I find that sensors tend to be sort of close to nominal, good dongles very accurate, and cheap dongles could be most of 100 ppm off.

FosterSM, 2023-04-09:
Thanks for the reply, I did not want to imply that there is a bug in the rtl_433 I was just trying to understand the behaviour I am seeing and to get a better understanding of the data I am receiving.

gdt, 2023-04-09:
I mentioned about bug because you filed an issue vs a discussion....

FosterSM, 2023-04-09:
I will move this to the discussion list sorry.

gdt, 2023-04-09:
Moved to #2478

closed 2023-04-09 by gdt