merbanan/rtl_433 #2479: OOK_MC_ZEROBIT decoder behaves unexpectedly with a non-Manchester prefix, possibly leading to incorrect decodes

klohner, 2023-04-11:
In trying to work on improvements to protocol 70, the Honeywell 345 MHz decoder, I'm running into issues with how rtl_433 applies the OOK_MC_ZEROBIT decoder returns data.

For example, I'm looking at 
[rtl_433_tests/tree/master/tests/honeywell_5816)/g002_344.975M_250k.cu8](https://github.com/merbanan/rtl_433_tests/blob/75c5a508a942aa8c3a09dc57f7b335e03f3ea654/tests/honeywell_5816/g002_344.975M_250k.cu8).  

Granted, this is not a particularly clean signal with high bitrate, but it's a good example to illustrate this issue.  This signal is OOK, even though it appears ASK, (see [issue 1174397711](https://github.com/merbanan/rtl_433/issues/2015#issue-1174397711) for background).

Here's a URH representation of of one message in the demodulated signal
[image]

I can ask rtl_433 to get that PCM data with
`rtl_433 -r tests/honeywell_5816/g002_344.975M_250k.cu8 -X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408"
`
and it'll tell me this is:
`{162}ffffffffeaaaaaacd54ad52ad52acd54cd2ad2d20`
or:
`0y111111111111111111111111111111111110101010101010101010101010110011010101010010101101010100101010110101010010101011001101010101001100110100101010110100101101001000`

This is fine and I wouldn't expect better when looking at this signal.  The part just before the Manchester data is being seen as pulses, but once the real pulses start, they all get decoded properly.

I can feed that data to bitbench and have it decode the useful Manchester data.  I just have it look for a preamble of `{2}8` and do a "Manchester (G.E. Thomas)" decode and invert. to get 
`y11111111111110100000111000011110000111101000001010011110011001` which is `{62}fffa0e1e1e829e64` (I have to be careful in bitbench with how it represents a final partial nibble in hex!).  This is data I'm trying to get rtl_433 to provide.

[bitbench](https://triq.net/bitbench#c=%7B162%7Dffffffffeaaaaaacd54ad52ad52acd54cd2ad2d20%20%5BI%20want%20to%20decode%20to%20%7B62%7Dfffa0e1e1e829e64%5D&f=b%0Ahhhhhhhhhhhhhhhh%0Ahhhh%20hhhh%20hhhh%20hhh%20bbbb&a=Preamble&m=y10&s=1&i=true&d=MC&cw=4)

But when I ask rtl_433 for this data with

`rtl_433 -r tests/honeywell_5816/g002_344.975M_250k.cu8 -X "n=Honeywell,m=OOK_MC_ZEROBIT,s=136,l=136,r=408,invert"
`
it tells me this is `{64}8000838787a0a799`.

Oh, you think, just don't invert it.  Nope.  That's not the issue.

`rtl_433 -r tests/honeywell_5816/g002_344.975M_250k.cu8 -X "n=Honeywell,m=OOK_MC_ZEROBIT,s=136,l=136,r=408"
code: {64}7fff7c78785f5866
`

This [bitbench](https://triq.net/bitbench#c=0y00%200x%7B62%7Dfffa0e1e1e829e64%20%20%5B%7B62%7Dfffa0e1e1e829e64%20Correct%20and%20desired%20decoding%2C%20pre-padded%20with%20two%200%20bits%20for%20alignment%5D&c=%7B64%7D8000838787a0a799%20%5B%7B64%7D8000838787a0a799%20-X%20%22n%3DHoneywell%2Cm%3DOOK_MC_ZEROBIT%2Cs%3D136%2Cl%3D136%2Cr%3D408%2Cinvert%22%5D&c=%7B64%7D7fff7c78785f5866%20%5B%7B64%7D7fff7c78785f5866-X%20%22n%3DHoneywell%2Cm%3DOOK_MC_ZEROBIT%2Cs%3D136%2Cl%3D136%2Cr%3D408%22%5D&c=%5B%5D&f=b&cw=4)  illustrates the issue.

```
0011111111111110100000111000011110000111101000001010011110011001 [{62}fffa0e1e1e829e64 Correct and desired decoding, pre-padded with two 0 bits for alignment]
1000000000000000100000111000011110000111101000001010011110011001 [{64}8000838787a0a799 -X "n=Honeywell,m=OOK_MC_ZEROBIT,s=136,l=136,r=408,invert"]
0111111111111111011111000111100001111000010111110101100001100110 [{64}7fff7c78785f5866 -X "n=Honeywell,m=OOK_MC_ZEROBIT,s=136,l=136,r=408"]

```

It looks like the `-X "n=Honeywell,m=OOK_MC_ZEROBIT,s=136,l=136,r=408,invert"` decoder is getting tripped up with the invalid Manchester "1111111" bits prefix and starts decoding out of sync until it realizes this and then adjusts.

The issue is that this impacts downstream decoding.  A valid message should start with a prefix like "{14}fff8" and now this doesn't appear.  It gets difficult to compensate for this because the transition we need to look for to signify the end of the prefix is now gone in the OOK_MC_ZEROBIT decoding, and the last two bits of valid Manchester data seem to be missing from the OOK_MC_ZEROBIT decoding.

I'd image at this point that changing/fixing the behavior of OOK_MC_ZEROBIT would negatively impact lots of existing decoders.

Perhaps instead, there's an opportunity to add options like "decode_zbmc" for (zero-bit manchester) G.E. Thomas Convention and "decode_obmc"  for (one-bit manchester) IEEE 802.3 Convention to the flex decoder so I could do something like:

`-X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408,invert,preamble={4}5,decode_obmc"` 
for the normal Manchester decoding the way that 
`-X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408,invert,preamble={4}5,decode_dm"`
works for a differential Manchester decoding?

Thoughts?

zuckschwerdt, 2023-04-11:
The problem is not the invalid data but the unaligned start of MC with non-recoverable alignment, i.e. is it `1010`… or `0101`…? You only know that when you hit the `11`. The slicer then re-syncs at that point.
We could buffer all short transitions until we find the alignment, but that would be a complicated algorithm if we also handle all possible errors.

zuckschwerdt, 2023-04-11:
Oh and `ZEROBIT` is not about G.E.T vs IEEE but the handling of the first edge.

klohner, 2023-04-11:
Ah, okay.  Then how about this:

- Allow the existing "preamble=<bits>" to be recognized by OOK_MC_ZEROBIT (preamble doesn't seem to have any effect on OOK_MC_ZEROBIT now).
- Create a new "sync=<bits>" option, also recognized by OOK_MC_ZEROBIT, that gets evaluated before preamble (if that also exists).  It might be useful to utilize both a sync and preamble in some cases.

This way I could do something like:

`-X "n=Honeywell,m=OOK_MC_ZEROBIT,sync={24}555556,s=136,l=136,r=408"`

Or, maybe better, create new "decode_mc_get" and "decode_mc_ieee" options to handle the decode, and have these decoding terminate at the first invalid symbol pair after successfully applying sync/preamble

-X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408,sync={24}555556,decode_mc_ieee"

I'm hoping to get a result like `{60}ffe838787a0a799`.  Maybe there's a way?

zuckschwerdt, 2023-04-11:
Yes, some additional options would be nice. It needs to be general and obvious though.

I don't think `decode_pwm` would work. A decoder operates on bits, and pwm operates on pulses.

What do you mean by "terminate on the first non-PWM group"?

klohner, 2023-04-11:
True, pwm operates on pulses in a pulse period.  But from my experience with many samples, pwm signals have a nice duration ratio between a short and long pulse, and the pwm periods are constant. 

For example, in my signal above, the pcm decoding is:

`{2002}9249249249c69a4d24926936db49249a49249249249249271a69349249a4db6d249269249249249249249c69a4d24926936db49249a49249249249249271a69349249a4db6d249269249249249249249c69a4d24926936db49249a49249249249249271a69349249a4db6d249269249249249249249c69a4d24926936db49249a49249249249249271a69349249a4db6d249269249249249249249c69a4d24926936db49249a49249249249249271a69349249a4db6d249269249249249249249c69a4d24926936db49249a49249249249249271a69349249a4db6d249269249249249249249c69a4d24926936db49249a49249249249249271c0`

It's essentially repeating the same shorter message a lot of times.  Here's a PCM decoding of just a bit of that signal as binary:

`y100100111000110100110100100110100100100100100100110100100110110110110110100100100100100100110100100100100100100100100100100100100100100100100100100100100100111000110100`

The signal is really a PWM signal, but every PWM period is 480μs, and the pulses in that period are either 160μs or 320μs.  In this PCM decode, I see either "100" or "110" for logical PWM "0" or "1" bits.  From my experience, the vast majority of PWM signals quantize into tidy ratios like this.

I'd define this PWM period as "3" since groups of 3 symbols could be pwm_decoded as either "0" or "1".  I've seen other PWM signals where the PWM period is "4" because a PCM decoding gives groups of "1000" or "1110" for PCM "0" or "1" logical bits.  Some PWM periods quantize into slightly larger groups, but not too often.

But between "rows" of repetition of data in my example, there is PCM "111000".  "111" is not really a "valid" PCM bit group (I'd posit that the only "valid" PCM groups start with "1", end with "0", have exactly 1 transition, and the ratio of time of the sum of "1" pulses to the total PWM period is either <50% or >50%)

So, I'd like to have a "decode_pwm=3" option on PCM data where:

- I could specify some kind of "sync" and "preamble" to the PCM data (not pwm decoded data) to tell the "decode_pwm" decoder where to start  (i.e.  sync={8}71,preamble={6}e0 or somesuch that looks at the PCM data bits and not any kind of PWM decoding).  This would allow me to synchronize the beginning of a decode_pwm just after a "111000"
- decode_pwm=3 would translate groups of three PCM bits, i.e. "100" to a logical 0 bit, "110" to a "1" bit.  It would inherently know that a 33% duty cycle is <50%, so that's a 0 a 66% duty cycle is >50% so that's 1.  But as soon as it encounters a "111" group, it knows that's invalid and terminates there, giving that "111" back to the buffer.
- Then it uses the PCM "preamble" or maybe some kind of PCM "row_separator" bitstring to split the rest of the data in the buffer into rows and try more decode_pwm decodings.

So, that's where I'd get the `{48}a409f0200000` decoded data, or may rows, of `{48}a409f0200000` data (if there's some facility to specify row separator bitstrings.

zuckschwerdt, 2023-04-11:
I see. You can already do that (and much more general) with "symbol_zero", "symbol_one", "symbol_sync".
Some examples are here: https://github.com/merbanan/rtl_433/issues?q=symbol_zero

klohner, 2023-04-12:
That's great!  Thanks to those "symbol_zero" and "symbol_one" options, I was able to formulate a workaround for both the behavior of OOK_MC_ZEROBIT (for Manchester decoding the Honeywell Series 5800 sensors signal) and even get a PWM decode in my desired format for Honeywell ActivLink devices.  Thank you.

```
rtl_433 -r tests/honeywell_5816/g002_344.975M_250k.cu8 -X "n=Honeywell,m=OOK_PCM,s=136,l=136,r=408,preamble={24}555556,symbol_zero={2}8,symbol_one={2}4"
{48}838787a0a799

```

```
rtl_433 -r g001_916.8M_1000k.cu8 -X "n=ActivLink,m=FSK_PCM,s=160,l=160,r=640,preamble={7}70,symbol_zero={3}8,symbol_one={3}c"
{48}a409f0200000

```
[g001_916.8M_1000k.zip](https://github.com/merbanan/rtl_433/files/11206956/g001_916.8M_1000k.zip)

closed 2023-04-12 by klohner

rct, 2023-04-14:
Maybe this is a dumb question, but is it possible to apply what you've learned to the existing builtin Honeywell decoder?  it seems like you get a few more successful decodes.