merbanan/rtl_433 #2482: Decoding AREXX TSN-70E

inonoob, 2023-04-16:
Dear all, 

I have a bunch of AREXX TSN-70E on my desk. I would like to work with them. 

some background info: 

the sensor: https://www.arexx.com/templogger/html/de/products.php 

the inside view with the chips: https://domoticx.com/arexx-module-tsn-th70e/ 

I managed after some time to find the hopefully right settings to decode with rtl_433 the signals: 

`rtl_433 -X 'n=AREXX,m=FSK_PWM,s=208,l=424,r=456,g=0,t=0,y=24`

With that I manged to catch a lot of packages but after several days of trying I can't figure the coding out :(. 

I have found the following: 

for the Temperature: 

T = p0 + p1* SOt 

p0= -39.6
p1= 0.01 
SOt= Digital readout 

for the Humidity: 

RH = p0 + SOrh *(p1 + SOrh *p2)

p0=-4
p1=0.0405
p2=-2.8e-6
SOrh = Digital readout 

Also from the data I could understand that the ID switch between the different information send. For example if the id is 20010 for temperature, the humidity will be reported as 20011. 

in addition it seems that "87" could be the sensor typ as it is the same for all my AREXX TSN-70E.

19868 Sensor 
{95} 00 00 87 AE 33   F6 8E  FF F3 B7 73 F8

{97} 00 00 87 AE D9   DD 5B FF FC ED D7 BE 8

{99} 00 00 87 AE 33   F7 7B FF FE 76 EF BF 4

{94} 00 00 87 AE D9   D3 2E FF F3 B7 5D F0

{95} 00 00 87 AE 33   ED 49 FF E7 6E DF F4

{97} 00 00 87 AE D9   DA DB 7F F9 DB BB FD 8

{98} 00 00 87 AE 33   E7 BD FF FC ED DF BD 8

{95} 00 00 87 AE D9   DA FA FF FC ED DC 74

{95} 00 00 87 AE 33   E3 75 FF F3 B7 6F F0

20010 Sensor 91 77 ==> could be temperatur ID
20011 Sensor 92 77 ==> could be humidity ID 

{97} 00 00 87 92 77   F6 9D FF   F9 DB BB FE 0

{93} 00 00 87 91 77   B3 4B FF   E7 6E CB E8

{95} 00 00 87 91 77   AD DF 7F   F9 DB BA F8

{97} 00 00 87 92 77   ED FB 7F   FC ED DA FD 0

{95} 00 00 87 91 77   AB 7A FF   F3 B7 6D F6

{94} 00 00 87 92 77   E4 6D FF   E7 6E ED D8

{96} 00 00 87 91 77   B5 FB FF   FC ED DD BA

{95} 00 00 87 92 77   EA DF 7F   F9 DB AB F0

{95} 00 00 87 91 77   B9 FA FF   F9 DB A7 F2

{95} 00 00 87 92 77   EC CF FF   F9 DB B1 EC

19946 Sensor B8 B3 
19947 Sensor B9 33

{95} 00 00 87 B9 33   EC CE FF   F3 B7 67 F8

{94} 00 00 87 B8 B3   B3 35 FF   E7 6E DB F4

{95} 00 00 87 B8 B3   BE BB 7F   F9 DB BA F8

{97} 0 00 08 7B 93 3E 7F 1F FF 9D BB DF D0

{94} 00 00 87 B8 B3   AA B9 FF   E7 6E DB EC

{94} 00 00 87 B9 33   E1 E5 FF   E7 6E D7 D8

{95} 00 00 87 B9 33   EE 2D FF   F3 B7 75 BC 

{94} 00 00 87 B8 B3   AA B9 FF   E7 6E DB 7C

{96} 00 00 87 B9 33   ED BF 7F   FC ED D8 BE

{99} 00 00 87 B8 B3   B7 FD FF   FE 76 EF FB C 

{94} 00 00 87 B8 B3   BB 55 FF   F3 B7 5B 74

{92} 00 00 87 B9 33   EC 99 FF   E7 6E D1 A

{95} 00 00 87 B8 B3   9F 7D 7F   F9 DB BE D8

Question is: 

Did I decode it correctly with rtl_433 ? 
Does anyone has an idea how to decode it. 

[Arexx.zip](https://github.com/merbanan/rtl_433/files/11243819/Arexx.zip)

zuckschwerdt, 2023-04-17:
PWM is very unlikely with FSK. If you look close you can see that short/long gaps don't pair with short/long pulses -- there is data in the gaps. This is Manchester coding. This should work:

`rtl_433 -X 'n=AREXX,m=FSK_MC_ZEROBIT,s=208,l=208,r=450'`

But only two of the posted signals decode well. The signal is clipping (too loud).

The good codes there are
`55555555aafa15b2e90f6cfffaf77b1ce3`
`55555555aafa14b2f90e51fffaf77b1ae4`

Best to capture the codes as PCM first and inspect carefully, then later decode the MC:
`rtl_433 -X 'n=AREXX,m=FSK_PCM,s=208,l=208,r=450'`

zuckschwerdt, 2023-04-17:
Oh, I see now why some files don't work: 500k is not a valid sample rate. You should have seen:
```
Invalid sample rate: 500000 Hz
[sdr_set_sample_rate] Failed to set sample rate.
```

inonoob, 2023-04-17:
Dear Zuckschwerdt, 

many thanks for pointing in the right direction... I'm sorry still trying to figure all out. 

I will try to get more samples. Back to the drawing board :).  

Best regards

zuckschwerdt, 2023-04-17:
You can just run ` rtl_433 -X 'n=AREXX,m=FSK_PCM,s=208,l=208,r=450'` and capture the codes. For now there is no need for additional sample files.

inonoob, 2023-04-17:
Hey Zuckschwerdt, 

many thanks for you patience here. I try to dig into the topic. Might you explain why now you are using FSK_PCM and not FSK_MC_ZEROBIT any longer ? 

Best regards

inonoob, 2023-04-17:
I've tried to see the signal with urh but I can't reproduce what rtl_433 is giving as data stream.

zuckschwerdt, 2023-04-17:
> why now you are using FSK_PCM and not FSK_MC_ZEROBIT any longer ?

PCM is the raw data, we can inspect it more easily. MC is an additional step which can be done later (e.g. in https://triq.net/bitbench ) and will hide errors and make it hard to see what's really transmitted.

So: collect PCM, verify it's really plain MC, decode "by hand", then later when all is well switch do MC directly.

inonoob, 2023-04-17:
Hey, 

to be honest, I'm currently puzzled with the result: 

with FSK_PCM i got the following data: 

`cccccccccccccccd33335532d52d34ab52d2cd54d2b35555553354d4d535354b2ccc
cccccccccccccccd33335532d52b34ab5532cb3534ab5555553354d4d5352d352ccb
66666666666666669999aa9956659a59aa959a65a696aaaaaa99aa6a6a9a69aa5a598
cccccccccccccccd33335532cad5355552d2d34cd2b35555553354d4d5354d2aad2c
cccccccccccccccd333355334ccb34ad552b34d3352b5555553354d4d534d2acd52b
cccccccccccccccd33335532accd34b3532cd2d54b355555553354d4d5354b34b4ac
cccccccccccccccd333355334ccd34ad532b54d54ab55555553354d4d5355334d4d4
66666666666666669999aa9956659a59aa959a5a6699aaaaaa99aa6a6a9a69555a558
cccccccccccccccd33335532d52d34ab52d2cd54d2b35555553354d4d535354b2cb4
66666666666666669999aa9965699aaaaa996599a9a5aaaaaa99aa6a6a9a966a56958
cccccccccccccccd333355334ccb34ad552b34d3352b5555553354d4d534d2acd4d3
cccccccccccccccd33335532accd34b3532cd32b2b335555553354d4d5354d2cb354
cccccccccccccccd33335532d52b34ab5532cb4cd34d5555553354d4d5352d2b2cb3
cccccccccccccccd333355334ccd34ad532b54d32d4d5555553354d4d53552d4d4cc
cccccccccccccccd33335532cad5355552d2d3532aab5555553354d4d5354cccacd4
66666666666666669999aa9965699aaaaa996599a9a5aaaaaa99aa6a6a9a966a56698
66666666666666669999aa99a6659a56aa959a666699aaaaaa99aa6a6a9a69a56a658
66666666666666669999aa9956659a59aa959a5a6699aaaaaa99aa6a6a9a695559a98
cccccccccccccccd33335532d52d34ab52d2d2ab2d4d5555553354d4d5354b352cac
cccccccccccccccd333355334ccd34ad532b54d32d4d5555553354d4d53552d4d4b4`

Would you suggest I try to find the Id of the sensor in this data stream ? 

Best regards

inonoob, 2023-04-17:
Thanks again for your support and pointing me in the direction.

MacH-21, 2023-04-17:
Try not to be confused you are nearly there.

Check out issue #2388
I use,  
rtl_433 -R 0 -X "n=Arexx,m=FSK_MC_ZEROBIT,s=210,l=210,r=2500"
Reads Arexx RF-Data
Arexx sensors do use manchester coding, The sensors transmit approx every 45 seconds alternating Temperature/Humidity 

@zuckschwerdt found this data- 
55555555aafa15b2e90f6cfffaf77b1ce3
55555555aafa14b2f90e51fffaf77b1ae4
using bitbench invert above lines of Hex

aaaaaaaa5505ea4d16f09300050884e31c
aaaaaaaa5505eb4d06f1ae00050884e51

aaaaaaaa5505ea4d16f09300050884e31c
aaaaaaaa Preamble
55 		 Sync
05		 Length of data in bytes
ea4d	 Id of Sensor (little-endian)_4dea = 19946 - even number = Temperature
16f0	 Raw Temperature value from sensor "Sensirion SHT-10"
93       Checksum
00050884e31c The rest of data can be dumped

aaaaaaaa5505eb4d06f1ae00050884e51
aaaaaaaa Preamble
55		 Sync	
05		 Length of data in bytes
eb4d	 Id of Sensor (little-endian)_4deb = 19947 -  odd number = Humidity
06f1	 Raw Humidity value from sensor "Sensirion SHT-10"
ae		 Checksum	
00050884e51	 The rest of data can be dumped

inonoob, 2023-04-17:
So many thanks that it !!! To contribute a bit I did also find the CRC check how it done. 

They us the CRC-8/MAXIM 

the use the length of the data in bytes + the sensor id + the value. 

05 eb4d 06f1 ==> CRC-8/MAXIM results in ae.  

I will check will all sensor and see how to integrate it into rtl_433

inonoob, 2023-04-17:
many thanks to you @zuckschwerdt for pointing me in the right direction and again your patience and big thanks to @MacH-21 for finding the solution. 

You just helped me to avoid using this ugly Arexx box to extract the data.

MacH-21, 2023-04-17:
@inonoob 
In issue #2388 Christian @zuckschwerdt found checksum I was looking at all the data which is not needed, thanks again Christian.

Maybe Christian can spoon feed me how to add a protocol to rtl_433,
I cannot follow instructions on contributions section (old grey cells) 
I use the windows10 OS and can just about compile a latest rtl_433.exe

Also maybe merge #2388 & #2482

zuckschwerdt, 2023-04-17:
I'll try to implement something in the next days. For a detailed example this should do:
https://github.com/merbanan/rtl_433/wiki/Adding-a-new-remote-device-and-writing-the-decoder-C-code#how-to-add-the-decoder-and-write-new-code

MacH-21, 2023-04-17:
Thanks again,
I will check it out, but I easily lose the thread these days.

zuckschwerdt, 2023-04-20:
There is now a decoder in #2487 -- let's discuss and finish this there.

closed 2023-04-20 by zuckschwerdt