merbanan/rtl_433 #2489: Support Acurite Tower Sensor

ifuchs, 2023-04-23:
I finally got rtl_433 working with Home Assistant and 4 of my 9 sensors are working fine (those that are Acurite 609TXC. However the other 5 are Acurite Tower Sensors. I know that RTL_433 can read these sensors and show the temperature but there is a problem with the MQTT information and so HA shows the temp and humidity as Unknown. I am not precisely clear about where the problem lies but it would be great if these sensors could work with HA.

zuckschwerdt, 2023-04-23:
Did you check the topics sent for this sensor? Can you post an examples? E.g. from `mosquitto_sub -t 'rtl_433/#' -v`

rct, 2023-04-23:
The Acurite "tower sensors" (592TXR and related) have been supported in rtL_433 for quite a few years. 

Since you mention Home Assistant, are you using the add-on with auto-discovery?

You can go into device info with Home Assistant see what MQTT topics you are subscribed to for each sensors and the most recent messages. 

Note: If you've changed the channel switch since the MQTT configuration was created, the data will be posted to a different topic.

A good place to get help with the Home Assistant add-on is https://community.home-assistant.io/t/home-assistant-add-on-rtl-433-with-mqtt-auto-discovery/260665

ifuchs, 2023-04-23:
Here is what one of the sensors showing Unknown is showing in mosquitto_sub:
rtl_433/raspberrypi/events {"time":"2023-04-23 13:02:56","model":"Acurite tower sensor","id":9064,"sensor_id":9064,"channel":"A","temperature_C":22.2,"humidity":46,"battery_low":0}
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/id 9064
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/sensor_id 9064
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/channel A
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/temperature_C 22.200001
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/humidity 46
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/battery_low 0

ifuchs, 2023-04-23:
@rct I am using Auto discovery to create the devices but not the rtl_433 addin as I am sending the data from a Raspberry Pi.

ifuchs, 2023-04-23:
@rct Subscribed topics:
rtl_433/raspberrypi/devices/Acurite_tower_sensor/A/9064/temperature_C as well as the humidity. But it shows 0 recent messages even though MQTTtool shows that they are being received.

ifuchs, 2023-04-23:
Unrelated question: I am trying to send data from 2 RPis to HA at the same time. How do I specify a unique client id for each connection? (e.g. if I issue rtl_433 -F "mqtt://192.168.3.135:1883,user=homeassistant,pass=pswd,retain=1" from both RPis, the second connection will fail).

zuckschwerdt, 2023-04-23:
> How do I specify a unique client id for each connection?

The computers need a distinct hostname (i.e. not the default of `raspberrypi`) or at least a different device selection, e.g. `-d 0`

ifuchs, 2023-04-23:
Thanks, hostname problem fixed. After naking that change I no longer get connection failed but now I get:
MQTT Connected...
MQTT Connection established.
cb transfer status: 1, canceling...
WARNING: async read failed (-5).

Library error -5, exiting...
Reattaching kernel driver failed!

After this happens, the RTL is no longer accessible until the Pi is rebooted.

ifuchs, 2023-04-23:
When the 2nd Pi is connected (lasts about 60 seconds before getting the async read error), I am seeing another issue where Temperature and Humidity are unknown even though I can see the updates in MQTTtool. This time the Models are: Nexus_Temperature_Humidity and OS BHTR968

ifuchs, 2023-04-23:
Not sure if this helps, but I note that the sensors that work who up hierarchically directly under the Model Acurite_609_TXC_Sensor whereas those that do not work are 1 level lower under A and C. Id=222 also reports Unknown.
[image]

rct, 2023-04-23:
When you change the hostname for one or both RPIs you will need to change the topic you are subscribed to.   

Yes the Tower sensors (and any of the "My Acurite" compatible sensors that have a channel switch will all have a default topic path of 

`rtl_433/<hostnamee>/<model>/<channel>/<id>/#` 

If you want to be able to use which ever RPI/SDR picks up the sensor (I do this to make sure I have the whole house covered), wildcard the host portion with a '+'

`rtl_433/+/Acurite Tower/A/9064/#`

How is rtl_433 sending data to MQTT?   I was assuming you were using `-F mqtt`, but I just noticed your tower devices are showing up as 'Acurite_tower_sensor`.  The default model string is `Acurite-Tower`

If you are seeing the messages in MQTT explorer, mosquitto_sub, then it seems like it is down to your Home Assistant configuration for those sensors.   It would help if you posted the config -- The YAML if you configured manually, and 
It would help if you posted the

ifuchs, 2023-04-23:
I changed the hostname of one of the RPis, however, for some reason I am no longer able to access the stick now (No supported devices found) and even a reboot which usually fixes the problem is not working. As a result, I will turn my attention to getting the 5 sensors on the 1st RPi working. I am sending the rtl data to HA with this command:
rtl_433 -F "mqtt://192.168.3.135:1883,user=homeassistant,pass=pass,retain=1". I used the RTL_433 auto discovery add-in on HA, I did not configure them manually so I don't think that will help diagnose the problem.

zuckschwerdt, 2023-04-24:
@rct good catch with the model names. This must be an ancient `19.08` version or earlier.

ifuchs, 2023-04-24:
Version of Acurite? I am not sure where to fix this. There is rtl_433 and rtl_433 Autodiscovery addin in HA. Or is the workaround to make manual changes in the HA configuration for each of these HA devices?

rct, 2023-04-24:
@ifuchs - It looks like you have a very old version of rtl_433 running in your Home Assistant add on since you aren't getting the current model names from the past few years.

Between the old version of rtl_433, and the software not being able to find the RTL-SDR, it sounds like you are struggling with the add-on and should be seeing help from the people who produced that add-on. 

As I mentioned above:  A good place to get help with the Home Assistant add-on is https://community.home-assistant.io/t/home-assistant-add-on-rtl-433-with-mqtt-auto-discovery/260665

ifuchs, 2023-04-24:
@zuckschwerdt Not sure if you saw my comment above but I am perfectly happy upgrading rtl_433 on to RPis to fix the MQTT issue I am having if that is where the problem lies (or even if not). I guess I'll see if any of the problems persist after upgrading. *TBH, Until now, I had no idea I was running an outdated version.

ifuchs, 2023-04-24:
New version fixed the problem! Once I can power cycle the other Pi, we'll see if a new version helps with the async problem. Thanks for your help.