merbanan/rtl_433 #2492: Help Appreciated. Unknown new signal near me every 16s at 868.3MHz

Pascal66, 2023-04-25:
This is perturbing everything and have no clues, a lot of short or long
tested at triq without success
Included one of the tons received
[g001_868.5M_2048k.zip](https://github.com/merbanan/rtl_433/files/11324378/g001_868.5M_2048k.zip)

ProfBoc75, 2023-04-26:
Salut Pascal, 

try this :

```
rtl_433 -r g001_868.5M_2048k.cu8 -Y minmax -X "name=test,m=FSK_PCM,s=58,l=58,r=1000"
```

But your sample was done at 2048k (sampling rate) not the default at 250k,
There is a DC offset also that avoid to decode the signal properly.

fineoffset have such signal every ~16 s, with same modulation short=long=58 µs, but not the same value (not same preamble).

May be put some other samples at -f 868M and -s 250k here , to see .

Pascal66, 2023-04-26:
Resalut
I've this with your paramters
[triq](https://triq.org/pdv/#AAB103001800300160819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080819080A0809055
)and you're right, i've a lot fineoffset too

closed 2023-04-28 by Pascal66