merbanan/rtl_433 #2494: rtl_433 failing

ifuchs, 2023-04-27:
I have been successfully running the command below for 2 days or so when it suddenly failed with:

bitbuffer_add_bit: Warning: row count limit (50 rows) reached
cb transfer status: 5, canceling...
Reattaching kernel driver failed!

`
rtl_433 -F "mqtt://192.168.3.135:1883,user=homeassistant,pass=pass,retain=1"
`

ifuchs, 2023-04-27:
I guess this is caused by some noise? Is three a way to automatically restart when this happens?

zuckschwerdt, 2023-04-27:
> bitbuffer_add_bit: Warning: row count limit (50 rows) reached

That's just a warning and won't cause any issue. Are the fails maybe a unstable power supply? I encountered that multiple times on Raspis with SDR.

zuckschwerdt, 2023-04-27:
Oh and, yes, the current rtl_433 has a `-D restart` option.
```
  [-D restart | pause | quit | manual] Input device run mode options.
	Supported input device run modes:
	  restart: Restart the input device on errors
	  pause: Pause the input device on errors, waits for e.g. HTTP-API control
	  quit: Quit on input device errors (default)
	  manual: Don't start an input device, waits for e.g. HTTP-API control
	Without this option the default is to start the SDR and quit on errors.
```

gdt, 2023-09-26:
This seems much more likely hardware than bug, so I'm closing.

closed 2023-09-26 by gdt