merbanan/rtl_433 #2499: Strange whx080

Pascal66, 2023-05-04:
```
time      : 2023-05-04 16:00:18.549246
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 78
Battery   : 1            Temperature: -2.0 C       Humidity  : 169 %         Wind Direction: 293
Wind avg speed: 117.50   Wind gust : 19.58         Total rainfall: 1029.0    Integrity : CRC
Modulation: FSK          Freq1     : 868.3 MHz     Freq2     : 868.4 MHz
RSSI      : -10.0 dB     SNR       : 26.1 dB       Noise     : -36.1 dB
```
Currently 26°C 46% Hum 0.3 KmH Wind

gdt, 2024-06-04:
abandoned by submitter

closed 2024-06-04 by gdt

TheButterZone, 2026-09-10:
Well, I just got my first & only decode of this; unfortunately I'm never running the correct flags for debugging one-offs & I may never see it again.

> time      : 2026-09-10 13:45:36
> model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 255
> Battery   : 1            Temperature: 5.9 F        Humidity  : 255 %         Wind Direction: 0         Wind avg speed: 193.94 mi/h                         Wind gust : 193.94 mi/h
> Total rainfall: 45.4 in  Integrity : CRC