merbanan/rtl_433 #2500: Cannot receive Driveway alarm signal

ifuchs, 2023-05-09:
I'm trying to create an alert system using a Driveway alarm (https://www.amazon.com/dp/B075HCDGYW) which says on it that it runs at 433 Mhz. I ran rtl_433 on a Mac with an RTL_SDR stick and cannot seem to see this signal. Any suggestions?

zuckschwerdt, 2023-05-09:
You need to load the config file for this sensor, e.g. `rtl_433 -c conf/DrivewayAlert.conf`
Locate the conf file first though. I should be `/usr/local/etc/rtl_433/DrivewayAlert.conf` for Brew and `/opt/local/etc/rtl_433/DrivewayAlert.conf` for MacPorts. If that fails then download the file here: https://github.com/merbanan/rtl_433/tree/master/conf

closed 2023-05-09 by zuckschwerdt

ifuchs, 2023-05-09:
Thanks, that looked promising but alas, it did not work as hoped. I ran Gqrx and found that the sensor seems to be sending at 434.599Mhz (rather than 433.92)
[image]
Does the conf need to be modified or perhaps other parms to rtl_433? I am trying to use this within a python script. Wuld something this simple work:
    rtl_433_cmd = f"rtl_433 -F json -f 434599000}"

zuckschwerdt, 2023-05-09:
If the signal is at 434.6M then this will work: `rtl_433 -c /path/to/DrivewayAlert.conf -f 434.65M -F json`
Note the 50k offset to the center frequency.

ifuchs, 2023-05-10:
Thanks. I will gather some samples for analysis later today. Looking at the waterfall from Gqrx makes me wonder if this sensor is sending its signal and receiving a response from the sounder in the house. Does that make sense and if so, is what I am trying to do with rtl_433 not going to work? (My goal is to detect the signal so that the python script can send a notification).

zuckschwerdt, 2023-05-10:
Detecting a "doorbell"-style transmission is a common use-case for rtl_433, yes.
It would be very unusual for such a device to have bidirectional communication.
To post-process the received signal have a look at various example Python scripts: https://github.com/merbanan/rtl_433/tree/master/examples

ifuchs, 2023-05-10:
OK. I'll give it a shot. Do I get initial samples with rtl_433 -f 434.65M -S unknown?

ifuchs, 2023-05-10:
The receiver's label says:

Wireless Alarm Receiver
Model: 40405
ID:01230211
Frequency: 433MHz FSK+FHSS

ifuchs, 2023-05-10:
[Archive 2.zip](https://github.com/merbanan/rtl_433/files/11443540/Archive.2.zip)
These are the .cu8 files I captured using rtl_433 -f 434.65M -S unknown.  It appears to confirm it is fsk but I don't know how to use what this tells us to run rtl_433 to detect the pulse(s) in the script.

zuckschwerdt, 2023-05-10:
`g005_434.65M_250k.cu8` looks kind of okay, but you need to lower the gain. Either use more distance or remove the antenna.

ifuchs, 2023-05-10:
I was right next to the transmitter which is not where I will be receiving the alert. It's just a bit more difficult to trigger it and do the sampling without 2 people. Are you suggesting I do this again with the RTL-SDR in the house? Or do we have enough information now to use RTL_433 for my use case?

ifuchs, 2023-05-10:
I'm afraid you lost me. I'm not sure what that command is supposed to do (if I run it and trigger the alarm, nothing seems to happen) and I did not understand the reference to MC 3000/3000. Looking at it on that website shows a pretty clear pulse. I reran rtl_433 -f 434.65M -S unknown without the antenna and got this:
[g006_434.65M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/11445437/g006_434.65M_250k.cu8.zip)
I feel as if I am closer to having an rtl_433 command that will indicate when this pulse is received but TBH, I don't know what it is.

zuckschwerdt, 2023-05-10:
`MC 3000 3000` are the values to enter in the first three fields of the pdv decoder: Slicer Short/Long.
The data seems to be Manchester coded and prefixed with a desync (longer symbols).

And the command will recover the raw data:
```
rtl_433 -R 0  -X 'n=name,m=FSK_PCM,s=3000,l=3000,r=9000' g005_434.65M_250k.cu8
...
codes     : {100}e75334b4ad35554d555334aa8
...
```

Drop your latest capture on pdv and observe that no data is present -- just a "beep".

ifuchs, 2023-05-10:
Yes, I see that "beep" which I presume is the alert. Sorry, if this should be obvious but based on this what rtl_433 commandline can I therefore use in my python script that will output something when this beep is received?

ifuchs, 2023-05-10:
Or do I have to use something like rtl_fm to demod the signal and then look for the amplitude to be > some threshold. This might be less than reliable.

zuckschwerdt, 2023-05-11:
That "beep", a longer straight red line is a single tone. There is no data. It's not a proper code. No even the cheapeast doorbells transmit just a "beep" -- there is always a code like you see in `g005_434.65M_250k.cu8`.

ifuchs, 2023-05-11:
Yes, I’m sure you’re right, and this alarm system is capable of having
multiple transmitters so it must have a way of identifying the alarms which
can be associated different tones. The question is what additional data do
I need or do we have enough and if so, what command can I use to trigger
when the alarm is going off?

zuckschwerdt, 2023-05-11:
Just like you did before: `rtl_433 -f 434.65M -S unknown` with some distance or without the antenna. Try a few times and inspect the recorded samples in https://triq.org/pdv/ -- then send a good one.

ifuchs, 2023-05-11:
If rtl_433 -R 0  -X 'n=name,m=FSK_PCM,s=3000,l=3000,r=9000' g005_434.65M_250k.cu8
shows
codes     : {100}e75334b4ad35554d555334aa8
and assuming that is repeatable, is the command to use in the script merely: rtl_433 -f 434.65M -S unknown -F json and then trigger on that code?

zuckschwerdt, 2023-05-11:
You need to use `-S unknown` to get a good sample. Analyze that sample and refine the `-X` with that info.

After that is done you always use e.g. `rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=3000,l=3000,r=9000' -F json`  to get codes, then validate them and effect triggers.