merbanan/rtl_433 #2503: best approach to identifying?

great9, 2023-05-13:
What's your best approach to identifying a signal?
Here's [the link to trig.org](https://triq.org/pdv/#AAB01B0501003000680108003C271480808080808080808080808080809155+AAB0150501003000680108003C271480809080808080808155+AAB0130501003000680108003C27148080808090808155+AAB00F0501003000680108003C271480809155+AAB0100501003000680108003C27149080808155+AAB0100501003000680108003C27148080809155+AAB0130501003000680108003C27148080808080908155+AAB0140501003000680108003C2714808080808080809155+AAB0100501003000680108003C27149080808155+AAB00D0502003000680108003C27149155+AAB00F0502003000680108003C271480809155+AAB00D0501003000680108003C27149155+AAB00E0501003000680108003C2714908155+AAB00E0501003000680108003C2714809155+AAB00E0502003000680108003C2714839155+AAB00D0501003000680108003C2714A455) for reference

zuckschwerdt, 2023-05-13:
Maybe the Introduction to pulse formats can help: https://triq.org/rtl_433/PULSE_FORMATS.html

Summarize what kind of pulses and gaps you see, and in what relation (e.g. are certain pulses always followed by certain gaps? Or are pulses and gaps not dependant at all?).

Here you will conclude that it's MC at 50 µs half-bit width, i.e. 10kBit/s

gdt, 2023-09-26:
question answered

closed 2023-09-26 by gdt