merbanan/rtl_433 #2507: Puzzling intermittency

ifuchs, 2023-05-21:
My friend got a driveway alert which sends a 433.655 Mhz signal into his house and sets off a sounder. He wanted a way to get these alerts on his phone so I wrote a python script for him that receives this signal, looks for the code that is unique to the xmtr, and sends him a Pushover alert. Everything works...until it doesn't. I have the script logging to the console (and a file) and sometimes someone comes in his driveway, the sounder goes off and there is nothing on the console (an obviously no Pushover). This particular driveway alert is supposed to operate over a mile distance and it has line of sight to the RTL about 150 ft away. He had the RTL-SDR plugged into his Mac and I thought perhaps it was a power issue, so we added a pwered hub with a 2.5A supply. I also swapped the RTL- with another working one just to be sure it wasn't the card. 
Here is the part of the script which calls rtl_433:
`
# Run the rtl_433 command and capture the output
p = subprocess.Popen(['rtl_433', '-R', '0', '-X', 'n=name,m=FSK_PCM,s=3000,l=3000,r=9000', '-f', '433655000', '-F', 'json'], stdout=subprocess.PIPE)
`
and here is the console output from this afternoon. This shows receiving the signal (Pushover was sent as expected). The beam was tripped twice since then and the sounder went off but there was nothing more in the log.

`
@Jays-iMac-2 Documents % ./Driveway_alert_logging.py
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/jayfriedman/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.655MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 3, 'data': '0'}], 'codes': ['{3}0']}
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 8, 'data': 'f8'}], 'codes': ['{8}f8']}
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 100, 'data': 'e75334b4ad35554d555334aa8'}], 'codes': ['{100}e75334b4ad35554d555334aa8']}
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 8, 'data': 'f8'}], 'codes': ['{8}f8']}
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 100, 'data': 'e75334b4ad35554d555334aa8'}], 'codes': ['{100}e75334b4ad35554d555334aa8']}
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 8, 'data': 'f8'}], 'codes': ['{8}f8']}
{'time': '2023-05-21 15:06:43', 'model': 'name', 'count': 1, 'num_rows': 1, 'rows': [{'len': 100, 'data': 'e75334b4ad35554d555334aa8'}], 'codes': ['{100}e75334b4ad35554d555334aa8']}
`
I can post the full script if that would help.

zuckschwerdt, 2023-05-22:
- Do not plug the receiver directly in a USB port, avoid noise and use a short USB cable.
- Watch the levels with `-M level` to know if there too much or few gain. Perhaps try a fixed gain if the auto-gain gets bad results. Also use `-Y autolevel` to adjust to the input.
- The `level` option will also show the estimated signal frequency. Check that it is well in band but not exactly centered.
- Watch the noise with `-M noise` to know if there are problems like jamming.

ifuchs, 2023-05-22:
I will try your suggestions tomorrow, but do you think that the level adjustments or the freq being slightly off could explain the behavior we are seeing where it works fine and suddenly stops working?

zuckschwerdt, 2023-05-22:
Yes, wrong levels or frequency will suddenly drop messages.
With the `noise` output your script can easily test if `rtl_433` is still alive and working.

ifuchs, 2023-05-24:
The script is still running as I continue to get the noise levels. When it works the rtl_433 command shows this:

rtl_433 -R 0 -X 'n=name,m=FSK_PCM,s=3000,l=3000,r=9000' -f 434.6M -F json
rtl_433 version 22.11 branch  at 202211191645 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/ihf/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 223 device decoding protocols [ ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 434.600MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 3, "data" : "0"}], "codes" : ["{3}0"]}
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 8, "data" : "f8"}], "codes" : ["{8}f8"]}
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 100, "data" : "e75334b4ad35554d555334aa8"}], "codes" : ["{100}e75334b4ad35554d555334aa8"]}
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 8, "data" : "f8"}], "codes" : ["{8}f8"]}
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 100, "data" : "e75334b4ad35554d555334aa8"}], "codes" : ["{100}e75334b4ad35554d555334aa8"]}
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 8, "data" : "f8"}], "codes" : ["{8}f8"]}
{"time" : "2023-05-23 16:22:26", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 100, "data" : "e75334b4ad35554d555334aa8"}], "codes" : ["{100}e75334b4ad35554d555334aa8"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 3, "data" : "0"}], "codes" : ["{3}0"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 8, "data" : "f8"}], "codes" : ["{8}f8"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 100, "data" : "e75334b4ad35554d555334aa8"}], "codes" : ["{100}e75334b4ad35554d555334aa8"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 8, "data" : "f8"}], "codes" : ["{8}f8"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 100, "data" : "e75334b4ad35554d555334aa8"}], "codes" : ["{100}e75334b4ad35554d555334aa8"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 8, "data" : "f8"}], "codes" : ["{8}f8"]}
{"time" : "2023-05-23 16:22:48", "model" : "name", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 100, "data" : "e75334b4ad35554d555334aa8"}], "codes" : ["{100}e75334b4ad35554d555334aa8"]}

My script triggers on the codes" : ["{100}e75334b4ad35554d555334aa8"]}. After working several times today, it no longer shows this output..only the noise levels which I did not include here. Sometimes it begins working again and then stops.

If I run gqrx, the waterfall looks like this when the alarm is sent. I do not understand the 3 lines as I would have expected just one transmission but the sounder that goes off seems to be coincident with the uppermost line. What is so puzzling is that it works a few times and then stops showing the transmission. The specific device is: https://www.amazon.com/dp/B075HCDGYW
[image]

Can someone make sense of all of this? My friend does not want to hear the sounder every time the beam is broken so the purpose of the script is to send a pushover to his phone rather than having the sounder. Any help would be much appreciated.

gdt, 2023-09-26:
Is this still a valid issue?   Could the submitter summarize where we are and what is to be done?

ifuchs, 2023-09-26:
My friend decided to return the driveway alert so I will close this issue.

closed 2023-09-26 by ifuchs