merbanan/rtl_433 #2509: Old decode versus New decode

8um1c4, 2023-05-23:
When decoding signals with -A option of I found a difference between an old and a new version of rtl_433.

Old version:

RSSI: -2.1 dB SNR: 18.6 dB Noise: -20.7 dB
Frequency offsets [F1, F2]:   -1858,      0	(-7.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB01308010194032C0550103816C00B6C021C3FC48355+AAB01608010194032C0550103816C00B6C021C3FC49080808455+AAB01808010194032C0550103816C00B6C021C3FC4908291A2829555+AAB01408010194032C0550103816C00B6C021C3FC4868555+AAB01608010194032C0550103816C00B6C021C3FC49690928755
Attempting demodulation... short_width: 384, long_width: 804, reset_limit: 5828, sync_width: 1632
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=384,l=804,r=5828,g=0,t=0,y=1632'
_**pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] { 8} ba        : 10111010 
[01] { 8} b1        : 10110001**_ 

New version:
Level estimates [high, low]:  15340,     66
RSSI: -0.3 dB SNR: 23.7 dB Noise: -23.9 dB
Frequency offsets [F1, F2]:   -1990,      0	(-7.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
view at https://triq.org/pdv/#AAB01308010194032C0550103416C00B6C02203FC48355+AAB01608010194032C0550103416C00B6C02203FC49080808455+AAB01808010194032C0550103416C00B6C02203FC4908291A2829555+AAB01408010194032C0550103416C00B6C02203FC4868555+AAB01608010194032C0550103416C00B6C02203FC49690928755
Attempting demodulation... short_width: 384, long_width: 804, reset_limit: 5828, sync_width: 1632
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=384,l=804,r=5828,g=0,t=0,y=1632'

The old version has some bit information.
Is there any way of seeing this information in the new version of rtl_433?

Many thanks.

8um1c4, 2023-05-23:
Another example:

OLD:

RSSI: -0.1 dB SNR: 21.3 dB Noise: -21.4 dB
Frequency offsets [F1, F2]:    4674,      0	(+17.8 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB0160401019003200F70271480808080808080808080808255+AAB04C0401019003200F70271481908190908190908181908190818181818190819081909081909090908190819081908181818181909090818181908190909090909090909090909090908190908355
Attempting demodulation... short_width: 376, long_width: 780, reset_limit: 3956, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=376,l=780,r=3956,g=840,t=162,y=0'
**_pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2 
[00] {12} ff f0                      : 11111111 1111
[01] {66} a4 d7 d4 85 5f 1d 00 02 40_** 

NEW:

RSSI: -4.1 dB SNR: 20.9 dB Noise: -25.0 dB
Frequency offsets [F1, F2]:    4907,      0	(+18.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
view at https://triq.org/pdv/#AAB01604010194031C0F6C271480808080808080808080808255+AAB04C04010194031C0F6C271481908190908190908181908190818181818190819081909081909090908190819081908181818181909090818181908190909090909090909090909090908190908355
Attempting demodulation... short_width: 376, long_width: 772, reset_limit: 3952, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=376,l=772,r=3952,g=856,t=158,y=0'

ProfBoc75, 2023-05-23:
Hi, 

Try this :
>rtl_433 -A -a 42

8um1c4, 2023-05-23:
As I understand -a is used for testing only. Can it be used long time and with lots of protocols? Or is it unsafe?
Is "42" the only option or are there more?

Many thanks for your help.

merbanan, 2023-05-23:
-a is deprecated, we don't give support on using that option. Feel free to use it as you see fit.

zuckschwerdt, 2023-05-23:
With a quick test this really seems to be a bug. The "Attempting demodulation..." should be followed with the actual data.
We recently changed the log output, I'll take a closer look now.

zuckschwerdt, 2023-05-24:
The log output has been added back to the analyzer now. It needs an overhaul later but again works.

commit ac36025a (2023-05-24): minor: Add log output to pulse analyzer (closes #2509)
closed 2023-05-24 by zuckschwerdt