merbanan/rtl_433 #2512: Suggestion: add dewpoint calculation for sensors support temperature and relative humidity

sheilbronn, 2023-05-26:
Hello, 
I've been implementing a [dewpoint ](https://en.wikipedia.org/wiki/Dew_point)calculation (approximation) in my [rtl_433 wrapper script](https://github.com/sheilbronn/rtl2mqtt).
It calculates the dewpoint temperature according to the Antoine approximation which works quite well across a broad range of temperature and humidity:
```
                # Antoine approximation (as AWK script)
                es = 6.112 * exp((17.67 * temp) / (temp + 243.5))
                ea = hum/100 * es
                gamma = log(ea/6.112)
                dewpointtemp = (243.5 * gamma) / (17.67 - gamma)
```

**However, I would like to suggest some command-line parameter for rtl_433 (e.g. `-M dewpoint`) that outputs this dewpoint temperature calculation for any sensor reading including temperature and humidity**, e.g. (as MQTT):

```
rtl/433/Nexus-TH/1 {battery_ok:0, temperature:15, humidity:24, dewpoint:-5}
rtl/433/Bresser-3CH/2 {battery_ok:1, temperature:14, humidity:68, dewpoint:8}
rtl/433/Nexus-TH/3 {battery_ok:1, temperature:22.5, humidity:60, dewpoint:14.5}
```
Rationale: It would be more efficient to do this once within the C code of rtl_433, rather having to do this programmatically for each home automation software... 

What do you think about this?

References:
* A dewpoint calculator: https://www.calculator.net/dew-point-calculator.html
* The wikipedia entry lists other approximations: https://en.wikipedia.org/wiki/Dew_point

zuckschwerdt, 2023-05-26:
It would seem a good addition to allow some post-processing, similar to the SI/customary conversion.
We can't do it currently and it's not likely to be added as a default, but a future feature to "enrich" data would be nice!

sheilbronn, 2023-05-26:
Thanks for the fast reply!
What would be the reason that this cannot be done at the moment?
Wouldn't a preprocessor directive help?
(Sorry if this question has an abvous answer.)

zuckschwerdt, 2023-05-26:
It's not a technical problem. The project needs to do-one-thing-well -- and we currently focus on translating radio messages as-is and then rely on bridges and integration scripts to add extras like e.g. rain counter derivations.

If we figure out how to split the core functionality from the nice-to-have extras not everybody wants or may want different, hopefully in a flexible and configurable way, then we'll add all these interesting things.

gdt, 2023-05-26:
It might make sense to add this, as well as suppressing of duplicates, in rtl_433_mqtt_relay.py

sheilbronn, 2023-05-27:
> It might make sense to add this, as well as suppressing of duplicates, in rtl_433_mqtt_relay.py

Yes, this seems like a good idea, too!
One then also might want to add `rtl_433_mqtt_relay.py` to the other Items mentioned in the wiki "[Built with rtl_433](https://github.com/merbanan/rtl_433/wiki/Built-with-rtl_433%21)" since it also provides helpful features regarding MQTT integration into a smart home environment (although I'm not sure whether this would stretch the scope of this Wiki page). One could then prefer the much lighter `rtl_433_mqtt_relay.py` over my rather heavy-weight bash daemon rtl2mqtt that already has dewpoint calculation and duplicate suppression.

Thanks @zuckschwerdt for explaining the rationale above, I totally agree with you now.

gdt, 2023-09-26:
Part of #2635

closed 2023-09-26 by gdt