merbanan/rtl_433 #2515: How calculated Nexus protocol negative temperature?

meloman27, 2023-05-29:
Hi everyone, 
In the sources of the nexus protocol, it is indicated that the temperature is calculated as temperature / 10. This works for positive temperature, but how is negative temperature calculated? This is not in the source code, but the compiled rtl_433 shows the correct negative temperature. Thank you.

zuckschwerdt, 2023-05-29:
The temperature field is already a signed value. It's 12 bit 2's complement.

closed 2023-05-29 by zuckschwerdt

meloman27, 2023-05-29:
> The temperature field is already a signed value. It's 12 bit 2's complement.

Please decode this payload if formula temperature / 10
(model     : Nexus-TH     House Code: 221
Channel   : 1            Battery   : 1             Temperature: -2.20 C      Humidity  : 28 %

[01] {36} dd 8f ea f1 c0 : 11011101 10001111 11101010 11110001 1100

zuckschwerdt, 2023-05-29:
```
dd 8f ea f1 c
    ^ ^^
```
0xfea (`1111 1111 1111 1111 1110 1010`) is -22

You will need to know how numbers are represented in a (modern) computer. To get started read about 2's complement.

meloman27, 2023-05-29:
Thanks for the answer, I really don't understand everything, that's why I asked the question. I am making a bash script to decode a temperature sensor, but not everything can be implemented. A positive temperature is normally extracted, but a negative one is not.