merbanan/rtl_433 #2516: RadioLib & Rtl_433

Pascal66, 2023-06-01:
Is someone have succesfully send a packet in FSK, with an HeltecLora or else with SX127x, wich can be decoded by rtl_433 ?

I can send a good packet, wich I can handle with a Flex decoder, but not by the concerned decoder (doesnt matter wich decoder)

zuckschwerdt, 2023-06-02:
A flex decoder uses the same mechanism as the built-in decoders. Is it a checksum mismatch? Try to max verbosity with `-vvvv` or better for a specific decoder with `-R 42:vvvv`

closed 2023-06-24 by Pascal66

netspoon, 2024-09-25:
@Pascal66 - I realise this is a closed ticket - but could you share example code to send an FSK packet using a HeltecLora board ?   I am trying something similar with no success (or enough knowledge)...