merbanan/rtl_433 #2519: -E quit option does not output to MQTT

bdf0506, 2023-06-06:
When running the program with the `-E quit` option, it will listen for one event and then quit as expected. However, if configured to output to MQTT, it will NOT output to MQTT before it quits.

If you do the `-E hop` option, it posts to MQTT as expected.

zuckschwerdt, 2023-06-06:
Yes, this is a known problem. The main program exit does not care about pending network events. We are on the way to fix that -- likely it is already fixed with current master? Which version did you test?

gdt, 2023-09-26:
Needs retest with current master.   7-day timer because that was requested in June and hasn't happened, and the issue is believed fixed.

gdt, 2024-06-04:
submitter has abandoned the issue.

closed 2024-06-04 by gdt