merbanan/rtl_433 #2520: rtl_433 does not show sensors anymore

timoon, 2023-06-06:
I have a RTL SDR with R820T tuner running on a Raspberry PI 2 since a long time and with rtl_433 it receives data from my 5 temperature sensors (1 LaCrosse-TX and 4 Ambientweather-F007TH).

Very recently I bought another RTL SDR, it has a Fitipower FC0012 tuner. I plugged it in my computer and with rtl_433 I was able receive signals from all my temperature sensors listed above. But this only worked for a few times. Firstly the device stopped receiving after restarting rtl_433, now I am not receiving any signals on rtl_433 no matter how many times I do unplugging and restarting. On the Raspberry PI 2 it still works like a charm. And I am still able to see the signals with the new device on gqrx:
[image]

I also tried tuning in on a FM radio station with the new device and it worked quite well.

The output of the "old" device and the working rtl_433 looks like this:
```
#rtl_433 -F csv > /tmp/temp_3.csv
rtl_433 version 21.12-93-g8228f0d4 branch master at 202204010924 inputs file rtl_tcp RTL-SDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/alarm/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 185 out of 217 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```
And on the new device:
```
#rtl_433 
rtl_433 version -128-NOTFOUND branch master at 202211192216 inputs file rtl_tcp RTL-SDR SoapySDR with TLS
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/timotius/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 191 out of 223 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-197 199 201-215 217-223 ]
Found Fitipower FC0012 tuner
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 25000 Hz, 40.0 us
```
I have attached a sample of 60s as mentioned in other threads here, I gzipped and uploaded it here.
[rogue_auto.cu8.gz](https://github.com/merbanan/rtl_433/files/11659761/rogue_auto.cu8.gz)

Looking forward to get some helpful feedback. Thanks in advance :-) Tim

zuckschwerdt, 2023-06-06:
IIRC the Fitipower FC0012 has problems with auto-gain. Search for mentions of that or try manual gain.

timoon, 2023-06-07:
Thanks @zuckschwerdt , I tried with -g 25 and received a lot of devices (from the neighborhood) that I have never seen before, but the sensors in my house are still not received well, only 1/5 (the one at greatest distance and lowest battery state) is seen.

After restarting rtl_433 with the same -g 25 there was no reception at all. Seems to me like a hardware defect.

zuckschwerdt, 2023-06-07:
Too much gain can distort nearby signals. So the first observation is normal. Don't know why the gain is not reproducible.
You can fix some of the gain problems using `-Y autolevel` to optimize the accepted signals.

timoon, 2023-07-05:
Thanks for your support.
I bought another TV stick and identified a faulty antenna as cause for the disappearing signals.

closed 2023-07-05 by timoon