merbanan/rtl_433 #2521: r900: rtl_433 vs rtlamr

robbat2, 2023-06-06:
Hi,

I see that rtl_433 recently got R900 support, and I'd love to convert over from RTLAMR, because RTLAMR pegs 1 cpu core at 100%, even on powerful systems (see https://github.com/bemasher/rtlamr/issues/276).

I'm fortunate enough to have multiple SDR units, so I was able to try and do side-by-side comparisons, and 
This was mentioned in #2180 and I have tried both `-Y auto` & `-Y classic` but rtl_433 sees significantly fewer R900 meters than rltlamr does.

This two identical Nooelec, NESDR SMArt v5 units, directly connected to a repurposed Wyse thin terminal. Identical antenna which are placed 2ft apart. The antenna are on top of my roof (30ft up), and I live in a semi-rural area.

```
# bash  /root/compare-meters.sh 
=== RTLAMR
xxxx369xx 1
xxxx101xx 7
xxxx568xx 2
xxxx629xx 6
xxxx738xx 6
xxxx359xx 1
xxxx832xx 6
xxxx876xx 8
xxxx986xx 13
xxxx010xx 6
xxxx093xx 5
xxxx154xx 1
xxxx663xx 6
xxxx838xx 4
xxxx839xx 3
xxxx082xx 1
xxxx085xx 1
xxxx110xx 10
xxxx115xx 5
xxxx530xx 1
xxxx724xx 3
xxxx917xx 3
xxxx145xx 1
xxxx849xx 1
xxxx927xx 2
xxxx381xx 1
xxxx444xx 3
xxxx393xx 4
xxxx189xx 10
xxxx453xx 10
xxxx699xx 7
xxxx729xx 8
xxxx000xx 8
xxxx009xx 7
xxxx038xx 11
999999 RTLAMR Distinct: Meters 35 Msgs 172
=== RTL_433
 1
xxxx101xx 1
xxxx568xx 4
xxxx738xx 5
xxxx359xx 1
xxxx832xx 8
xxxx876xx 10
xxxx986xx 21
xxxx010xx 2
xxxx093xx 1
xxxx663xx 2
xxxx838xx 3
xxxx839xx 7
xxxx110xx 1
xxxx115xx 1
xxxx724xx 7
xxxx145xx 1
xxxx444xx 1
xxxx393xx 5
xxxx189xx 11
xxxx453xx 1
xxxx699xx 12
xxxx752xx 1
xxxx009xx 11
xxxx038xx 17
999999 RTL_433 Distinct: Meters 25 Msgs 135
```

robbat2, 2023-06-06:
Here's the comparison script:
```
#!/bin/bash
PREFIX=(redacted)
echo "=== RTLAMR"
grep -h -e '^{.*}$' /var/log/rtlamr/rtl*.20230606{00..04}.{log,json} 2>/dev/null |jq -c . |sort | uniq |jq '.Message|[.ID, .Consumption]|@csv' -r | sort | uniq | awk -F, '{msg[$1]++; msg_total++} END{for(meter in msg){num_meters++; print meter, msg[meter]}; printf "999999 RTLAMR Distinct: Meters %d Msgs %d\n", num_meters, msg_total}' |sort  |sed -r "s,^${PREFIX}([0-9]{3})[0-9]{2},xxxx\\1xx,g"
echo "=== RTL_433"
grep -h -e '^{.*}$' /var/log/rtl_433/rtl*.20230606{00..04}.{log,json} 2>/dev/null |jq -c . |sort | uniq |jq '[.id, .consumption]|@csv' -r | sort | uniq | awk -F, '{msg[$1]++; msg_total++} END{for(meter in msg){num_meters++; print meter, msg[meter]}; printf "999999 RTL_433 Distinct: Meters %d Msgs %d\n", num_meters, msg_total}' | sort  |sed -r "s,^${PREFIX}([0-9]{3})[0-9]{2},xxxx\\1xx,g"
```

robbat2, 2023-06-07:
Full day of data:
```
=== RTLAMR
xxxx575xx 1
xxxx994xx 1
xxxx369xx 2
xxxx101xx 29
xxxx568xx 16
xxxx629xx 17
xxxx738xx 53
xxxx359xx 10
xxxx832xx 27
xxxx876xx 45
xxxx986xx 71
xxxx010xx 17
xxxx051xx 2
xxxx093xx 26
xxxx126xx 1
xxxx154xx 1
xxxx663xx 27
xxxx805xx 2
xxxx815xx 1
xxxx838xx 21
xxxx839xx 1
xxxx839xx 26
xxxx082xx 2
xxxx085xx 1
xxxx110xx 30
xxxx115xx 52
xxxx939xx 1
xxxx166xx 4
xxxx530xx 1
xxxx724xx 28
xxxx726xx 1
xxxx917xx 10
xxxx145xx 2
xxxx849xx 3
xxxx927xx 4
xxxx964xx 2
xxxx967xx 1
xxxx436xx 1
xxxx381xx 1
xxxx444xx 12
xxxx393xx 31
xxxx189xx 36
xxxx453xx 57
xxxx699xx 40
xxxx729xx 47
xxxx752xx 12
xxxx000xx 42
xxxx009xx 24
xxxx038xx 93
xxxx975xx 1
xxxx196xx 1
xxxx486xx 2
999999 RTLAMR Distinct: Meters 52 Msgs 939
=== RTL_433
 1
xxxx509xx 1
xxxx101xx 3
xxxx568xx 16
xxxx738xx 28
xxxx359xx 10
xxxx832xx 27
xxxx876xx 39
xxxx986xx 71
xxxx010xx 3
xxxx093xx 8
xxxx663xx 4
xxxx838xx 10
xxxx839xx 1
xxxx839xx 30
xxxx110xx 2
xxxx115xx 8
xxxx724xx 28
xxxx145xx 2
xxxx444xx 2
xxxx393xx 20
xxxx189xx 36
xxxx453xx 3
xxxx699xx 40
xxxx729xx 11
xxxx752xx 2
xxxx009xx 24
xxxx038xx 93
999999 RTL_433 Distinct: Meters 28 Msgs 523
```

robbat2, 2023-06-07:
One thing i'm finding interesting here is that there are two water meters that RTL_433 picks up, but RTLAMR doesn't.  I checked 6 months of data and RTLAMR never picked them up once.

zuckschwerdt, 2023-06-07:
Very interesting comparison, thanks! Did you also use `-Y autolevel` to adapt the input thresholds? And perhaps `-M level -M noise` to observe the band?

robbat2, 2023-06-07:
Yes, I've used `-Y autolevel`.

my normal logging config:
```
report_meta noise:60
report_meta stats:1:600
report_meta time:iso:usec
report_meta protocol
```

Is there a way to get JSON format of the level & noise data (with timestamps etc).
I do have the autolevel non-json output, here's a quick statistical analysis on it, same day as above:
```
$ ... |grep  -E -e '^Auto Level: Estimated noise level is -[0-9.]+ dB, adjusting minimum detection level to -[0-9.]+ dB$'   |sort |uniq -c
      1 Auto Level: Estimated noise level is -27.8 dB, adjusting minimum detection level to -24.8 dB
      2 Auto Level: Estimated noise level is -27.9 dB, adjusting minimum detection level to -24.9 dB
      1 Auto Level: Estimated noise level is -28.2 dB, adjusting minimum detection level to -25.2 dB
      2 Auto Level: Estimated noise level is -28.3 dB, adjusting minimum detection level to -25.3 dB
      6 Auto Level: Estimated noise level is -28.4 dB, adjusting minimum detection level to -25.4 dB
      5 Auto Level: Estimated noise level is -28.5 dB, adjusting minimum detection level to -25.5 dB
     16 Auto Level: Estimated noise level is -28.6 dB, adjusting minimum detection level to -25.6 dB
     13 Auto Level: Estimated noise level is -28.7 dB, adjusting minimum detection level to -25.7 dB
     21 Auto Level: Estimated noise level is -28.8 dB, adjusting minimum detection level to -25.8 dB
     26 Auto Level: Estimated noise level is -28.9 dB, adjusting minimum detection level to -25.9 dB
     25 Auto Level: Estimated noise level is -29.0 dB, adjusting minimum detection level to -26.0 dB
     34 Auto Level: Estimated noise level is -29.1 dB, adjusting minimum detection level to -26.1 dB
     33 Auto Level: Estimated noise level is -29.2 dB, adjusting minimum detection level to -26.2 dB
     43 Auto Level: Estimated noise level is -29.3 dB, adjusting minimum detection level to -26.3 dB
     40 Auto Level: Estimated noise level is -29.4 dB, adjusting minimum detection level to -26.4 dB
     68 Auto Level: Estimated noise level is -29.5 dB, adjusting minimum detection level to -26.5 dB
     91 Auto Level: Estimated noise level is -29.6 dB, adjusting minimum detection level to -26.6 dB
     91 Auto Level: Estimated noise level is -29.7 dB, adjusting minimum detection level to -26.7 dB
     95 Auto Level: Estimated noise level is -29.8 dB, adjusting minimum detection level to -26.8 dB
     90 Auto Level: Estimated noise level is -29.9 dB, adjusting minimum detection level to -26.9 dB
     71 Auto Level: Estimated noise level is -30.0 dB, adjusting minimum detection level to -27.0 dB
     65 Auto Level: Estimated noise level is -30.1 dB, adjusting minimum detection level to -27.1 dB
     62 Auto Level: Estimated noise level is -30.2 dB, adjusting minimum detection level to -27.2 dB
     57 Auto Level: Estimated noise level is -30.3 dB, adjusting minimum detection level to -27.3 dB
     44 Auto Level: Estimated noise level is -30.4 dB, adjusting minimum detection level to -27.4 dB
     60 Auto Level: Estimated noise level is -30.5 dB, adjusting minimum detection level to -27.5 dB
     86 Auto Level: Estimated noise level is -30.6 dB, adjusting minimum detection level to -27.6 dB
     98 Auto Level: Estimated noise level is -30.7 dB, adjusting minimum detection level to -27.7 dB
    103 Auto Level: Estimated noise level is -30.8 dB, adjusting minimum detection level to -27.8 dB
     97 Auto Level: Estimated noise level is -30.9 dB, adjusting minimum detection level to -27.9 dB
     81 Auto Level: Estimated noise level is -31.0 dB, adjusting minimum detection level to -28.0 dB
     72 Auto Level: Estimated noise level is -31.1 dB, adjusting minimum detection level to -28.1 dB
     65 Auto Level: Estimated noise level is -31.2 dB, adjusting minimum detection level to -28.2 dB
     52 Auto Level: Estimated noise level is -31.3 dB, adjusting minimum detection level to -28.3 dB
     28 Auto Level: Estimated noise level is -31.4 dB, adjusting minimum detection level to -28.4 dB
     12 Auto Level: Estimated noise level is -31.5 dB, adjusting minimum detection level to -28.5 dB
      8 Auto Level: Estimated noise level is -31.6 dB, adjusting minimum detection level to -28.6 dB
      5 Auto Level: Estimated noise level is -31.7 dB, adjusting minimum detection level to -28.7 dB
      3 Auto Level: Estimated noise level is -31.8 dB, adjusting minimum detection level to -28.8 dB
      2 Auto Level: Estimated noise level is -31.9 dB, adjusting minimum detection level to -28.9 dB

```

gdt, 2023-09-26:
I think this is a question that has run its course.  If so, please close.    Otherwise, please explain and phrase as a defect report or enhancement request (or perhaps in a new issue).

mattfox27, 2023-12-01:
Did you ever get this working?  I'm trying to setup 2 radios as well to monitor 433 door sensors and a Neptune water meter.  Any tips?

dimatx, 2024-01-12:
My Neptune water meter works.

closed 2024-01-12 by zuckschwerdt