merbanan/rtl_433 #2522: Broken [196] Atech-WS308 protocol after release 21.12

jamaisx, 2023-06-06:
Used that protocol to read a Lidl 4-LD1558 hygro/temp sensor. Worked fine until last 21.12 release, with 22.x is not detected anymore. Checked sources but I' can't see any relevant change, perhaps some refactoring broke protocol?
 Let me know ho I can contribute to solve the issue.

zuckschwerdt, 2023-06-06:
History for context: https://github.com/merbanan/rtl_433/commits/master/src/devices/atech_ws308.c
There are no relevant changes since initial commit. I have one sample and it still works well.

Can you grab a sample and upload as zip here? https://triq.org/rtl_433/ANALYZE.html

zuckschwerdt, 2023-06-06:
Try also `-M level -M noise` to see the frequency and noise estimates. Needing `-s 1024k` indicates that `-f 433.92M` might not be optimal.