merbanan/rtl_433 #2523: Bresser 6-in-1 - wrong negative temperatures with Bresser 3-in-1 Professional Wind Gauge / Anemometer, PN 7002531

matthias-bs, 2023-06-06:
The Bresser 3-in-1 Professional Wind Gauge / Anemometer, PN 7002531 uses a different temperature encoding for negative values than the other sensors using the 6-in-1 protocol.

Please see here for a (rather lengthy) discussion: https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/42#issuecomment-1471544273

The outcome was the following workaround:
```
    // Look up the ID in an array
    f_3in1 = is_decode3in1(id_tmp);

    // temperature, humidity(, uv) - shared with rain counter
    temp_ok = humidity_ok = (flags == 0);
    if (temp_ok) {
        bool  sign     = (msg[13] >> 3) & 1;
        int   temp_raw = (msg[12] >> 4) * 100 + (msg[12] & 0x0f) * 10 + (msg[13] >> 4);
        float temp;
        
        // Workaround for 3-in-1 Professional Wind Gauge / Anemometer 
        if (f_3in1) {
            temp     = ((sign) ? -temp_raw : temp_raw) * 0.1f;
        } else {
            temp     = ((sign) ? (temp_raw - 1000) : temp_raw) * 0.1f;
        }
```

I did not find a way to distinguish the "Bresser 3-in-1 Professional Wind Gauge" from other sensors automatically.

zuckschwerdt, 2023-06-06:
Thanks! We have a precedent for such overlap: The Agimex Rosenborg 66796 collides with Fine Offset Electronics WH5 and has a different temperature calculation (actually the WH5 is different then all the others)
https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c

The solution is not great, one needs to manually setup the deocder for one or the other. Something more obvious would be preferred.

gdt, 2023-09-26:
@matthias-bs Are you able to figure out how to do this and open a PR?

ProfBoc75, 2023-09-27:
Hi @matthias-bs : Into the current device [bresser_6in1.c](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_6in1.c) there are already conditional formatting based on the presence or absence of : rain, temp, wind, moisture and uv, this could be extend to define the model.

Since it's a 3in1 model, you should have only wind speed, wind gust, temp and humidity (what I understood from [Bresser pdf](https://www.bresser.de/out/media/d178c89d89ce78110b0daf7c45d4e08d.pdf)) , this means that uv, wind direction and rain never report data (always 0xFF for example) and could be used to define the model, and we'll add a new conditional formatting which will calculate properly the temp accordingly.

Are you able to capture rf signal and share it here to analyze it and confirm my assumption ? (let it run few minutes and report here the result)
```
rtl_433 -R 172:v -f 868M
```

Thx.

gdt, 2023-09-28:
I don't think we want to merge support for configuring ids for what device is what type, but you are welcome to submit a PR for discussion.  It would of course have to be general and configurable.   I think someone will need to figure out how to tell these apart and adjust the decoder.   Realistically that is almost certainly someone who has one and is having  a problem -- as you can see from the issue list there is vastly more work to be done than the people who are regularly contributing.

matthias-bs, 2023-09-28:
Hi all, unfortunately I have neither this "Bresser 3-in-1 Professional Wind Gauge" nor a SDR receiver. I just tried to pass my insights from [matthias-bs/BresserWeatherSensorReceiver/issues/42#issuecomment-1471544273]( https://github.com/matthias-bs/BresserWeatherSensorReceiver/issues/42#issuecomment-1471544273) back to this project.

gdt, 2023-09-29:
Thanks for the report and for clarifying.

matthias-bs, 2023-11-29:
Here is an automatic solution:

```
temp = ((sign) ? (temp_raw - 1000) : temp_raw) * 0.1f;

// Correction for Bresser 3-in-1 Professional Wind Gauge, PN 7002531
if (temp < -50.0) {
    temp = -(100.0 + temp);
}
```

zuckschwerdt, 2023-11-29:
In the condition I would use 
```
            temp     = ((sign) ? -temp_raw : temp_raw) * 0.1f;
```
but yes, that could work as we reasonably expect to never get a value below -50 C.

matthias-bs, 2023-11-29:
Or even simpler:
```
temp = ((sign) ? (temp_raw - 1000) : temp_raw) * 0.1f;

// Correction for Bresser 3-in-1 Professional Wind Gauge, PN 7002531
if (temp < -50.0) {
    temp = -temp_raw* 0.1f;
}
```
BTW: As far as specified in the manuals, the lower limit of the temperature range is -40°C.

commit e0680251 (2023-11-30): Fix temperature for Bresser 3-in-1 Wind Gauge (closes #2523)
closed 2023-11-30 by zuckschwerdt